pymmcore-plus/pymmcore-plus #6: Specify minimum MicroManager Version

ianhi, 2021-07-01:
With build `gamma1 20210211` running the example code gave me this error:

```python
2021-07-01 15:11:56.781 | DEBUG    | pymmcore_remote._client:ensure_server_running:118 - No server found, creating new mmcore server
2021-07-01 15:11:57.382 | DEBUG    | pymmcore_remote._client:__exit__:56 - closing pyro client

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_8868/2357010630.py in <module>
      4     # 'demo' is a special option for the included CMMCorePlus
      5     # that loads the micro-manager demo config
----> 6     mmcore.loadSystemConfiguration("demo")
      7     print(mmcore.getLoadedDevices())

~\mambaforge\envs\control\lib\site-packages\Pyro5\client.py in __call__(self, *args, **kwargs)
    474         for attempt in range(self.__max_retries + 1):
    475             try:
--> 476                 return self.__send(self.__name, args, kwargs)
    477             except (errors.ConnectionClosedError, errors.TimeoutError):
    478                 # only retry for recoverable network errors

~\mambaforge\envs\control\lib\site-packages\Pyro5\client.py in _pyroInvoke(self, methodname, vargs, kwargs, flags, objectId)
    241                     return _StreamResultIterator(streamId, self)
    242                 if msg.flags & protocol.FLAGS_EXCEPTION:
--> 243                     raise data  # if you see this in your traceback, you should probably inspect the remote traceback as well
    244                 else:
    245                     return data

OSError: Line 7: Device,DHub,DemoCamera,DHub
Failed to load device "DHub" from adapter module "DemoCamera" [ Failed to load device adapter "DemoCamera" from "C:\Program Files\Micro-Manager-2.0gamma\mmgr_dal_DemoCamera.dll" [ Incompatible device interface version (required = 70; found = 69) ] ]
```

which was resolved when I updated to the current nightly build. 

One small thing of note is that I had to restart my jupyter kernel in order to have the micromanager installer be able to replace `mmgr_dal_DemoCamera.dll` which may be a helpful thing to note if the fix for above is ever included in a troubleshooting section.

tlambert03, 2021-07-01:
yeah it's an issue, I asked Nico about this actually... unfortunately, there's no ideal way to know what device interface any given download from micromanger has without checking the git repo, or downloading at looking.  And it's not just a minimum version either, it's a maximum too since we have to be working with a pymmcore version that matches the library version.

there is an `install_mm.py` helper script in the root directory here that will help install a specific version for you ... but that too will go out of date when pymmcore updates itself (fortunately, pymmcore puts the device interface in the [version number](https://pypi.org/project/pymmcore/))

anyway, it would be a great feature here to a) make the helper script install a compatible version for the given pymmcore device interface (that might require a little manual annotation) and b) provide a perhaps an even more useful error message when someone runs into this

tlambert03, 2025-06-09:
this has since been significantly improved, both via the `mmcore install` just doing the right thing, as well as error messages clarifying the situation.  We can now check what device interface has been downloaded, and we know if it's a mismatch

closed 2025-06-09 by tlambert03