pymmcore-plus/pymmcore-plus #10: Add explanation of `register_callback` vs `RegisterCallback`

ianhi, 2021-07-03:
As in the title

tlambert03, 2021-07-03:
will add to docstrings... but before I do, lemme take a stab here and see if it makes sense. (it also needs some refactoring and renaming)

`registerCallback` is the native method from the [MMCore API](https://valelab4.ucsf.edu/~MM/doc/MMCore/html/class_c_m_m_core.html#aad1cc3b2feeb819f1c800f2805a8549e).  Unfortunately, as far as I can tell, it can only install a _single_ object (an instance of `pymmcore.MMEventCallback`) ... which makes it hard to do that standard `x.connect(callback)` pattern.  Hence this whole "other" callback system.
The API there is just a proof of principle, but the `register_callback` snake case method only lives on the RemoteMMCore  proxy object, it doesn't work for synchronous usage. The purpose of _that_ method is to install a callback object in the thread/process where the _proxy_ lives. For instance, in the [qt_example](https://github.com/tlambert03/pymmcore-plus/blob/main/examples/qt_integration.py) we use it to install a `QObject` that can receive callbacks from the server, and re-emit them in the local process for any local objects that may need to react.

Definitely confusing wording.  will try to clean that up

ianhi, 2021-07-05:
Ah I see. Also I noticed that if you want to follow signals like `onMDAFrame` you need to either use `register_callback`  or if you want to not use the remote functionality you need to call (what will then be confusingly named) `connect_remote_callback` with a custom object that has a `receive_core_callback` method. 

> egisterCallback is the native method from the MMCore API. Unfortunately, as far as I can tell, it can only install a single object (an instance of pymmcore.MMEventCallback) ... which makes it hard to do that standard x.connect(callback) pattern. Hence this whole "other" callback system.

Maybe the take here is that pymmcore-plus should be providing it's own callback system that incorporates both the standard signals and the new ones? This is what `remote` is currently doing, but probably makes sense to move that into the core part. 

As for the form of how to connect to signals is there a good reason to prefer the current style with methods on an object vs a matplotlib style of `mmcore.connect('onMDAFrame', my_function)`. The second form feels more natural to me, but that may just be due to familiarity with matplotlib. Or at the very least it would be nice to provide a convenience for users that accepts a function or perhaps an object to inherit from:

```python
class MMCorePlusCallback(pymmcore.MMEventCallback):
    def receive_core_callback(self, signal_name, args):
         # hasattr etc..
```

tlambert03, 2021-07-05:
> Maybe the take here is that pymmcore-plus should be providing it's own callback system that incorporates both the standard signals and the new ones? This is what remote is currently doing, but probably makes sense to move that into the core part.

absolutely agree.

> As for the form of how to connect to signals is there a good reason to prefer the current style with methods on an object vs a matplotlib style of mmcore.connect('onMDAFrame', my_function)

I'm just influenced by Qt, where `object.signal.connect(callback_function)` is the pattern.   On this one, I think I have to say that I prefer method approach to the string approach since it's more IDE-friendly (i.e. you can immediately tab complete the various callback names, and get a linting error if you try to connect to a non-existent one).  

I'm actually working on at this very moment on a pure-python Qt signals-slots implementation and will make a small package soon... (currently it's a [PR here](https://github.com/napari/superqt/pull/7), but I think i'll probably release it on its own).  Maybe once that is out, we can circle back here and throw out some various patterns for a refactor

closed 2021-07-15 by tlambert03