pymmcore-plus/pymmcore-plus #13: MDA auto writing to OME-Zarr?

ianhi, 2021-07-06:
What do you think of having the MDA function automatically handle the writing of data to a zarr on disk that conforms to https://ngff.openmicroscopy.org/latest/ ?

I looked into this a little bit but can't seem to find any easy to use `write to ome-zarr` libraries, only ones for reading. 

Alternatively, or perhaps in addition, it would be good to provide an example in `run_mda.py` of how to save the data to disk in a safe way. I was working on this when I realized that I really should be following the ome format.

ianhi, 2021-07-06:
A quick example that doesn't write to the OME spec, but does write to a zarr store:
```python
import zarr
class MDA_zarr_store:
    def __init__(self, core, store_name, img_shape=(512, 512), dtype=np.uint16):
        """
        Parameters
        ----------
        core : MMCorePlus
        store_name : str
            Should accept .format(run=INT)
        img_shape : (int, int)
        dtype : numpy dtype
        """
        self._core = core
        self._store_name = store_name
        self._run_number = -1
        self._img_shape = img_shape
        self._dtype = dtype
        self._core.connect_remote_callback(self)

    def receive_core_callback(self, signal_name, args):
        """Not for normal usage. Delegates callbacks to approriate methods."""
        if signal_name == "onMDAStarted":
            self._onMDAStarted(*args)
        elif signal_name == "onMDAFrameReady":
            self._onMDAFrame(*args)

    def _onMDAStarted(self, sequence):
        self._current_sequence = sequence
        self._run_number += 1
        self._z = zarr.open(
            self._store_name.format(run=self._run_number),
            mode="w",
            shape=sequence.shape + self._img_shape,
            dtype=self._dtype,
        )
        self._z.attrs["axis_order"] = sequence.axis_order + "yx"

    def _onMDAFrame(self, img, event):
        self._z[tuple(event.index[a] for a in self._current_sequence.axis_order)] = img

    def disconnect(self) -> None:
        """Disconnect this object from the MMCorePlus callbacks"""
        self._core.disconnect_remote_callback(self)
```
```python
MDA_storer = MDA_zarr_store(mmc, 'data/mda_test_{run}.zarr')
mmc.run_mda(sequence)
```

tlambert03, 2021-07-06:
hey, I definitely think some sort of builtin writers would be useful.  both ome.tiff and zarr would be great defaults.  I think it'd be nice if we could think of an API that has the core object calling the writer (rather than passing control of the core to the writer).  maybe something like this?

```python
from typing_extensions import Protocol  # typing extensions for 3.7 support

class MDAListener(Protocol):
    def start(self, seq: useq.MDASequence) -> None: ...
    def cancel(self, seq: useq.MDASequence) -> None: ...
    def finish(self, seq: useq.MDASequence) -> None: ...
    def frame(self, img: np.ndarray, event: useq.MDASequence) -> None: ...
    def pause(self, paused: bool) -> None: ...
```

and then the core can dispatch to the methods when ready...  just playing with the idea though.  not immediately sure it's better.  thoughts?

ianhi, 2021-07-06:
> I think it'd be nice if we could think of an API that has the core object calling the writer (rather than passing control of the core to the writer). 

:+1: I think that overall makes more sense except that in the sense that it is sort of special casing itself around using the signal mechanism that a user would have to use.  I came up with what I did by working within the bounds of the signal mechanism. But I like having something like a `writer` kwarg  to `MDA`.

Is there a clear set of libraries that are best for writing to these formats? My current top choice would probably be [aicsimageio](https://allencellmodeling.github.io/aicsimageio/index.html) in part due their push towards having good xarray support. But if there is a clear leader in the OME spec writing/reading in python space it makes sense to use that rather than any reimplementation?

tlambert03, 2021-07-06:
> I think that overall makes more sense except that in the sense that it is sort of special casing itself around using the signal mechanism that a user would have to use.

definitely true... though they're already kinda bound to the signals and their names in the other example too right?  I think it might be worth pondering this API a bit. not that that should stop you from creating a preliminary zarr writer.

> Is there a clear set of libraries that are best for writing to these formats?  My current top choice would probably be aicsimageio

That would probably be my choice too!  great library, great developers.  As for zarr, have you already played with the [writer](https://github.com/ome/ome-zarr-py/blob/master/ome_zarr/writer.py) in [ome-zarr-py](https://github.com/ome/ome-zarr-py)?

tlambert03, 2026-03-02:
ome-writers does this well now