pymmcore-plus/pymmcore-plus #15: Docs page

ianhi, 2021-07-08:
It would be good have a docs page (e.g. readthedocs) in addition to the now growing examples repo. I think this is a big time helper with discoverability. It's also nice to have structured place where we can put things like the explanation of connecting to signals.

----
FWIW readthedocs now supports [MyST[(https://myst-parser.readthedocs.io/en/latest/) which is what I would likely choose when starting a docs from scratch. Although that's aspirational as I never actually have used it.

tlambert03, 2021-07-08:
Yeah myst is awesome.  I've been using jupyter-book for docs in my last few projects, and loving it.  Unfortunately that's not compatible with rtd, but it works fine for GitHub pages

ianhi, 2022-02-24:
thinking about this again in the context of https://github.com/tlambert03/pymmcore-plus/issues/81. Do you have a preference for what docs system  to go with?

tlambert03, 2022-02-25:
Agreed, would be great.  I don't have any strong preferences.  sphinx with myst seems like a great place to start!