pymmcore-plus/pymmcore-plus #17: RemoteMMCore signal breaks if calling mmcore method in callback

ianhi, 2021-07-12:
I was working on updating napari-micromangaer to the new callback system an noticed that I wasn't able to get past loading a config. Tracking it down it looks like something goes wrong when you call a core method from inside a callback:

```python
from pymmcore_plus import RemoteMMCore
class blah_blah:
    def __init__(self, cfg='demo_config.cfg'):
        self._core = RemoteMMCore()
        self._core.systemConfigurationLoaded.connect(self._loaded)
        self._core.loadSystemConfiguration(cfg)

    def _loaded(self):
        print('pre camera')
        cam_device = self._core.getCameraDevice()
        print("got camera")
```
(Above is a pared down version of what breaks in napari-micromanger)

```python
2021-07-12 16:56:13.981 | DEBUG    | pymmcore_plus.client._client:ensure_server_running:161 - No server found, creating new mmcore server

here

---------------------------------------------------------------------------
ConnectionClosedError                     Traceback (most recent call last)
<ipython-input-5-23efa5e7068a> in <module>
      2 
      3 # mmcore = RemoteMMCore(verbose=True)
----> 4 blah = blah_blah()

<ipython-input-4-69f3e1643c9a> in __init__(self, cfg)
      3         self._core = RemoteMMCore()
      4         self._core.systemConfigurationLoaded.connect(self._loaded)
----> 5         self._core.loadSystemConfiguration(cfg)
      6 
      7     def _loaded(self):

~/mambaforge/envs/micro/lib/python3.8/site-packages/Pyro5/client.py in __call__(self, *args, **kwargs)
    474         for attempt in range(self.__max_retries + 1):
    475             try:
--> 476                 return self.__send(self.__name, args, kwargs)
    477             except (errors.ConnectionClosedError, errors.TimeoutError):
    478                 # only retry for recoverable network errors

~/mambaforge/envs/micro/lib/python3.8/site-packages/Pyro5/client.py in _pyroInvoke(self, methodname, vargs, kwargs, flags, objectId)
    222                 return None  # oneway call, no response data
    223             else:
--> 224                 msg = protocol.recv_stub(self._pyroConnection, [protocol.MSG_RESULT])
    225                 if config.LOGWIRE:
    226                     protocol.log_wiredata(log, "proxy wiredata received", msg)

~/mambaforge/envs/micro/lib/python3.8/site-packages/Pyro5/protocol.py in recv_stub(connection, accepted_msgtypes)
    187     """
    188     # TODO decouple i/o from actual protocol logic, so that the protocol can be easily unit tested
--> 189     header = connection.recv(6)  # 'PYRO' + 2 bytes protocol version
    190     ReceivingMessage.validate(header)
    191     header += connection.recv(_header_size - 6)

~/mambaforge/envs/micro/lib/python3.8/site-packages/Pyro5/socketutil.py in recv(self, size)
    433 
    434     def recv(self, size: int) -> bytes:
--> 435         return receive_data(self.sock, size)
    436 
    437     def close(self) -> None:

~/mambaforge/envs/micro/lib/python3.8/site-packages/Pyro5/socketutil.py in receive_data(sock, size)
    157                     err = ConnectionClosedError("receiving: not enough data")
    158                     err.partialData = data  # store the message that was received until now
--> 159                     raise err
    160                 return data  # yay, complete
    161             except socket.timeout:

ConnectionClosedError: receiving: not enough data
```

tlambert03, 2021-07-13:
thanks @ianhi, this is helpful, and concerning... i wonder if I might need to undo some of #14 ... 
in any case, now is as good as any to get the test coverage way up before we start extending.  I'll look into this, and let me know if you come across any additional insights

ianhi, 2021-07-14:
reading around I think https://pyro5.readthedocs.io/en/latest/clientcode.html#pyro-callbacks may hold the promise of a solution

tlambert03, 2021-07-14:
yeah, unfortunately that didn't do it.  played with it last night.  I just need to go back and look closer at how I implemented it before the changeover

ianhi, 2021-07-14:
I wonder if this is stretching the bounds of pyro5 a little bit? Even in it's callback [examples](https://github.com/irmen/Pyro5/tree/master/examples/callback) there is no example of the client callback calling the server again in it's callback. (although that it was working before pysygnal implies hope)

tlambert03, 2021-07-14:
> (although that it was working before pysygnal implies hope)

yep, I think I just can't make it the same object... before, there was a completely independent local object that was managing the callbacks...

tlambert03, 2021-07-14:
ok, turns it worked before because the callback object was a `QObject`, and is now a plain python object with `psygnals`.  If I rollback to v0.1.0 everything works with `QCoreCallback`, but then if I make an identical object using psygnal, it doesn't work:

```py
from psygnal import Signal

class PsygnalCallback:
    propertiesChanged = Signal()
    propertyChanged = Signal(str, str, object)
    channelGroupChanged = Signal(str)
    configGroupChanged = Signal(str, str)
    systemConfigurationLoaded = Signal()
    pixelSizeChanged = Signal(float)
    pixelSizeAffineChanged = Signal(float, float, float, float, float, float)
    stagePositionChanged = Signal(str, float)
    XYStagePositionChanged = Signal(str, float, float)
    exposureChanged = Signal(str, float)
    SLMExposureChanged = Signal(str, float)
    MDAFrameReady = Signal(object, object)
    MDAStarted = Signal(object)
    MDACanceled = Signal()
    MDAPauseToggled = Signal(bool)
    MDAFinished = Signal(object)

    def receive_core_callback(self, signal_name, args):
        if signal_name.startswith("on"):
            signal_name = signal_name[2:]
        if not signal_name[1].isupper():
            signal_name = signal_name[0].lower() + signal_name[1:]
        # let it throw an exception.
        getattr(self, signal_name).emit(*args)
```

This probably has something to do with the way callbacks are in Qt (which isn't always  an immediate synchronous callback... like psygnal's).  I don't think it's stretching the bounds of Pyro5, but concurrency is always hard, so we just need to figure out to let the call to `self._core.loadSystemConfiguration(cfg)` (which pyro treats as synchronous by default) receive the full response from the server before the stream is interrupted by another call to the server.

tlambert03, 2021-07-14:
another note to self:

this works:
```py
    core = RemoteMMCore()

    @core.systemConfigurationLoaded.connect
    def _cb():
        with RemoteMMCore() as _core:  # this is fine, it connects to the same core
            print("device is", _core.getCameraDevice())

    core.loadSystemConfiguration()
```

but this doesn't
```py
    core = RemoteMMCore()

    @core.systemConfigurationLoaded.connect
    def _cb():
        # core is busy calling `loadSystemConfiguration`
        print("device is", core.getCameraDevice())

    core.loadSystemConfiguration()
```

tlambert03, 2021-07-15:
This is definitely an interesting challenge, but probably not high enough priority for me at the moment (given that we have a satisfactory workaround by using Qt).  So, putting some general thoughts here for myself or whoever picks this up:

- An important rule about pyro Proxies is that you [cannot share Proxy objects among threads](https://pyro5.readthedocs.io/en/latest/clientcode.html#proxies-connections-threads-and-cleaning-up). One single thread ‘owns’ a proxy. It is possible to explicitly transfer ownership to another thread (using `_pyroClaimOwnership`)
- As encouraged in the [pyro callbacks docs](https://pyro5.readthedocs.io/en/latest/clientcode.html#pyro-callbacks) `RemoteMMCore` proxy [creates a new thread](https://github.com/tlambert03/pymmcore-plus/blob/37b04166745e1c289ad2b0d62290ee42acaa42d0/pymmcore_plus/client/_client.py#L91) for the callback object.
- The problem comes when a callback in the callback thread tries to directly call a method on the remote core object. 
   ```py
    @core.events.systemConfigurationLoaded.connect
    def cb():
        core.someMethod()  # problem
    core.loadSystemConfiguration()
    ```
    In a synchronous scenario, this is a problem because the remote proxy hasn't finished receiving bytes from the  `loadSystemConfiguration` at the moment `someMethod` is called.   This leads to the `Pyro5.errors.ConnectionClosedError: receiving: not enough data` error observed above.
- Qt signals solves this by doing smart things behind the scenes with Direct vs Queued connections (such that it effectively queues the callback for when the `loadSystemConfiguration` call has finished.
- We can "simulate" that a little bit with the non-Qt psygnal based callback object by using `.emit(..., asynchronous=True)` and using some `threading.Lock` or `threading.Condition` to handle the queuing behavior.  When I do that, I get around the `not enough data` error... but then I run into the typical pyro error of `Pyro5.errors.PyroError: the calling thread is not the owner of this proxy, create a new proxy in this thread or transfer ownership.` ... and I haven't yet found the correct place to "claim ownership" of the proxy to get around this.
- what's not clear to me is why the Qt queued connection case doesn't _also_ run into this issue, since according to the Qt docs "The_ slot is invoked when control returns to the event loop of the receiver's thread. **The slot is executed in the receiver's thread**."  So it seems like it should be violating the first rule mentioned above.

I'm sure there's a solution here somewhere that doesn't involve using Qt at all.  It's possible there may be some features that could be added upstream at psygnal, or perhaps it needs to be special cased here..
I'd like a clean end-user API to be the ultimate goal here (that is: ideally the remote client will be a strict drop-in replacement for the synchronous use case).

for the case of napari-micromanager, we'll just keep using the Qt system, which is added back in #19

ianhi, 2022-02-17:
I believe this is fixed now - and also soon to be obsolete by way of threads! (although the requirement of qt remains)

closed 2022-02-17 by ianhi