pymmcore-plus/pymmcore-plus #18: Tidbits from pyro5 docs

ianhi, 2021-07-14:
I was reading around the pyro5 docs due to #17 and noticed a few things that may be worth considering in the future:

1. https://pyro5.readthedocs.io/en/latest/tipstricks.html#binary-data-transfer-file-transfer
> Pyro wasn’t designed to transfer large amounts of binary data (images, sound files, video clips): the protocol is not optimized for these kinds of data. The occasional transmission of such data is fine but if you’re dealing with a lot of them or with big files, it is usually better to use something else to do the actual data transfer (file share+file copy, ftp, http, scp, rsync).

And then it recommends some alternatives

2. https://pyro5.readthedocs.io/en/latest/tipstricks.html#pyro-and-numpy essentially claims that numpy won't work, but it sure seems to here. Did you do any trickery or did we get lucky with the types we're using here?

tlambert03, 2021-07-14:
> Did you do any trickery or did we get lucky with the types we're using here?

I did some trickery: I don't serialize the array, i put it in a shared memory buffer and just pass a pointer:

https://github.com/tlambert03/pymmcore-plus/blob/37b04166745e1c289ad2b0d62290ee42acaa42d0/pymmcore_plus/_serialize.py#L29-L41

and then use that on the client side:

https://github.com/tlambert03/pymmcore-plus/blob/37b04166745e1c289ad2b0d62290ee42acaa42d0/pymmcore_plus/_serialize.py#L43-L47

tlambert03, 2021-07-15:
@ianhi, in the interest of keeping issues to "actionable" items... can we close this?  did understanding the SharedMemory trick answer your question?

ianhi, 2021-07-15:
Yes indeed. thanks!

--- 

I think one issue is that there isn't really a good place to ask questions other than issues. I imagine my questions will come up for others in the future as well so it's nice to have them somewhere permanent public and searchable (github discussions?)

closed 2021-07-15 by ianhi

tlambert03, 2021-07-15:
I definitely don't mind questions being asked in issues!  If you feel like you got it answered, you can go ahead and close them, otherwise feel free to ask follow up questions