pymmcore-plus/pymmcore-plus #21: Signal systems seems to break something on a real microscope

ianhi, 2021-07-16:
Went to give this a shot on a real microscope today and ran into a confusing bug with the latest release `0.1.3`

The following code works for `pymmcore.CMMCore` but fails for `pymmcore_plus.CMMCorePlus` on the `setConfig("Channel", "BF")`. By fails I mean that in jupyter my kernel would crash after ~1-2 seconds and no error messages would written anywhere.

```python
# %matplotlib ipympl
import matplotlib.pyplot as plt
from mpl_interactions import ipyplot as iplt
import os

from pymmcore_plus import CMMCorePlus
import pymmcore

mmcore = CMMCorePlus() # broken
# mmcore = pymmcore.CMMCore() # works! hooray

mm_dir = "C:/Program Files/Micro-Manager-2.0gamma"
mmcore.setDeviceAdapterSearchPaths([mm_dir])
mmcore.loadSystemConfiguration(os.path.join(mm_dir, "MMConfig_olympus_v2_1.cfg"))

# snapping images works in both cases
mmcore.setAutoShutter(True)
mmcore.snapImage()
img = mmcore.getImage()
plt.figure()
ctrls = iplt.imshow(img, vmin_vmax=("r", img.min(), img.max()))

print(mmcore.getAvailableConfigGroups())
# ('Channel', 'Objective')

print(mmcore.getAvailableConfigs('Channel'))
# ('BF', 'GFP', 'YFP')

# Potentially crash python
mmcore.setConfig('Channel', 'BF')

# good house keeping if we make it this far
# if it crashed prior then you'll probably need to restart the microscope (and potentially computer)
mmcore.unloadAllDevices()
```

@kobayashikirschvink suggested that that the major difference between pymmcore and the plus version was the callbacks and it seems that commenting out the connection to the core signals (including what's sent by `setConfig`) prevents the crash. In particular eliminating these lines:

https://github.com/tlambert03/pymmcore-plus/blob/58bd71537e02425dee3b68a9010701cca7efe906/pymmcore_plus/core/_mmcore_plus.py#L33-L34

and the above code runs with no issue. I'm going to try to debug this a bit more tomorrow (but only in person on account of needing to restart the scope when things go awry) but wanted to post in case you have any insights into this or thoughts on how to extract more of an error message.

For completeness below is the config file we are using

<details>
 <summary>Config File</summary>

```
# Reset
Property,Core,Initialize,0

# Devices
Device,COM6,SerialManager,COM6
Device,XYStage,Prior,XYStage
Device,Shutter-1,Prior,Shutter-1
Device,HamamatsuHam_DCAM,HamamatsuHam,HamamatsuHam_DCAM
Device,Olympus IX83,OlympusIX83,Olympus IX83
Device,Objective,OlympusIX83,Objective
Device,Dichroic 1,OlympusIX83,Dichroic 1
Device,EpiShutter 1,OlympusIX83,EpiShutter 1
Device,DiaShutter,OlympusIX83,DiaShutter
Device,TransmittedIllumination 1,OlympusIX83,TransmittedIllumination 1
Device,Transmitted Aperture,OlympusIX83,Transmitted Aperture
Device,FocusDrive,OlympusIX83,FocusDrive
Device,Shutter-2,Prior,Shutter-2

# Pre-init settings for devices
Property,COM6,AnswerTimeout,500.0000
Property,COM6,BaudRate,9600
Property,COM6,DataBits,8
Property,COM6,DelayBetweenCharsMs,0.0000
Property,COM6,Fast USB to Serial,Disable
Property,COM6,Handshaking,Off
Property,COM6,Parity,None
Property,COM6,StopBits,1
Property,COM6,Verbose,1
Property,XYStage,Port,COM6
Property,Shutter-1,Port,COM6
Property,Shutter-2,Port,COM6

# Pre-init settings for COM ports

# Hub (parent) references
Parent,Objective,Olympus IX83
Parent,Dichroic 1,Olympus IX83
Parent,EpiShutter 1,Olympus IX83
Parent,DiaShutter,Olympus IX83
Parent,TransmittedIllumination 1,Olympus IX83
Parent,Transmitted Aperture,Olympus IX83
Parent,FocusDrive,Olympus IX83

# Initialize
Property,Core,Initialize,1

# Delays
Delay,XYStage,1000.0

# Focus directions
FocusDirection,FocusDrive,0

# Roles
Property,Core,Camera,HamamatsuHam_DCAM
Property,Core,Shutter,Shutter-1
Property,Core,Focus,FocusDrive
Property,Core,AutoShutter,1

# Camera-synchronized devices

# Labels
# Objective
Label,Objective,5,6-UPLSAPO60XW
Label,Objective,4,5-UPLSAPO40X
Label,Objective,3,4-LUCPFL20XPH2
Label,Objective,2,3-UPLSAPO20X
Label,Objective,1,2-UPLSAPO10X
Label,Objective,0,1-UPLSAPO4X
# Dichroic 1
Label,Dichroic 1,7,8-NONE
Label,Dichroic 1,6,7-NONE
Label,Dichroic 1,5,Cy5
Label,Dichroic 1,4,TxRed
Label,Dichroic 1,3,YFP
Label,Dichroic 1,2,GFP
Label,Dichroic 1,1,DAPI
Label,Dichroic 1,0,1-NONE

# Configuration presets
# Group: Objective
# Preset: 40X
ConfigGroup,Objective,40X,Objective,Label,5-UPLSAPO40X

# Preset: 60XW
ConfigGroup,Objective,60XW,Objective,Label,6-UPLSAPO60XW

# Group: Channel
# Preset: YFP
ConfigGroup,Channel,YFP,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,Channel,YFP,HamamatsuHam_DCAM,Exposure,300.0090
ConfigGroup,Channel,YFP,Dichroic 1,Label,YFP
ConfigGroup,Channel,YFP,EpiShutter 1,State,1
ConfigGroup,Channel,YFP,DiaShutter,State,0
ConfigGroup,Channel,YFP,TransmittedIllumination 1,Brightness,0
ConfigGroup,Channel,YFP,Core,Shutter,Shutter-2
ConfigGroup,Channel,YFP,Olympus IX83,TouchPanel,0
ConfigGroup,Channel,YFP,Olympus IX83,FrontPanel,0
ConfigGroup,Channel,YFP,Shutter-1,State,0
ConfigGroup,Channel,YFP,Shutter-2,State,0

# Preset: GFP
ConfigGroup,Channel,GFP,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,Channel,GFP,HamamatsuHam_DCAM,Exposure,300.0090
ConfigGroup,Channel,GFP,Dichroic 1,Label,GFP
ConfigGroup,Channel,GFP,EpiShutter 1,State,1
ConfigGroup,Channel,GFP,DiaShutter,State,0
ConfigGroup,Channel,GFP,TransmittedIllumination 1,Brightness,0
ConfigGroup,Channel,GFP,Core,Shutter,Shutter-2
ConfigGroup,Channel,GFP,Olympus IX83,TouchPanel,0
ConfigGroup,Channel,GFP,Olympus IX83,FrontPanel,0
ConfigGroup,Channel,GFP,Shutter-1,State,0
ConfigGroup,Channel,GFP,Shutter-2,State,0

# Preset: BF
ConfigGroup,Channel,BF,HamamatsuHam_DCAM,Binning,2x2
ConfigGroup,Channel,BF,HamamatsuHam_DCAM,Exposure,50.0080
ConfigGroup,Channel,BF,Dichroic 1,Label,1-NONE
ConfigGroup,Channel,BF,DiaShutter,State,1
ConfigGroup,Channel,BF,TransmittedIllumination 1,Brightness,100
ConfigGroup,Channel,BF,EpiShutter 1,State,0
ConfigGroup,Channel,BF,Core,Shutter,Shutter-1
ConfigGroup,Channel,BF,Olympus IX83,TouchPanel,0
ConfigGroup,Channel,BF,Olympus IX83,FrontPanel,0
ConfigGroup,Channel,BF,Shutter-1,State,1
ConfigGroup,Channel,BF,Shutter-2,State,0

# PixelSize settings

```
</details>

ianhi, 2021-07-16:
I should also note that on the same computer, same python kernel etc, both `plus` and stock pymmcore run the example just fine if I load the demo config instead. So it's possible something is broken in the config, but in that case it would still be good if we could figure out how to make things fail more gracefully.

----

Have you had a chance to use `pymmcore_plus` on a real scope yet? I have access to another one (nikon instead of olympus) where I may be able to try this tomorrow as well to see if this is particular to the system or something deeper.

tlambert03, 2021-07-16:
thanks for the report!  I agree with @kobayashikirschvink (and your observations certainly confirm) that the callbacks are the primary "opportunity" for an error in `pymmcore_plus` and we can definitely do some more there to catch and report errors.  We can definitely add a layer of error catching there which will hopefully (?) prevent a full crash, but hard to say at this point (the fact that you don't get a stack trace at all kinda suggests something on the C side, but not sure).

When you get a moment, can you try connecting a plain, empty, pymmcore callback: `mmcore.registerCallback(pymmcore.MMEventCallback())` (but keep the _callback_relay part commented out). You can also try making a more manual callback object with only a subset of methods implemented:

```py
class MyCallbacks(pymmcore.MMEventCallback):
    def onPropertiesChanged(self):
        print("propertiesChanged")
    def onPropertyChanged(self, dev_name: str, prop_name: str, prop_val: str):
        print("onPropertyChanged", dev_name, prop_name, prop_val)
```

... or wrap the re-emission process here in a try/except block:

https://github.com/tlambert03/pymmcore-plus/blob/58bd71537e02425dee3b68a9010701cca7efe906/pymmcore_plus/core/_mmcore_plus.py#L156 

> Have you had a chance to use pymmcore_plus on a real scope yet?

only cameras! so you're definitely treading new territory! :)

tlambert03, 2021-07-16:
one more observation:
if I force an exception in the callback (~line 156 in mmycore_plus.py)
```py
        def reemit(self: _MMCallbackRelay, *args):
                raise ValueError
```
then I also get a jupyter kernel error with no exception printed... so let's definitely start there:

```py
        def reemit(self: _MMCallbackRelay, *args):
            try:
                getattr(self._emitter, sig_name).emit(*args)
            except Exception as e:
                print(f"ERROR: {e}")

```

ianhi, 2021-07-16:
I came in and so was able to debug in person and am thoroughly bogged down in weirdness (this may end up not being a pymmcore-plus issue, but may have lessons for it)

It turns out that when I register a pure `pymmcore.MMEventCallback` there is no issue and it prints out all the callbacks to the terminal (Though there is some multiple calling of callbacks perhaps)
(These prints come from here: https://github.com/micro-manager/mmCoreAndDevices/blob/97f82b9ef02788d8b6e9aa70283eb10c7c645a6e/MMCore/MMEventCallback.h)
```
onPropertyChanged() Core Shutter Shutter-1
onPropertyChanged() DiaShutter State 1
onConfigGroupChanged() Channel BF
onPropertyChanged() TransmittedIllumination 1 Brightness 100 
onConfigGroupChanged() Channel BF
onPropertyChanged() EpiShutter 1 State 0
onConfigGroupChanged() Channel BF 
```

However if I switch to using a subclass of that object that doesn't introduce any changes then I do see the crash:

```python
class MyCallbacks(pymmcore.MMEventCallback):
    def __init__(self):
        super().__init__()

# cb = pymmcore.MMEventCallback()
cb = MyCallbacks()
mmcore.registerCallback(cb)
```

However, I only see this crash when changing the config, if I do something like set the `X` position then the callback works the same as the pymmcore version.

This is deeply confusing because I thought that subclassing an object like that should be exactly the same as having the same object

tlambert03, 2021-07-16:
ugh... I feared something like this :/

> this may end up not being a pymmcore-plus issue, but may have lessons for it ...
>
> This is deeply confusing because I thought that subclassing an object like that should be exactly the same as having the same object

yep, I think you're right... and yeah, it's definitely confusing.  If there is a fix, I suspect it would be introduced in the SWIG wrapping process itself (i.e. [this file](https://github.com/micro-manager/pymmcore/blob/main/pymmcore/pymmcore_swig.i))... or even worse, in the C code itself.   That is, it's possible that the callback system is subtly incompatible with the `SwigPyObject` wrapping `pymmcore.MMEventCallback` (see [here](http://www.swig.org/Doc4.0/Python.html#Python_nn28) for a bit more, and note that pymmcore currently uses the `-builtin` flag) but that's just a guess.

You're saying this only works with a certain device though right?  Like, the demo config is fine?
If so, and you're _really_ interesting in digging in 😂 , i wonder if you could remove devices one at a time from the config?

tlambert03, 2021-07-16:
man... the C code is _very_ minimal.  it simply calls the callback in two places:  ([here](https://github.com/micro-manager/mmCoreAndDevices/blob/65c776897c065f37d78440f9be58c85bcfe6e1d7/MMCore/MMCore.cpp#L6899), and [here](https://github.com/micro-manager/mmCoreAndDevices/blob/65c776897c065f37d78440f9be58c85bcfe6e1d7/MMCore/MMCore.cpp#L6926))... _really_ hard to see what could be going wrong there.

ianhi, 2021-07-16:
> or even worse, in the C code itself.

:' (

--- 
One important piece of information I forgot (!!) is that I tried this this morning on different microscope (nikon te-2000, probably different python and micromanager versions) and it didn't have this same issue. 

I would like to fix this as, but for the meantime on this system will just disable the core callbacks and live without them.

> You're saying this only works with a certain device though right? Like, the demo config is fine?

Yes I can try next week to spend some time going piece by piece and seeing what happens. The feedback loop is pretty slow unfortunately as I seem to need to both restart the scope and log in/log out in order to free up all the connections after the crash.

ianhi, 2021-07-16:
>  it simply calls the callback in two places: 

I saw that as well. But I think it must be being called from somewhere else because other methods on that object are getting called (e.g. `onPropertyChanged` is used in either of the places you linked)

tlambert03, 2021-07-16:
> One important piece of information I forgot (!!) is that I tried this this morning on different microscope (nikon te-2000, probably different python and micromanager versions) and it didn't have this same issue.> The feedback loop is pretty slow unfortunately as I seem to need to both restart the scope and log in/log out in order to free up all the connections after the crash.

oof! that's annoying.

ianhi, 2021-08-12:
Fixed by https://github.com/micro-manager/pymmcore/pull/50 hoooooorayyyyy!!

ianhi, 2021-08-12:
We maybe should require the next release of pymmcore in order to avoid anyone else running into this?

closed 2021-08-12 by ianhi

tlambert03, 2021-08-12:
really great work @ianhi!  i agree... we should pin the new version