pymmcore-plus/pymmcore-plus #23: ENH: Override `mmcore.getLoadedDevicesOfType`

ianhi, 2021-07-16:
## Problem

Currently `mmcore.getLoadedDevicesOfType(...)` only accepts `ints` which come from the variables defined by `pymmcore` such as:
```python
pymmcore.CoreDevice
pymmcore.GalvoDevice
....
```

The typing and docstring on shift-tab doesn't suggest that this is the way to access them 

## Proposed Solution

Override this method and accept strings such as `Focus`, `Core`, `Stage` etc which we then map to the appropriate `int`.

If this seems good I'm happy to open a PR

tlambert03, 2021-07-16:
I've got a PR where I'm turning those random integers into [proper enums](https://github.com/tlambert03/pymmcore-plus/pull/16/files#diff-02329d9130680789c8bd41ab357c51be5e44fb92f60eefb454cabc5974beca17R6).

Once that's in, I'd be fine with accepting an int, string, or enum... and we could just do:
```py
val = int(DeviceType[x] if isinstance(x, str) else DeviceType(x))
```

will ping when that's in

tlambert03, 2021-07-17:
#16 is now in if you'd like to submit a PR.  #16 handled a few cases of returning special enums, but doesn't handle function parameters.

other functions that use DeviceType:
```py
def deviceTypeBusy(self, devType: DeviceType) -> bool: ...
def getLoadedDevicesOfType(self, devType: DeviceType) -> Tuple[str]: ...
def waitForDeviceType(self, devType: DeviceType) -> None: ...
```

tlambert03, 2023-07-27:
yep, this works now:
```python
from pymmcore_plus import CMMCorePlus, DeviceType

core = CMMCorePlus()
core.loadSystemConfiguration()
core.getLoadedDevicesOfType(DeviceType.CameraDevice)  # ('Camera',)
```

closed 2023-07-27 by tlambert03