pymmcore-plus/pymmcore-plus #30: question: How does cancelling an MDA work?

ianhi, 2021-07-31:
If I start a remotemmcore and then start an MDA then the core thread seems like it should be constantly blocked: https://github.com/tlambert03/pymmcore-plus/blob/654715380ba81c35762ea084b0e943ac78014a7e/pymmcore_plus/core/_mmcore_plus.py#L166-L181

but if I call `mmcore.cancel()` then somehow the underlying thread can break out of that loop to update the value of `_cancelled`, how is this happening?

tlambert03, 2021-07-31:
it's a good question!  I need to dig a bit more through the pyro code to give you a satisfying answer.  but my assumption is that there are multiple threads at play.  And as long as there is some thread switching and an object that can be accessed across threads, then it's possible.  Simple example:

```py
from threading import Thread
import time

class C:
    _canceled = False

    def run(self):
        while not self._canceled:
            time.sleep(0.5)
            print("running")
        print("done")

    def cancel(self):
        self._canceled = True

c = C()
thread = Thread(target=c.run)
thread.start()

# run the above... then when you're ready:
# c.cancel()
```

I wouldn't blame you for not being super satisfied with that answer... but the key lies in Pyro, and I need to keep digging

ianhi, 2021-08-02:
I'm actually fairly satisfied now that I know that all the ingredients to make this work in a sensible way are here. I didn't know that threads could modify variables on other threads like that, but with that knowledge this feels much less spooky.

Thanks!

closed 2021-08-02 by ianhi

tlambert03, 2021-08-02:
>  I didn't know that threads could modify variables on other threads like that,

yep!  this is the whole "problem" with threads actually!  Threads share memory space, which can include python objects, it's convenient when you _want_ different threads to mutate state on a single python object (like here), but tricky when you start getting conflicting thread-local state (i.e. race conditions)