pymmcore-plus/pymmcore-plus #43: How to recover if connection to pyro server lost

ianhi, 2021-08-14:
I've been running into this error recently:

```
ctionResetError                      Traceback (most recent call last)
~\miniconda3\envs\control-pymm\lib\site-packages\Pyro5\socketutil.py in send_data(sock, data)
    181         try:
--> 182             sock.sendall(data)
    183             return

ConnectionResetError: [WinError 10054] An existing connection was forcibly closed by the remote host

During handling of the above exception, another exception occurred:

ConnectionClosedError                     Traceback (most recent call last)
~\AppData\Local\Temp/ipykernel_10752/3470287651.py in <module>
----> 1 mmcore.getExposure()

~\miniconda3\envs\control-pymm\lib\site-packages\Pyro5\client.py in __call__(self, *args, **kwargs)
    474         for attempt in range(self.__max_retries + 1):
    475             try:
--> 476                 return self.__send(self.__name, args, kwargs)
    477             except (errors.ConnectionClosedError, errors.TimeoutError):
    478                 # only retry for recoverable network errors

~\miniconda3\envs\control-pymm\lib\site-packages\Pyro5\client.py in _pyroInvoke(self, methodname, vargs, kwargs, flags, objectId)
    217             protocol.log_wiredata(log, "proxy wiredata sending", msg)
    218         try:
--> 219             self._pyroConnection.send(msg.data)
    220             del msg  # invite GC to collect the object, don't wait for out-of-scope
    221             if flags & protocol.FLAGS_ONEWAY:

~\miniconda3\envs\control-pymm\lib\site-packages\Pyro5\socketutil.py in send(self, data)
    430 
    431     def send(self, data: bytes) -> None:
--> 432         send_data(self.sock, data)
    433 
    434     def recv(self, size: int) -> bytes:

~\miniconda3\envs\control-pymm\lib\site-packages\Pyro5\socketutil.py in send_data(sock, data)
    185             raise TimeoutError("sending: timeout")
    186         except socket.error as x:
--> 187             raise ConnectionClosedError("sending: connection lost: " + str(x))
    188     else:
    189         # Socket is in non-blocking mode, use regular send loop.

ConnectionClosedError: sending: connection lost: [WinError 10054] An existing connection was forcibly closed by the remote host
```

and when I close my jupyter kernel I get warnings that the subprocess is still running. Is there an easy way to try to kill off the server? Or to reconnect without rejection (I'm remote controlling so I need to be able to unload the devices to be able to reset things. https://github.com/tlambert03/pymmcore-plus/issues/42)

Interestingly I seem to be able to send some commands before this happens. In this case I was able to set the config but then got this error on querying the exposure. This is with the latest pymmcore with the threads fix, so it's possible that three is some error in the pymmcore-plus callbacks that's being eaten by pyro.

tlambert03, 2021-08-14:
yeah, I've seen this too ... haven't had time to dig in.  

> Is there an easy way to try to kill off the server?

There's a little bit of logic that already exists to detect and attach to an existing process, or kill an existing one.  So you could start there.  For instance, see [`ensure_server_running`](https://github.com/tlambert03/pymmcore-plus/blob/244efd06bb1c957611e5e3f6e3b721655fecec28/pymmcore_plus/client/_client.py#L113) for the code that tries to detect (in the remote process) whether there is an existing server, or whether a new one is needed.
You'll also see in there the attempt to clean up any rogue servers once we reattach by using [`_get_remote_pid`](https://github.com/tlambert03/pymmcore-plus/blob/244efd06bb1c957611e5e3f6e3b721655fecec28/pymmcore_plus/client/_client.py#L80) to get the process id of the rogue process so it can be  killed with `proc.kill`

but it'd be great to figure out why we're losing the connections, and try to reconnect without killing the server

ianhi, 2021-08-14:
> yeah, I've seen this too ... haven't had time to dig in.

Has this been recent? I kept some old version of this installed for a while because I had local changes eliminating the callbacks so perhaps this is related to changes that no greater than ~30 days old?

tlambert03, 2021-08-14:
possible?  I did see this within the last month.   my gut feeling is that it doesn't have anything to do with recent changes, but your guess is as good as mine!

ianhi, 2021-09-13:
See https://github.com/tlambert03/napari-micromanager/pull/54/checks?check_run_id=3591154951 for a nice capturing of these sorts of errors. maybe it's occuring when some unhandled error happens on the remote and for some reason pyro isn't propogating it

tlambert03, 2023-07-27:
remote is removed in #208

closed 2023-07-27 by tlambert03