pymmcore-plus/pymmcore-plus #44: Config objects not serializable for pyro

ianhi, 2021-08-20:
Your nice new config objects seem to only work on CMMCorePlus, but not remote:

```python
from pymmcore_plus import RemoteMMCore, CMMCorePlus
mmc = RemoteMMCore() # broken
# mmc = CMMCorePlus() # works

mmc.loadSystemConfiguration('demo')
config = mmc.getConfigGroupState('Channel')
```
```python
---------------------------------------------------------------------------
SerializeError                            Traceback (most recent call last)
/tmp/ipykernel_40955/3511616965.py in <module>
      3 
      4 mmc.loadSystemConfiguration('demo')
----> 5 config = mmc.getConfigGroupState('Channel')

~/mambaforge/envs/pymm/lib/python3.8/site-packages/Pyro5/client.py in __call__(self, *args, **kwargs)
    474         for attempt in range(self.__max_retries + 1):
    475             try:
--> 476                 return self.__send(self.__name, args, kwargs)
    477             except (errors.ConnectionClosedError, errors.TimeoutError):
    478                 # only retry for recoverable network errors

~/mambaforge/envs/pymm/lib/python3.8/site-packages/Pyro5/client.py in _pyroInvoke(self, methodname, vargs, kwargs, flags, objectId)
    234                 if self._pyroRawWireResponse:
    235                     return msg
--> 236                 data = serializer.loads(msg.data)
    237                 if msg.flags & protocol.FLAGS_ITEMSTREAMRESULT:
    238                     streamId = bytes(msg.annotations.get("STRM", b"")).decode()

~/mambaforge/envs/pymm/lib/python3.8/site-packages/Pyro5/serializers.py in loads(self, data)
    424 
    425     def loads(self, data):
--> 426         return msgpack.unpackb(self._convertToBytes(data), raw=False, object_hook=self.object_hook, ext_hook=self.ext_hook)
    427 
    428     def default(self, obj):

msgpack/_unpacker.pyx in msgpack._cmsgpack.unpackb()

~/mambaforge/envs/pymm/lib/python3.8/site-packages/Pyro5/serializers.py in object_hook(self, obj)
    456     def object_hook(self, obj):
    457         if "__class__" in obj:
--> 458             return self.dict_to_class(obj)
    459         return obj
    460 

~/mambaforge/envs/pymm/lib/python3.8/site-packages/Pyro5/serializers.py in dict_to_class(cls, data)
    231                     return SerializerBase.make_exception(exceptiontype, data)
    232         log.warning("unsupported serialized class: " + classname)
--> 233         raise errors.SerializeError("unsupported serialized class: " + classname)
    234 
    235     @staticmethod

SerializeError: unsupported serialized class: pymmcore_plus.core._config.Configuration
```

tlambert03, 2021-08-20:
Oops! Thanks, forgot to add a serializer.

tlambert03, 2021-08-20:
oh actually... this fails with the native class too right?
```
config = mmc.getConfigGroupState('Channel', native=True)  # this falls back to the builtin behavior
```

```
SerializeError: unsupported serialized class: pymmcore.pymmcore_swig.Configuration
```

ianhi, 2021-08-20:
Yup - they both fail, is that a bigger problem to fix?

tlambert03, 2021-08-20:
it will be easier to fix for the new "non-native" types... but neither is terribly hard to fix.  just need to define functions that turn instances of these classes into a dict, and back into an instance.  will work on it

PR #46 (2021-08-21): Fix serialization of Configuration and Metadata objects
fixes #44 

This still doesn't fix the case of using the `native=True` option (returning a native `pymmcore` objects) ... but those should be less likely.
Tried to find a way to make sure we can serialize everything needed... but it was difficult
closed 2021-08-21 by tlambert03