pymmcore-plus/pymmcore-plus #51: Autofocus in mda

ianhi, 2021-09-01:
Cross linking with https://github.com/tlambert03/useq-schema/issues/4

But perhaps prior to that it would be nice to have a signal emitted in MDA after the hardware is set up. Something similar to the pycromanager `post_hardware_hook` (https://pycro-manager.readthedocs.io/en/latest/apis.html?highlight=post_hardware#pycromanager.Acquisition)

tlambert03, 2021-09-02:
yep, we generally need to add hooks!  But I agree, this feels like something that the core run_mda should look for

ianhi, 2023-01-19:
I've been working towards using autofocus with pymmc+ and napari-micromanager. I did all these experiments on a nikon TE-2000 with the first generation of nikon's PFS. (heads up @fdrgsp that I got this working :) )

In this case we have two Focus devices (PFS, and Focus (a standard z-stage)).

The basic way to do autofocus+z-stack in an MDA is to 
1. At the start of each position use PFS to find the 0 position of the z-stack
2. Use the z-stage to perform a z-stack

I think that we can do some intelligent guessing w.r.t which stage to use - however, I think that that should live at the level of napari-micro and the default mda-engine does strictly as instructed by the `MDASequence`. Thus there will need be an extension of the `ZPlan`

In the interim I've made this custom engine that performs autofocusing which hopefully can serve as a basis for adding this to the default engine.

```python
from pymmcore_plus.mda import MDAEngine
from useq import MDAEvent, MDASequence

class AutoFocusEngine(MDAEngine):
    def setup_sequence(self, sequence: MDASequence) -> None:
        super().setup_sequence(sequence)
        self._z_rel = sequence.z_plan.positions()
        if "autofocus" in sequence.metadata:
            auto_meta = sequence.metadata["autofocus"]
            self._auto_device = auto_meta["autofocus_device"]
            self._rel_device = auto_meta["rel_focus_device"]
        self._last_pos = -1

    def setup_event(self, event: MDAEvent):
        if event.x_pos is not None or event.y_pos is not None:
            x = event.x_pos if event.x_pos is not None else self._mmc.getXPosition()
            y = event.y_pos if event.y_pos is not None else self._mmc.getYPosition()
            self._mmc.setXYPosition(x, y)

        if event.z_pos is not None:
            if event.index["p"] != self._last_pos:
                # moved to a new position
                # figure out what the PFS-Offset was
                pfs_z = event.z_pos - self._z_rel

                # Set the PFS value
                self._mmc.setPosition(self._auto_device, pfs_z[0])

                # Perform the autofocus
                self._mmc.fullFocus()

                # Get the position of the stage
                self._ref_z = self._mmc.getPosition(self._rel_device)

                # make sure we turn PFS off
                self._mmc.enableContinuousFocus(False)
                self._mmc.waitForSystem()
                self._last_pos = event.index["p"]
            # move in the z stack
            z_pos = self._ref_z + self._z_rel[event.index["z"]]
            self._mmc.setPosition(self._rel_device, z_pos)

        if event.channel is not None:
            self._mmc.setConfig("Filter", event.channel.config)

        if event.exposure is not None:
            self._mmc.setExposure(event.exposure)

        self._mmc.waitForSystem()
```

and was using it with this sequence:

```python
from napari_micromanager._mda_meta import SequenceMeta
from useq import MDASequence

mda = MDASequence(
    metadata={
        "autofocus": {"autofocus_device": "PFS-Offset", "rel_focus_device": "Focus"},
        "napari_mm_sequence_meta": SequenceMeta(
            mode="mda"
        ),  # hack to make napari micro recognize this mda
    },
    stage_positions=[
        (65336.0, 37236, 1.72),
        (65148.00, 37224.00, 1.72),
        (64849.00, 37247.00, 1.72),
        (64586.00, 37271.00, 1.72),
    ],
    channels=[
        {"config": "BF", "group": "Filter", "exposure": 10},
        {"config": "GFP", "group": "Filter", "exposure": 200},
        {"config": "dsred", "group": "Filter", "exposure": 200},
    ],
    time_plan={"interval": 60 * 10, "loops": 6 * 16},
    z_plan={"range": 4, "step": 0.5},
    axis_order="tpcz",
)
```

tlambert03, 2023-01-20:
looks good to me, including some representation in useq-schema.  I'd like to have a quick peek on the scope to see how fullFocus and getSet position are working with the two stages.

the one thing I find myself wondering is whether _all_ autofocus devices end up working similarly in micromanager, or if that interface leaves much to interpretation (i.e. do they all "find the 0 position of the z-stack"?  or so some mediate the maintenance of distance between the lens and the coverslip in slightly different ways).  That's a bit out of our hands if it is variable, but just something i wonder

PR #204 (2023-07-27): feat: update engine with autofocus using user Action
This pr is a followup of #199 but uses the new `MDAEvent` `Action` logic for hardware autofocus: https://github.com/pymmcore-plus/useq-schema/pull/94.

closes #199
closes #51

Linked to https://github.com/pymmcore-plus/useq-schema/pull/96.
closed 2023-07-27 by tlambert03