pymmcore-plus/pymmcore-plus #66: how should finding local MM install behave?

ianhi, 2021-12-01:
https://github.com/tlambert03/napari-micromanager/pull/80 is currently failing because the local_install part of `find_micromanager` doesn't detect the micromanager installation that was added to that folder. This is because it uses the location of the file in pymmcore:

https://github.com/tlambert03/pymmcore-plus/blob/7d9df92a73c15551f3bd22c75baf104a644fc5c2/pymmcore_plus/_util.py#L22

this seems like maybe a vestige of everything being in napari-micromanager. Perhaps a more sensible behavior is to change that line to:

```
 local_install = list(Path(".").glob(f"**/Micro-Manager*{sfx}")) 
```
?

So that it's local to wherever the python was run from?

ianhi, 2021-12-01:
I guess it could search if it exists in any of the directories in the PATH

tlambert03, 2022-02-28:
i think this can be closed, as this has been pretty stable now.  For the record though, I never like using anything like `Path(".")` that changes behavior depending on the current directory of whatever is executing the program

closed 2022-02-28 by tlambert03