pymmcore-plus/pymmcore-plus #77: squash merging not maintaining commit text?

ianhi, 2022-02-19:
The way #73 was merged squashed the commits (which is fine with me) but a bit disconcertingyl seems to have lost all the text of subcommits: https://github.com/tlambert03/pymmcore-plus/commit/2023d2335d3543714d89ad3990da025e5b285447 which is less great because there was useful information in some of them. Not a big deal so long as it all remains on github but nice to have that information duplicated in the commit body (if this is an easy fix)

tlambert03, 2022-02-19:
would need to look into rewriting that commit message.  keep your branch around for now

tlambert03, 2022-02-28:
gonna close this as not really actionable.  Looked through some of the commit messages.  In many cases, they could actually be added as comments to the code, which I would prefer.  For example:

> Has to be this way (single underscore global variable) because
swig seems to prevent us from modifying class attributes, and
double underscore triggers name mangling in a class method:

could be a comment in code rather than in commit

closed 2022-02-28 by tlambert03