pymmcore-plus/pymmcore-plus #79: Issue with synchronized decorators

tlambert03, 2022-02-20:
```py
In [1]: from pymmcore_plus import CMMCorePlus

In [2]: mmc = CMMCorePlus()

In [3]: mmc.loadSystemConfiguration()

In [4]: mmc.setPosition(0.0)
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
Input In [4], in <module>
----> 1 mmc.setPosition(0.0)

File ~/miniconda3/envs/napdev/lib/python3.9/site-packages/wrapt/decorators.py:469, in synchronized.<locals>._synchronized(wrapped, instance, args, kwargs)
    463 @decorator
    464 def _synchronized(wrapped, instance, args, kwargs):
    465     # Execute the wrapped function while the original supplied
    466     # lock is held.
    468     with lock:
--> 469         return wrapped(*args, **kwargs)

TypeError: setPosition() missing 1 required positional argument: 'val'
```

cc @ianhi, can you take a look?

ianhi, 2022-02-20:
Ah not actually an issue with the decortator but rather an issue with now having `setPosition` be a method on the python side. Previously it was only on C++ side so `setPosition` was overloaded:

[image]

but I only implemented one of them (namely: `(str, float)`) in python. Is there a good way to do overloading in python? I found https://docs.python.org/3/library/typing.html#typing.overload but that doesn't work because it doesn't seem to have run time effects.

ianhi, 2022-02-20:
I don't know if there is a good way to do this in python while maintaining reasonable argument names. I guess we could do *args but hten you lose typing information, or have `stageLabel` sometimes be a float (which isn't great)

ianhi, 2022-02-20:
I guess that the issues is that `setPosition` would ideally have been `setPosition(position, stageLabel)` which would have allowed for easy translation to python as `setPosition(positoin, stageLabel=None)`, but thats not the owrld we live in :(

tlambert03, 2022-02-20:
oh good!  that's easier to fix :)

> I guess we could do *args but then you lose typing information

that's where `typing.overload` comes in.  Something like this should work

```python
    @overload
    def setPosition(self, position: float) -> None:
        """Sets the position of the stage in microns."""
    @overload
    def setPosition(self, stageLabel: str, position: float) -> None:
        """Sets the position of the stage in microns."""

    @synchronized(lock)
    def setPosition(self, *args: Union[str, float]) -> None:
        return super().setPosition(*args)

```

tlambert03, 2022-02-20:
in other words, the way you handle overloads in python is that you only get one function signature, and it has to be compatible with all of the ways you want to call it.  but `typing.overload` lets you provide as many signatures as you want, just for the sake of typing

ianhi, 2022-02-20:
I guess we can do this:

```python
from functools import singledispatch
@singledispatch
def setPosition(arg: str, position: Number=None):
    print("both")

@setPosition.register
def _(arg: int):
    print('ehre!')

```

tlambert03, 2022-02-20:
we don't really need to bring singledispatch into this, since pymmcore itself already handles the overload.  (we'd need singledispatch if _we_ were trying to implement the overload ... but simply passing *args along does everything we need)

ianhi, 2022-02-20:
I was hoping that reimplementing the overload would let us keep more typing information but if it's lost I suppose it's not hte end of the world

tlambert03, 2022-02-20:
it's not lost!  that's the whole point of typing.overload.  did you see that above?  With the above code using `typing.overload` you get this:

[image]

it's fantastic.  this is how typing overloads is done

tlambert03, 2022-02-20:
you can see all the typing overloads I added upstream here: https://github.com/micro-manager/pymmcore/blob/main/pymmcore/__init__.pyi

ianhi, 2022-02-20:
oooooh I like that. very good lets do that

PR #82 (2022-02-21): restore setPosition overloading
fix: #79
closed 2022-02-21 by tlambert03