pymmcore-plus/pymmcore-plus #100: Proposal - add a new MDA signal.

ianhi, 2022-03-03:
I know that long term we need a better `useq` to allow for full control of MDAs - so this is more of a stop gap suggestion:

Thoughts on adding a signal to be emitted between the `waitForSystem` and `snapImage`? 

https://github.com/tlambert03/pymmcore-plus/blob/2fcf5fce074cedb8c940c472111db3055857f9e2/pymmcore_plus/core/_mmcore_plus.py#L517-L519

I think this is would be equivalent to pycromanager's `post_hardware_hook` (https://pycro-manager.readthedocs.io/en/latest/acq_hooks.html#acq-hooks) which I've found to be useful in the past. 

Use cases include making slight adjustments to positioning, or potentially throwing in a an outside of Micromanager acquisition there. For example my laser exposures aren't z-stacked, and also can't be captured by useq (currently) and there is no good way to hook into the MDA loop and do the laser exposure when I'm at Zrel=0.

tlambert03, 2022-03-03:
Definitely ok with this.

Since all the signals coming out of run_mda are custom to pmmc+ anyway, we can add more no problem.

tlambert03, 2023-07-27:
For now, I think we're in a good place with flexibility without the need for additional signals and hooks.  I believe many of pycromanagers hooks are a result of working within a ZMQ inter-process kind of setup, where pickling class instances isn't so trivial.  Here, we have much more flexibility, so before we just re-implement those patterns, I'd like to pause at the current state. So, for reference, here is how you could subclass our current engine and perform all of these hooks:

```python
from typing import Iterable, Iterator

from pymmcore_plus import CMMCorePlus
from pymmcore_plus.mda import MDAEngine
from useq import MDAEvent

# subcla
class MyEngine(MDAEngine):
    def setup_event(self, event: MDAEvent):
        # pre-hardware hook
        super().setup_event(event)
        # post-hardware hook

    def exec_event(self, event: MDAEvent):
        super().exec_event(event)
        # self._mmc.getImage() <- image_proces_fn
        # post-camera hook
    
    def event_iterator(self, events: Iterable[MDAEvent]) -> Iterator[MDAEvent]:
        # event_generation_hook_fn
        yield from events

core = CMMCorePlus.instance()
core.mda.set_engine(MyEngine())

core.mda.run(...)
```

closed 2023-07-27 by tlambert03