pymmcore-plus/pymmcore-plus #101: Make an MDA object that's owned by pymmcore

ianhi, 2022-03-03:
It's a bit awkward to have MMCore have methods like `cancel` and  properties like `pause` interspersed with other methods as it is not longer obvious that they are associated with MDAs. The same also goes for the signals, we've intermixed MDA signals with normal microscope control signals. 

What about having an `mda`  object on core that would handle all these things.

```python
# events
@core.mda.events.frameReady.connect
def newFrame(img, event):
    print(img.shape)

# cancelling/pausing
core.mda.cancel()
core.mda.toggle_pause()

# running mda - keep on core for reason below
core.run_mda(seq) # internally calls self.mda.run(seq)
```

This could also be a way towards having pluggable acquisition engines. By deafult we would load the current loop, but you could register your own engine with `core.register_mda_engine`

tlambert03, 2022-03-03:
> What about having an mda object on core that would handle all these things.

love it

ianhi, 2022-03-04:
fixed in #102

closed 2022-03-04 by ianhi