pymmcore-plus/pymmcore-plus #117: MDA: Add `Keep Shutter Open` Option to default engine

ianhi, 2022-03-22:
In micromanager there is the option to keep the shutter open for a Z stack. The default MDA engine should support this option

x-ref: https://github.com/tlambert03/useq-schema/issues/38

ianhi, 2022-03-22:
This may require more tracking of the state of the mda (or something new in useq) to be able to know when we've switched to a new channel. That's actually probably a generally useful thing in useq, to have some way of denoting on an event that it is the first or last event in a given run of time, or position, or channel, or Z.

tlambert03, 2023-07-27:
note that https://github.com/pymmcore-plus/useq-schema/pull/107 added `MDAEvent.keep_shutter_open` ... Now pymmcore-plus just needs to implement it

PR #237 (2023-07-31): feat: implement keep shutter open for engine
closes #117 .  implements support for leaving the shutter open if `MDAEvent.keep_shutter_open` is True
closed 2023-07-31 by tlambert03