pymmcore-plus/pymmcore-plus #119: PyQt import errors

ianhi, 2022-03-29:
Maybe something weird with the napari imports rather than an issue here - but running into this error:

```python
from napari.layers import Points
from pymmcore_plus import CMMCorePlus
core = CMMCorePlus.instance()
```

gives
```
AttributeError                            Traceback (most recent call last)
Input In [1], in <cell line: 3>()
      1 from napari.layers import Points
      2 from pymmcore_plus import CMMCorePlus
----> 3 core = CMMCorePlus.instance()

File ~/mambaforge/envs/pymm/lib/python3.10/site-packages/pymmcore_plus/core/_mmcore_plus.py:84, in CMMCorePlus.instance(cls, mm_path, adapter_paths)
     82 global _instance
     83 if _instance is None:
---> 84     _instance = cls(mm_path, adapter_paths)
     85 return _instance

File ~/mambaforge/envs/pymm/lib/python3.10/site-packages/pymmcore_plus/core/_mmcore_plus.py:96, in CMMCorePlus.__init__(self, mm_path, adapter_paths)
     93 if adapter_paths:
     94     self.setDeviceAdapterSearchPaths(adapter_paths)
---> 96 self.events = _get_auto_core_callback_class()()
     97 self._callback_relay = MMCallbackRelay(self.events)
     98 self.registerCallback(self._callback_relay)

File ~/mambaforge/envs/pymm/lib/python3.10/site-packages/pymmcore_plus/core/events/__init__.py:23, in _get_auto_core_callback_class(default)
     22 def _get_auto_core_callback_class(default=CMMCoreSignaler):
---> 23     if _qt_app_is_running():
     24         from ._qsignals import QCoreSignaler
     26         return QCoreSignaler

File ~/mambaforge/envs/pymm/lib/python3.10/site-packages/pymmcore_plus/_util.py:58, in _qt_app_is_running()
     56 for modname in {"PyQt5", "PySide2", "PyQt6", "PySide6"}:
     57     if qmodule := sys.modules.get(modname):
---> 58         QtWidgets = getattr(qmodule, "QtWidgets")
     59         return QtWidgets.QApplication.instance() is not None
     60 return False

AttributeError: module 'PyQt5' has no attribute 'QtWidgets'

```

tlambert03, 2022-03-29:
i suspect you have multiple backends installed,  can you print output of pip list?

ianhi, 2022-03-29:
Interestingly I don't seem to:

```
(pymm) ian@fermented-beans:~/Documents/oss/micro/raman-mda-engine$ mamba list | grep -E 'pyside|PyQt|pyqt'
pyqt5                     5.15.6                   pypi_0    pypi
pyqt5-qt5                 5.15.2                   pypi_0    pypi
pyqt5-sip                 12.9.1                   pypi_0    pypi
(pymm) ian@fermented-beans:~/Documents/oss/micro/raman-mda-engine$ pip list | grep -E 'pyside|PyQt|pyqt'
PyQt5                         5.15.6
PyQt5-Qt5                     5.15.2
PyQt5-sip                     12.9.1
```

note that I can import napari, but if I import the points layer then things break.

tlambert03, 2022-03-29:
still suspect environment.  try a clean env and let me know if it still happens

ianhi, 2022-03-29:
same in a fresh environment

```
mamba create -n test python=3.9
conda activate test
pip install pymmcore-plus PyQt5 napari 
python -c  "from napari.layers import Points;from pymmcore_plus import CMMCorePlus;core = CMMCorePlus.instance()"
```

tlambert03, 2022-03-29:
well!  :joy:  ... ok, will take a look

tlambert03, 2022-03-29:
ah, ok, sorry: yeah this is a latent bug in `_qt_app_is_running`.  The problem is that the top level `PyQt5` doesn't actually import `QtWidgets` (i.e., it has no _attribute_ `QtWidget`) ... so the only reason this code has generally worked before is that someone else had presumably already imported `PyQt5.QtWidgets` (like all the `_qt` code in napari when you instantiate a viewer).  but when you import `napari.layers` (but _don't_ create a Viewer) it looks like something has imported the top level PyQt5 module, such that it's in `sys.modules` ... but `PyQt5.QtWidgets` isn't there...  So, we should probably check that `PyQt5.QtWidgets`, etc is in sys.modules, or import it explicitly after detecting that PyQt5 is there.

closed 2022-04-23 by tlambert03