pymmcore-plus/pymmcore-plus #134: Registry system for `MDAEngines` and `MDAWriters`

ianhi, 2022-07-09:
## Problem

Currently there is no way for a GUI to know what `MDAEngine`s (and in the future `MDAWriters` are installed. This makes it tricky for napari-micromanager to be able to offer a list of writers or engines to be selected by the user.

## Potential solution

`pymmcore-plus` should provide a way for other code to get a list of all valid  `MDAEngine`s (or writers) that are available in the current environment. Something like:

```python
def get_available_engines() -> list["some way to import the engine"]
```

I think that entry_points are the right way to accomplish this?

tlambert03, 2022-07-10:
How many MDA engines do we imagine existing?  An entry point certainly would be one way to accomplish this, but i feel like this was largely added as a hook specifically for pycromanager.  Are we expected this to be something that lots of people want to hook into?  If so, I think whether that's a good idea to begin with is worth discussion.  At first blush, I'd prefer a more configurable `run_mda` method, before leaping straight to plugins (which can quickly disperse useful functionality across an ecosystem without a huge amount of benefit)

ianhi, 2022-07-19:
>  An entry point certainly would be one way to accomplish this, but i feel like this was largely added as a hook specifically for pycromanager

Oh interesting. I actually considered the primary value of the swappable engines to be supporting extensions on top of the default mda-engine. For example when you want to implement custom behaviors like software autofocus, or in my case controlling non-MM devices. Right now I launch everything from a script so I can set this up, but we're trending towards having less technical users use [raman-mda-engine](https://github.com/ianhi/raman-mda-engine) so in an ideal world  they wouldn't have to open up a python interpreter but could just launch napari and be able to select our custom engine (or writers) from there.

That said configuring engines is a bit niche and I was actually writing this mostly with an eye towards a future `MDAWriter`. We should provide sensible default writers here, but at some point someone will also create custom writers for their own usecases and it would be nice if napari-micro had a way to discover them and offer to select them in a gui element.

tlambert03, 2022-07-19:
what I don't love about fully swapping out the engine with a single run method is that it puts way too much on the engine.  For example, your raman engine also has to emit events in just the right places or it will not truly be "swappable".  In other words `run_mda` is too complex of a protocol to make for a good plugin.  If it was just two engines (us and perhaps pycromanager) then it's easy to communicate and test... but entry points open it up to anyone.

If we want this to be truly pluggable, I think we need to tighten it up significantly, where the plugin just has to do things like `prep_hardware`, or `acquire`, or, indeed, `write_image`.

closed 2025-06-09 by tlambert03