pymmcore-plus/pymmcore-plus #144: should `events.imageSnapped` be emitted on `snapImage()` rather than `snap()`

tlambert03, 2022-09-22:
currently, the `imageSnapped` event _only_ gets emitted if someone use the `snap()` method that is introduced by pymmcore-plus ... but not if they use the `snapImage()` method that they're probably accustomed to using.  This makes it less useful for saying "an image is ready to be viewed", as it now depends very much on how the end-user is using pymmcore-plus.

I think we should perhaps change the `imageSnapped` event to be emitted by the `snapImage()` method, and have it emit nothing, rather than only upon using our custom `snap()` method.

then it would be:
```
@core.def.imageSnapped.connect
def _handle():
    core.getImage()
```

thoughts @ianhi @fdrgsp ?

fdrgsp, 2022-09-22:
@tlambert03 I agree!

ianhi, 2022-09-23:
In genera yes I agree However, thinking back I think I did it this way so that way because I was worried about the MDAEngine using `snapImage` and thus causing potential downstream issues

wl-stepp, 2023-11-03:
I'm having an additional thought on this. What do you guys think about a SnapEngine? We have to trigger the camera and some components for every snap. This means that snapImage does not return unless we check somewhere that it has been called and do the triggering. If we would have a SnapEngine that get's called whenever we want to snap, we could put that logic in there and have it automatically for everywhere you might want to snap (SnapButton, StageWidget etc).
One step further, the MDAEngine would also call the SnapEngine, so we can do setup for one event in the SnapEngine and sequence setups in the MDAEngine. the SnapEninge has a default that works as it is now and a different SnapEngine can be registered like the MDAEngine is.

tlambert03, 2023-11-03:
Definitely open to it!  
I do think that the moment of acquisition is one of the most critical parts, and honestly, the part that someone might *not* want to use micro-manager for (for example to use [acquire](https://github.com/acquire-project/acquire-python) or to run the camera as a slave rather than master).  So, having a nice abstraction around a modular frame grabber would be great.  I'm going to move this to a new issue since I think the `imageSnapped` emission is a parallel discussion.

PR #314 (2024-02-02): refactor: move imageSnapped event
this closes #144 by moving the `imageSnapped()` event from the `CMMCorePlus.snap()` method (which is new and unexpected for most people) to the `snapImage()` method (which is both more familiar and semantically correct).

This change is partially motivated by the confusion caused by this in https://github.com/pymmcore-plus/napari-micromanager/issues/305

Some considerations before merge:
1. since `snapImage()` no longer has access to the resulting image, it removes the numpy array as an argument to the callback, which is a breaking change.  Callbacks must now  [... cut]
closed 2024-02-02 by tlambert03