pymmcore-plus/pymmcore-plus #210: should `MDAEngine` be calling `getImage()`?

tlambert03, 2023-07-19:
While working on #115, it occurred to me that we might not want to have the engine actually call `getImage` in `EventPayload(image=self._mmc.getImage())`.  

One can imagine downstream applications where you might want to directly access that buffer somehow,  and this would have incurred an additional read event.  This becomes even more obvious with sequence events, where we would be reading a whole sequence of events into RAM.

So, i think we should think about whether it should be the responsibility of something else (either a viewer or a writer) to actually call `getImage()`

ianhi, 2023-07-25:
> So, i think we should think about whether it should be the responsibility of something else (either a viewer or a writer) to actually call getImage()

Is there a worry about synchronizing timing here if both the acquisition and writer are running on different threads? i.e. would `getImage` be guaranteed to still be current by the time a writer/napari-micro calls it?

tlambert03, 2023-07-25:
> Is there a worry about synchronizing timing here if both the acquisition and writer are running on different threads? i.e. would getImage be guaranteed to still be current by the time a writer/napari-micro calls it?

yep, that should definitely be a concern, and something that we should have an answer for before changing anything

tlambert03, 2025-06-09:
i'm not so concerned about this any more