pymmcore-plus/pymmcore-plus #300: mmcore install specific release

wl-stepp, 2023-12-18:
Hello everybody,

I'm having problems using the --release input for mmcore install:
[image]

Not sure what the release should look like and the error message seems to be truncated. 

[image]

wl-stepp, 2023-12-18:
I also think this doesn't really work. There is no "_win" in the folder name for me. Setting environment variable works however, so not urgent.

https://github.com/pymmcore-plus/pymmcore-plus/blob/78585577b50d12d30c9ae33c53e4fe447df651e6/src/pymmcore_plus/_util.py#L107C1-L116C58
```python
    # then look for an installation in this folder (from `pymmcore_plus.install`)
    sfx = "_win" if os.name == "nt" else "_mac"
    local_install = [
        p for p in PYMMCORE_PLUS_PATH.glob(f"**/Micro-Manager*{sfx}") if p.is_dir()
    ]
    if local_install:
        if return_first:
            logger.debug("using MM path from local install: %s", local_install[0])
            return str(local_install[0])
        full_list.extend([str(x) for x in local_install])
```

tlambert03, 2023-12-18:
That second if clause is more or less obsolete at this point (it used to install directly into the pymmcore plus folder, and made a _win folder at that time.  Now it installs into proper user space.  Sorry about the useless error message, will check later today and get back to you.  You should also be able to manually download and install it into the user folder, and if you use `mmcore list` (and if you have at least one other install) it will show you that folder.  I also think it’s in the readme or docs somewhere.

tlambert03, 2023-12-18:
ok, the error above is a windows specific issue with the `available_versions` parsing here:

https://github.com/pymmcore-plus/pymmcore-plus/blob/78585577b50d12d30c9ae33c53e4fe447df651e6/src/pymmcore_plus/install.py#L153

will try to post a fix later today

tlambert03, 2023-12-18:
fixed by #301 ... but one note @wl-stepp, if you have multiple versions installed, there still isn't a good way to tell mmcore plus which one to select (short of using the environment variable).  We need to save a current selection somewhere, and add something like a `mmcore use` command to select

PR #301 (2023-12-18): fix: fix available version lookup on windows
name parsing needed a fix in delimiter for windows

fixes #300
closed 2023-12-18 by tlambert03