pymmcore-plus/pymmcore-plus #303: OME Zarr writer doesn't fail gracefully with GeneratorMDASequence

tlambert03, 2023-12-30:
the Zarr writer is supposed to fail gracefully if an `Iterable[MDAEvent]` is used instead of MDASequence, but doesn't currently because:

1. the error will likely happen on every frame (not just once)
2. it doesn't properly handle the "non-sequence" proxy object `GeneratorMDASequence`

this is as much of an issue in the (non)-design of `GeneratorMDASequence` as it is in the zarr writer.  but something that should be improved (preferably without having to check `isinstance(thing, GeneratorMDASequence)`

jeskesen, 2025-04-25:
Hi @tlambert03 ,

> the Zarr writer is supposed to fail gracefully if an Iterable[MDAEvent] is used instead of MDASequence

I've run into this recently (though not specifically for Zarr), does this mean that one cannot use the supplied file writers using an Iterable[MDAEvent]?

tlambert03, 2025-04-25:
(side note: since you work on acquire... I do have some experiments refactoring all this code to try to be able to swap in acquire-zarr for tensorstore, borrowing the general strategy of defining a dimension schema from acquire-zarr, and then just appending files as they come off the scope.  It's a much nicer API of course, and I'd like to have something here like that too.  but that code is still a local experiment)

jeskesen, 2025-04-25:
> I think the best we can do for Iterable[MDAEvent] is to offer to just save every frame as it comes off the scope in a (t, y, x) arrangement.
The approach acquire takes is to assume data is coming in in the correct order and then append it as t,c,z,y,x or whatever subset is relevant.  We're not currently handling position, but that is actively being worked on.  In the pymmcore-plus context, do you think its reasonable to accumulate data (if acquired out of order) and flush it when the buffer is filled?

> I do have some experiments refactoring all this code to try to be able to swap in acquire-zarr for tensorstore

Is it something I could help with?  I was going to take a crack at that myself, so maybe its better if we're working together?

tlambert03, 2025-04-25:
> The approach acquire takes is to assume data is coming in the correct order and then append it as t,c,z or whatever subset is relevant. 

right... but with `Iterable[MDAEvent]` we potentially have no information at all about what is about to happen.  An iterable could be a generator, or even a `Queue` object that doesn't actually start with any events (i.e. a fully dynamic acquisition based on the feedback from image analysis or something).  We can make no assumptions ahead of time about what the correct or final order even is.   It could be a completely jagged array with a z stack, followed by a little burst of time lapse for a different channel, followed by ... whatever.

> In the pymmcore-plus context, do you think its reasonable to accumulate data (if acquired out of order) and flush the entire volume, volume+channel, etc, when the buffer is filled?

in the case of a pre-defined experiment, absolutely... but that's not what `Iterable[MDAEvent]` implies.  It's the broader type (i.e. not an `MDASequence`) that could be anything, and we can't always even peek ahead to look at what's coming.  That's really the key problem here.

if you *do*, however, have a nice, standard non-jagged XYZCTP sequence that you're trying to run, may I ask what structure you're actually passing in to `.run()`?  Can you not use `MDASequence`?  it's ok if not... that would be a good feature for us to support (i.e. infer the dimensions from a *`Sized`* sequence).

> Is it something I could help with? I was going to take a crack at that myself, so maybe its better if we're working together?

I would absolutely like some help one this!  👍 
The general idea I was going to go with is

1. define a function that accepts the `Iterable[MDAEvent]` (and *all* of it's possible subtypes, including `MDASequence`) and return a list of `DimensionInfo` objects (same concept as the `Dimension` object in acquire), if possible.
2. If it *is* possible, i.e. it's a standard ND array, then we're on the happy path, and then we could just create an abstract base class that has a few standard methods similar to `aqz.ZarrStream`, most notably an `append` method (i do think that's the correct API.  And that `.append` method is more or less all that pymmcore-plus really would have to care about (with some additional logic for saving metadata)
4. If it's not possible, we fallback to some custom 3d TYX structure... 

I'll try to find the code I was working on, but I have to step out now (and it wasn't much more elaborate than what I wrote above).

tlambert03, 2025-04-26:
this is where I was starting: https://github.com/tlambert03/ome-writers

basic example at https://github.com/tlambert03/ome-writers/blob/main/x.py
ABC at https://github.com/tlambert03/ome-writers/blob/main/src/ome_writers/_stream_base.py
aqz implementation at https://github.com/tlambert03/ome-writers/blob/main/src/ome_writers/_acquire_zarr.py

jeskesen, 2025-04-26:
Thank you,
  First off, Apologies for hijacking an issue that wasn't meant for this problem.  I'd only intended to ask about the writer failing by design in the Iterable[MDAEvent] case, but I really appreciate the conversation. 

  Some context: In the short term, I'm attempting to [port an existing repo from pycromanager to pymmcore-plus](https://github.com/czbiohub-sf/shrimPy/pull/171) as part of an inter-group collaboration.   Many of the data structures seemed to have good 1-to-1 analogs, so it was going smoothly... until I tried to start writing to files where many of the assumptions broke down.  

> right... but with Iterable[MDAEvent] we potentially have no information at all about what is about to happen. An iterable could be a generator, or even a Queue object that doesn't actually start [...]

Yeah, I can see how that could be a real tricky general problem to solve.  In our case, we can constrain the assumptions to an in-order, append-only acquisition.  From your description & a quick glance, it looks like your new ome-writers repo could be exactly what we need.  One thing that the existing repo does (as I've ported it) is create a new MDASequence for each CZXY block on the fly as it externally loops over time & position where they handle auto-focus and auto-exposure independently.  I suspect I could do this in a more integrated way in pymmcore-plus, however I'm trying to be as non-invasive to the acquisition loop as I can.

tlambert03, 2025-04-26:
> First off, Apologies for hijacking an issue that wasn't meant for this problem.

no worries! the general topic of writers is something that very much needs more eyes.   While we do have some existing logic... I'm by no means content with it, and I'd be more than happy to essentially throw it all away as we refine the patterns.  So, I'm happy to have you looking at it (particularly with your acquire-zarr experience).  

> I'm attempting to https://github.com/czbiohub-sf/shrimPy/pull/171 as part of an inter-group collaboration.

thanks for the context!  That's great, happy to see it. do feel free to reach out in general on that if you have any questions as you go :)

> From your description & a quick glance, it looks like your new ome-writers repo could be exactly what we need.

great.  Yes, I'd love to essentially see all of the data handlers ultimately replaced with something like that.  Any involvement you'd like to have would be welcome.  That little repo I started was just a playground, happy to move it anywhere (either within pymmcore-plus, or as an independent well-scoped repo) if entices you to work on it! :)

> I suspect I could do this in a more integrated way in pymmcore-plus, however I'm trying to be as non-invasive to the acquisition loop as I can.

Totally makes sense.  Your code looks like a nice bare minimum example.  I agree, stuff [like this](https://github.com/czbiohub-sf/shrimPy/blob/4f28d308ceee0b94f21ab285e27f734d8014e066/mantis/acquisition/acq_engine.py#L837-L856) could eventually be made a bit simpler, and you might find that a [custom MDAEngine](https://pymmcore-plus.github.io/pymmcore-plus/guides/custom_engine) suits some of your needs here.  And you can [tuck non-standard info](https://pymmcore-plus.github.io/pymmcore-plus/guides/custom_engine/#utilizing-mdaevent-metadata) into the `MDASequence` object to inform what your engine should do with it.

Anyway.  good stuff, thanks for dropping by and let me know how I can help

tlambert03, 2025-04-26:
side-note: if you ever get annoyed by having to return everything back from snake case to camel case, I could conceivably add a `__getattr__` convenience to `CMMCorePlus` ..., but it won't benefit from the type hinting (you'd be back to purely dynamic attributes)

jeskesen, 2025-04-28:
> side-note: if you ever get annoyed by having to return everything back from snake case to camel case, I could conceivably add a __getattr__ convenience to CMMCorePlus ..., but it won't benefit from the type hinting (you'd be back to purely dynamic attributes)

Thanks, I appreciate the offer... but I think I'm done with that part, and it wasn't that bad.

tlambert03, 2026-03-03:
i think this is finally/safely closed by https://github.com/pymmcore-plus/pymmcore-plus/pull/579

closed 2026-03-03 by tlambert03