pymmcore-plus/pymmcore-plus #306: repeating hardware triggering causes error in device

tlambert03, 2024-01-24:
from @linshaova:

when I use

```python
# enable hardware triggering
mmc.mda.engine.use_hardware_sequencing = True
```

and then try `mmc.run_mda()` a second time with the same `MDASequence` after the first run is finished, an error would appear complaining about the DA Z Stage (in my case, named "Galvo"):

```pytb
2024-01-24 09:29:10,883 - pymmcore-plus - INFO - (_runner.py:311) MDA Finished: z_plan=ZRangeAround(go_up=True, range=400.0, step=1.0)
Exception in thread Thread-10 (run):
Traceback (most recent call last):
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\threading.py", line 1052, in _bootstrap_inner
    self.run()
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\threading.py", line 989, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\site-packages\pymmcore_plus\mda\_runner.py", line 174, in run
    raise error
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\site-packages\pymmcore_plus\mda\_runner.py", line 168, in run
    self._run(engine, events)
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\site-packages\pymmcore_plus\mda\_runner.py", line 188, in _run
    engine.setup_event(event)
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\site-packages\pymmcore_plus\mda\_engine.py", line 129, in setup_event
    self.setup_sequenced_event(event)
  File "C:\Users\scape20\miniconda3\envs\pycroman\Lib\site-packages\pymmcore_plus\mda\_engine.py", line 273, in setup_sequenced_event
    core.loadStageSequence(zstage, event.z_sequence)
RuntimeError: Error in device "Galvo": (Error message unavailable) (2001)
 ```

tlambert03, 2024-01-24:
@linshaova, it would be great to try to reduce this error to a set of core calls, something like:

```python
import time

def run_sequence():
    zstage = core.getFocusDevice()
    core.loadStageSequence(zstage, [0.0, 1.0, 2.0])
    core.startStageSequence(zstage)
    core.startSequenceAcquisition(3, 0, True)
    while core.isSequenceRunning():
        time.sleep(0.01)
```

and then run that again.  If you get the same error (great!), then play with adding things like `core.stopSequenceAcquisition` between calls

marktsuchida, 2024-01-24:
It would also be useful to know what device is acting as the DA for the DA Z Stage.

tlambert03, 2024-01-24:
Sorry @marktsuchida :) this was pulled from an email conversation with Lin, and is related to this issue on imagesc: 
https://forum.image.sc/t/device-property-usesequence-seen-in-a-pycro-manager-tutorial-unavailable-in-my-trial/91089/5

so I gather the device acting as DA is a national instruments device, I believe lin is now using the NIDAQ device adapter, but may have also been using either NI100x and/or NIMultiAnalog.  @linshaova, can you confirm? 

btw: @marktsuchida, meet @linshaova and vice versa!  Two people I highly respect :)

marktsuchida, 2024-01-24:
Nice to meet you @linshaova.

If it's `NIDAQ`, the error `2001` is `ERR_SEQUENCE_RUNNING`. From the source code it looks like it's returned if you try to change the trigger input port or sample rate before stopping a running sequence. The code means the same thing in `NIMultiAnalog` and is also returned if you try to send a new sequence or set a (non-sequence) voltage before stopping a running sequence.

So I'd try `stopStageSequence()` after every run.

(The `Utilities` virtual devices currently lack a mechanism to transfer the correct error message for error codes of the physical device.)

tlambert03, 2024-01-24:
thanks @marktsuchida!  that was the key i needed to discover this PR from @nicost: https://github.com/micro-manager/AcqEngJ/pull/108 making a similar fix in AcqEngJ.  I'll make a similar fix here

PR #307 (2024-01-24): fix: avoid hardware errors while running sequences with NIDAQ stages
similar to https://github.com/micro-manager/AcqEngJ/pull/108

fixes #306 by calling `stopStageSequence` prior to `loadStageSequence` in the MDAEngine
closed 2024-01-24 by tlambert03