pymmcore-plus/pymmcore-plus #311: Request for opinions on data-saving APIs

tlambert03, 2024-01-26:
This is a call for opinions.  
CC @fdrgsp @ianhi @wl-stepp @kmdouglass @marktsuchida @dpshepherd @nng-thienphu @linshaova
(it's ok if you don't feel that up-to-date on the goings on of pymmcore-plus, it's kind of a high-level question)

With a protocol for data handlers/writers (like `OMEZarrWriter` and `ImageSequenceWriter`) in place, we need a higher level API for indicating to the runner how it should save data.  Currently, the MDA runner just "emits" frames as they are acquired and it's up to the end-user to attach any specific data handlers they want; this would aim to add a convenience layer on top of that.

## API for Specifying a Writer

There is already the low-level:

```python
writer = OMEZarrWriter("file.zarr")
with mda_listeners_connected(writer):
    core.mda.run(sequence)
``` 

but I was always hoping to offer a slightly nicer API than that.  Here are a few options (and note that all of these are mutually compatible with a user *also* attaching additional listeners using `mda_listeners_connected`)

### 1. Add an `output` field to `MDARunner.run`

(can also be named something other than `output`)

```python
core = CMMCorePlus()
core.mda.run(..., output='some/path.ome.zarr')  # pass a string
# or 
my_writer = CustomDataHandler()
core.mda.run(..., output= my_writer). # pass a data handler object 
```

The argument would be either:
- `str` - in which case a builtin Handler would be used based on extension.  Note that fsspec can be used to support remote filesystems.
    - `ome.zarr` - > `OMEZarrWriter`
    - `ome.tiff` -> `OMETiffWriter`
    - `dir` -> no extension would use `ImageSequenceWriter`
- A Handler object, (anything implementing the data handler protocol, meaning it has at least a `frameReady` method that accepts 1-3 arguments).  This could be, for example, something that would deskew and rotate images prior to writing them for @dpshepherd

### 2. Add an `output` field to `useq.MDASequence`

A similar thing could be added at a higher level to useq-schema's `MDASequence` object

```python
core = CMMCorePlus()
seq = MDASequence(output='some/path.ome.zarr', ...)
core.mda.run(seq)
```

The type semantics of `str` would be the same.  But since `useq-schema` should remain agnostic to the APIs of `pymmcore-plus`, it's harder to say exactly how a custom writer should be declared there in a way that retains `useq-schemas` agnosticism to micro-manager, imswitch, etc...

Also, as it requires an `MDASequence` object (which might not always be created in the cases of event-driven or non-deterministic sequences like what @wl-stepp and @kmdouglass might use) it might need to be done *in addition* to explicitly allowing `output` to be passed to the `run()` method.  That is, `run()` will first look to see if an `output` argument has been passed, and then fallback to an `MDASequence.output` field if it has been passed.

### 3. Give `MDARunner` a new `writer` or `handler` attribute that will always be used:

```python
core = CMMCorePlus()
my_writer = CustomDataHandler()
core.mda.set_handler(my_writer). # or `set_writer` or something
core.mda.run(...)
```

Note that this is rather similar to just using:

```python
my_writer = CustomDataHandler()
with mda_listeners_connected(writer):
    core.mda.run(...)
```

except that it permanently attaches `my_writer` to the runner, and you could call `run()` multiple times.
It doesn't prevent a user from attaching additional handlers if desired, it just essentially acknowledges that there is a usually a "special" handler that will be responsible for writing.

## File name incrementing

In all cases, we have to determine how the file name will be incremented if called twice.  Any ideas welcome here.

1. It could be left entirely to the Handler: each handler would be responsible for checking if the destination exists, and changing the name or raising accordingly.  This gives the most flexibility, but makes writing a handler significantly more cumbersome.
2. Add a new argument somewhere, perhaps `run(..., output='some_path', increment=some_function)`: where `some_function` is a `Callable[[str], str]]` - a function that takes the requested string and returns a new useable string (or raises).  This could have a default implementation that adds `_001` for example to the filename.
3. Just have pymmcore-plus magically increment filenames that already exist.

tlambert03, 2024-01-26:
fwiw, @fdrgsp and I think we favor option number 1

marktsuchida, 2024-01-26:
I like option 1, too. Option 3 adds (effectively) global state. And I also think data destination should be kept separate from useq.

dpshepherd, 2024-01-26:
I also favor option 1. I am curious if it makes sense to allow multiple outputs. Maybe too complicated?

For example, the API could:
- allow multiple outputs, to give parallel processing and writing.
- enforce one output, and the responsibility for passing the data to the writer is on the function that receives it. 

Not sure what is the best practice given the parallelism structure you have in mind.

tlambert03, 2024-01-26:
>  I am curious if it makes sense to allow multiple outputs. Maybe too complicated?

yep, I can see that being a reasonable feature request as well:

here's a possibility.  It could take any of `Path | str | None | object | Sequence[str | object]`.
I don't think there's any overlapping semantics in there.  It's flexible in what it allows, but pretty clear what happens in each case?

```python
class MDARunner:

    def run(
        self,
        events: Iterable[MDAEvent],
        output: Path | str | None | object | Sequence[str | object] = None,
    ) -> None:
        """Run the multi-dimensional acquistion defined by `events`.

        Parameters
        ----------
        events : Iterable[MDAEvent]
            An iterable of `useq.MDAEvents` objects to execute.
        output : str | None | object | Sequence[str | object], optional
            If a path or string, the path to save the output to (A writer will be
            selected automatically based on the file extension).
            If a writer object, the writer to use to save the output.  Must implement
            the `DataHandler` protocol (i.e. something that has frameReady, etc...).
            May also be a sequence of paths/writers, in which case the output will be
            saved to all of them.
            If None, no output will be saved.
        """
```

as for the exact mechanism of the parallelized writing, the ThreadRelay mentioned in https://github.com/pymmcore-plus/pymmcore-plus/issues/289#issuecomment-1837209136 should handle some basic cases (keeping the writers from blocking the progress of the experiment), but I'm not sure that it would be maximally efficient.  That is: writing might take longer than needed to "catch up" to after the experiment is finished.

wl-stepp, 2024-01-27:
Yes, 1 also sounds the best to me. I like how 2 doesn't have to touch the runner itself, but if it's to stay agnostic the only place would be a special key in the metadata field, right?

I think that one thing to consider for the filename adjustment is, that it should be communicated back up to the GUI. Would that be an event? Otherwise I think it might better be handled upstream. I have done incrementation on sequenceFinished from the widget that defines the path in the widget.

PR #313 (2024-02-15): feat: Add file saving `output` argument to MDARunner.run
This implements option number 1 and closes #311 

### 1. Add an `output` field to `MDARunner.run`

```python
core = CMMCorePlus()
core.mda.run(..., output='some/path.ome.zarr')  # pass a string
# or 
my_writer = CustomDataHandler()
core.mda.run(..., output= my_writer). # pass a data handler object 
```

The argument would be either:
- `str` - in which case a builtin Handler would be used based on extension.  Note that fsspec can be used to support remote filesystems.
    - `ome.zarr` - > `OMEZarrWriter`
    - `ome.tiff` -> `OMETiffWriter [... cut]
closed 2024-02-15 by tlambert03