pymmcore-plus/pymmcore-plus #318: towards using external trigger as master

linshaova, 2024-02-01:
Hi there,

I played with [NI-DAQmx Python package](https://nidaqmx-python.readthedocs.io/en/latest/) running side-by-side with Micro-manager. It looks promising in that, first of all, MM and the Python process can have independent control over the same NIDAQ device, as long as the controls are for different and non-conflicting resources on the device. One test I did is that I configure the MDA to run with hardware-sequenced focus stage while at the same time, I could run a pulse-train generation sequence via `nidaqmx` on the side, and oscilloscope traces confirms that both proceed correctly.

The reason I brought this up is that with `nidaqmx` being easily integratable with pymmcore-plus, we can start to think about using external triggers to sync the MDA, as opposed to using the camera's exposure-start trigger. For this to happen though, the `MDAEngine` needs another method similar to `setup_sequence()` that gets called _right after_ the externally-triggered camera starts its acquisition. In this method, we could for example start the pulse-train generator with some `nidaqmx` calls. How does that sound? Or does such a method already exist?

Cheers,
-lin

dpshepherd, 2024-02-01:
Hi all-

Our current strategy is to sub-class the acquisition engine with custom events. One is for waveform setup, one to start the camera in external mode, and then finally have the `MDAEngine` start. This way, we explicitly handle starting the camera.

In the end, we are trying to port our custom code from [generating waveforms on an NIDAQ for external master](https://github.com/QI2lab/mcSIM/blob/a645d4ed0a9949464d60ae0d5b0645ea44cb1a82/mcsim/expt_ctrl/program_sim_odt.py) and then [running an acquisition](https://github.com/QI2lab/napari-micromanager/blob/6117a2442d9a2a150740a1cf75febc0a0e0e90df/micromanager_gui/sim_odt_widget.py#L470) into a sub-class custom `MDAEngine` that re-uses all of the great work in this project so that we aren't writing our own acquisition engine.

linshaova, 2024-02-01:
Doug's comments probably suggest all necessary tools are already in place. I just have to learn more about the custom events.

One question regarding the steps Doug described: when do you start the waveform on the NIDAQ side with the `MEAEngine` start? because I imagine the correct way of implementing this sequence of events is first starting the camera in external-trigger mode, letting it wait for the trigger train, and sending the waveform from NIDAQ. which is exactly what my original question is about.

To answer @tlambert03's "naive" question: I naively think in the `MDAEngine` somewhere it'll call something like `cameraStartAcquisition()`; is that a somewhat correct understanding? If so, then this new method I imagined would be called right after this. 

-lin

tlambert03, 2024-02-01:
> To answer @tlambert03's "naive" question: I naively think in the MDAEngine somewhere it'll call something like cameraStartAcquisition(); is that a somewhat correct understanding? If so, then this new method I imagined would be called right after this.

:joy: so much naïveté going on here huh? :joy:
For a sequenced acquisition, the MDAEngine starts the sequence, and gives up control, right here:

https://github.com/pymmcore-plus/pymmcore-plus/blob/788ec247aab3efe213fd631b3aea1078ab45272a/src/pymmcore_plus/mda/_engine.py#L335-L343

and, if you'd like a lower level glimpse of exactly how that is controlling the various device adapters, see the C++ code for `startSequenceAcquisition` here:

https://github.com/micro-manager/mmCoreAndDevices/blob/1aa373bd769b312940d7336c8c7c54943b93340b/MMCore/MMCore.cpp#L2660

I think the reason I called it naive is that it's unclear to me in this flipped regime where the camera is not the master, whether/how micro-manager's `startSequenceAcquisition` is involved _at all_?  that is, if you look at the C++ code, you can see that a call to `MMCore.startSequenceAcquisition` just turns around and calls `camera->StartSequenceAcquisition` on the current camera device.  And, what's not entirely clear to me is whether that will still be the case when you make the camera the slave?  can you clarify that for me?

tlambert03, 2024-02-01:
can you guys have a look at https://github.com/pymmcore-plus/pymmcore-plus/pull/319 and see if it looks like it would be all you need?

dpshepherd, 2024-02-01:
Yes, if you call `startSequenceAcquisition(n_pictures_expected,0,True)`, it will block until all of the requested images arrive if it is setup for external triggering. Then, we start the DAQ waveform playback.

See [here](https://github.com/QI2lab/napari-micromanager/blob/6117a2442d9a2a150740a1cf75febc0a0e0e90df/micromanager_gui/sim_odt_widget.py#L1321) for that logic.

We have had to do some work to deal with XY stage triggering using pause loops on the DAQ, but this way you can run fully automated with fast timing.

dpshepherd, 2024-02-01:
> can you guys have a look at https://github.com/pymmcore-plus/pymmcore-plus/pull/319 and see if it looks like it would be all you need?

With a quick glance, that is a yes from me. 

The main hurdle for us now is how to setup the GUI (a DAQ waveform widget for example) and event handling correctly in a custom engine to take advantage of all of the functionality.

tlambert03, 2024-02-01:
great thanks. 

>The main hurdle for us now is how to setup the GUI (a DAQ waveform widget for example) and event handling correctly in a custom engine to take advantage of all of the functionality.

yeah, happy to help with that!  I see the [daq-python](https://github.com/nng-thienphu/daq-python) repo that @nng-thienphu started. that's exciting.   As for the event handling, i think you'll just want to tuck some metadata into the `MDASequence.metadata` or the `MDAEvent.metadata` field that you could look for in that `post_sequence_started` method (which will receive the full `MDAEvent` that triggered it), and use it to setup and trigger your waveforms.  Happy to discuss structural strategies on that over zoom if you want to sometime.

linshaova, 2024-02-01:
> Yes, if you call `startSequenceAcquisition(n_pictures_expected,0,True)`, it will block until all of the requested images arrive if it is setup for external triggering. Then, we start the DAQ waveform playback.

> See [here](https://github.com/QI2lab/napari-micromanager/blob/6117a2442d9a2a150740a1cf75febc0a0e0e90df/micromanager_gui/sim_odt_widget.py#L1321) for that logic.

I have hard time understanding how this logic works exactly because  `startSequenceAcquisition` blocks in the case of external triggering. Wouldn't it block there forever and never get a chance to run `self.daq.start_sequence()`?

dpshepherd, 2024-02-01:
> I have hard time understanding how this logic works exactly because `startSequenceAcquisition` blocks in the case of external triggering. Wouldn't it block there forever and never get a chance to [...]

I just cleaned up that post - realized I said slightly the wrong thing! It doesn't block execution, it just waits for images to arrive.

tlambert03, 2024-02-01:
To be fair, I incorrectly used the work "block" first :) I was thinking from the perspective of the camera, and not from the running process.  But I think we're all clear now.  Thanks!

linshaova, 2024-02-01:
All good! And yes, #319 looks like exactly what I was pitching!

dpshepherd, 2024-02-01:
@linshaova - given we are both interested in this, what sort of waveforms do you need the DAQ to produce? 

What would you need in a GUI for a user to setup these waveforms? We are trying to build something flexible but useful for NIDAQ use patterned after the different widgets that are in [pymmcore-widgets](https://github.com/pymmcore-plus/pymmcore-widgets).

linshaova, 2024-02-01:
My waveforms are scanning light-sheet related (one counter-output pulse sequence, a few AO sequences, and sometimes a DO sequence to control fluorescence channels) and relatively simple. I haven't thought about how a GUI would benefit this; maybe because in my case flexibility is not essential?

dpshepherd, 2024-02-02:
Got it. We are trying to build one DAQ widget for a few different microscopy approaches and we have some collaborators who prefer GUIs, so that is our motivation for the flexibility.

tlambert03, 2024-02-02:
gonna close this with https://github.com/pymmcore-plus/pymmcore-plus/pull/319 merged.  @linshaova, if you find additional considerations that should be discussed in the same thread, feel free to reopen.

closed 2024-02-02 by tlambert03

linshaova, 2024-06-21:
Sorry for the belated response. Long story short: only recently it became possible for me to do a field test of this added method `post_sequence_started` of `MDAEngine` and from what I can tell, it is actually unnecessary and if I used it, it would report some undecipherable NIDAQ error messages (possibly related to it being called in a separate thread but I'm not sure). It turns out if I just define a NIDAQ task and start it after `mmc.run_mda(mda_sequence)` in the main thread (i.e., without involving `post_sequence_started`), external triggers already work beautifully with hardware sequencing, probably thanks to the extensive use of `yield` in `MDAEngine`.

So in conclusion, I'm okay with taking this method out from `MDAEngine` interface.