pymmcore-plus/pymmcore-plus #350: discussion: metadata schema during MDA

tlambert03, 2024-05-14:
I'd like to begin nailing down a schema for metadata emitted during multi-dim acquisitions that consumers can rely on.  That is, I want to clarify the structure of the `meta` objects passed as the second argument to [`sequenceStarted`](https://pymmcore-plus.github.io/pymmcore-plus/api/events/#pymmcore_plus.mda.events._protocol.PMDASignaler.sequenceStarted) and the third argument to [`frameReady`](https://pymmcore-plus.github.io/pymmcore-plus/api/events/#pymmcore_plus.mda.events._protocol.PMDASignaler.frameReady).  This is tricky because some people will certainly want as much data as possible (which takes time) while others will want speed and minimal data on each frame.

These `meta` objects are currently dicts (but see below), and my plan is to:
1. have a `format` and `version` keys at the top level that define the structure of the metadata objects.  Versions will be X.Y, where `X` implies a breaking change (removal/renaming of a key or change in the type), while `Y` implies a backwards compatible change (addition of a non-required key).
2. allow users to set the `format` and `version` in the engine.  Something like `core.mda.engine.set_summary_metadata_format(format='full', version='1.0')`.  This way, if we later decide we need to make changes, any external consumers needn't be affected as long as they specify the format that they want.  (We'll just retain the ability to generate older versions indefinitely)

These formats will be accompanied by documentation that specifies the structure/schema that you can expect.

Some questions:

1. Does that general idea of a metadata format/version string sound good?
2. I'd *like* to depend on [msgspec](https://github.com/jcrist/msgspec), which is a really great high performance library for declaring structured schema, creating objects (faster than dicts), and encoding/decoding to/from json or msgpack (it will also make documenting these objects easier because it outputs to json schema, similar to pydantic).  It has no additional dependencies and has a great author.  Given that the final destination of most of this metadata will be serialization to dict, its [encoding/decoding performance](https://jcristharif.com/msgspec/benchmarks.html) will be nice.  So, first question: **does anyone mind if I take on that additional dependency**?
3. If we do use msgspec, **should the `meta` objects themselves remain `dict`**, with the "actual" metadata struct as the value of some `data` key:
    ```python
    meta = {
      "format": "some-string",
      "version": "1.0",
      "data": SomeStruct(),
    }
    ```
    **or should we relax the typing of `meta` from `dict` to `Any` such that `meta` *itself* could be an object** that has, for example, a fast `to_json` method:
    ```python
    meta = SomeStruct(format='some-string', version='1.0')
    ```
    Note that, in the latter case, if there was enough demand, these could be switchable with:
    ```python
    core.mda.engine.set_summary_metadata_format(format='full-struct', version='1.0')  # meta is a msgspec.Struct
    core.mda.engine.set_summary_metadata_format(format='full-dict', version='1.0')  # meta is a dict
    ````

If you're curious, here is a rough draft of what the summary metadata (at the beginning of the experiment) could look like:

<details>

<summary>Example Summary Metadata</summary>

```python
SummaryMetaV1(
    version='1.0',
    time='2024-05-14T15:46:08.457841',
    devices={
        'DHub': DeviceInfo(type='HubDevice', description='DHub', library='DemoCamera', name='DHub', label='DHub'),
        'Camera': DeviceInfo(type='CameraDevice', description='Demo camera', library='DemoCamera', name='DCam', label='Camera'),
        'Dichroic': DeviceInfo(type='StateDevice', description='Demo filter wheel', library='DemoCamera', name='DWheel', label='Dichroic'),
        'Emission': DeviceInfo(type='StateDevice', description='Demo filter wheel', library='DemoCamera', name='DWheel', label='Emission'),
        'Excitation': DeviceInfo(type='StateDevice', description='Demo filter wheel', library='DemoCamera', name='DWheel', label='Excitation'),
        'Objective': DeviceInfo(type='StateDevice', description='Demo objective turret', library='DemoCamera', name='DObjective', label='Objective'),
        'Z': DeviceInfo(type='StageDevice', description='Demo stage', library='DemoCamera', name='DStage', label='Z'),
        'Path': DeviceInfo(type='StateDevice', description='Demo light path', library='DemoCamera', name='DLightPath', label='Path'),
        'XY': DeviceInfo(type='XYStageDevice', description='Demo XY stage', library='DemoCamera', name='DXYStage', label='XY'),
        'White Light Shutter': DeviceInfo(type='ShutterDevice', description='Demo shutter', library='DemoCamera', name='DShutter', label='White Light Shutter'),
        'Autofocus': DeviceInfo(type='AutoFocusDevice', description='Demo auto focus', library='DemoCamera', name='DAutoFocus', label='Autofocus'),
        'LED': DeviceInfo(type='StateDevice', description='Demo State Device', library='DemoCamera', name='DStateDevice', label='LED'),
        'LED Shutter': DeviceInfo(type='ShutterDevice', description='State device used as a shutter', library='Utilities', name='State Device Shutter', label='LED Shutter'),
        'Camer2': DeviceInfo(type='CameraDevice', description='Demo camera', library='DemoCamera', name='DCam', label='Camer2'),
        'MC': DeviceInfo(type='CameraDevice', description='Combine multiple physical cameras into a single logical camera', library='Utilities', name='Multi Camera', label='MC'),
        'Core': DeviceInfo(type='CoreDevice', description='Core device', library='', name='Core', label='Core')
    },
    properties={
        'DHub': {},
        'Camera': {
            'AllowMultiROI': '0',
            'AsyncPropertyDelayMS': '2000',
            'AsyncPropertyFollower': '',
            'AsyncPropertyLeader': '',
            'Binning': '1',
            'BitDepth': '16',
            'CCDTemperature': '0.0000',
            'CCDTemperature RO': '0.0000',
            'CameraID': 'V1.0',
[... 290 lines cut ...]
                    "type": "array",
                    "minItems": 6,
                    "maxItems": 6,
                    "prefixItems": [{"type": "number"}, {"type": "number"}, {"type": "number"}, {"type": "number"}, {"type": "number"}, {"type": "number"}],
                    "items": false
                }
            },
            "required": ["settings", "pixel_size_um", "pixel_size_affine"]
        },
        "Setting": {"title": "Setting", "type": "object", "properties": {"device": {"type": "string"}, "property": {"type": "string"}, "value": {}}, "required": ["device", "property", "value"]},
        "PositionInfo": {
            "title": "PositionInfo",
            "type": "object",
            "properties": {"x": {"anyOf": [{"type": "number"}, {"type": "null"}]}, "y": {"anyOf": [{"type": "number"}, {"type": "null"}]}, "focus": {"anyOf": [{"type": "number"}, {"type": "null"}]}},
            "required": ["x", "y", "focus"]
        }
    }
}
```

</details>

... while metadata accompanying each frame will likely be *much* more limited.  Probably just the bare minimum timestamp, stage positions, channel and camera info.  (i.e., not the full state of the core as is currently done in MM).

CC @wl-stepp, @kmdouglass, @dpshepherd.  your input welcome, since these are likely objects you would be consuming.

tlambert03, 2024-05-14:
an example of how some of this stuff would be declared can be seen in: https://github.com/pymmcore-plus/pymmcore-plus/pull/351/files#diff-f133fe0432d4cb05d6966987e8c9e99f8dceaae8fb94a19b3d6810a19507e34a

wl-stepp, 2024-05-15:
Sounds good to me.
1. Yes, only predefined formats or would it make sense to enable passing specific keys?
2. Looks very sound to me. But I think most of our applications would work fine with dicts.
3. If the data is the struct anyways, it sounds better to me to have the version and format as well in there. More convenient that way to have them included when writing to disk, which definitely would be a good idea.

There wouldn't be a `core.mda.engine.set_image_metadata_format` then?

tlambert03, 2024-05-15:
> There wouldn't be a core.mda.engine.set_image_metadata_format then?

Yes that would be there too. There would be format for the summary metadata, and a (different) format for each frame metadata

dpshepherd, 2024-05-23:
1. Yes. As long as we can limit the amount for high-speed acquisitions through the set format for each frame.
2. msgspec looks like a good solution.
3. I can see us utilizing both the `dict` and `to_json` method for different applications. My preference would be to start with the higher performance one that writes a structured json to disk. Then add a helper function later to allow for return as a `dict` for those people that need to consuming in real-time.

tlambert03, 2024-06-29:
Ok, lots of work went into #358, which formalizes the schema.  Many thanks to @marktsuchida for all the helpful input.  In the end: I'm not using `msgspec.Struct` by default, but rather leaving metadata as simple `dict`s, with a well-defined schema that is now documented online at:  https://pymmcore-plus.github.io/pymmcore-plus/metadata/

More than happy to continue to make tweaks to it if anyone has further input, but closing this as completed

closed 2024-06-29 by tlambert03