pymmcore-plus/pymmcore-plus #375: Not able to write ome-zarr without msgspec

wl-stepp, 2024-09-23:
Hello everyone,

I just wanted to ask if the tests include a setting without msgspec? I just got a serialization error that I could fix by installing msgspec.
Maybe it should just go into required? This happened with an acquisition, where the grid was activated. I don't have time at the moment unfortunately, but can give a little more detail later if needed.

pymmcore-plus           0.11.1
pydantic                      2.9.2
psygnal                       0.11.1
useq-schema              0.4.7

```
Traceback (most recent call last):
  File "f:\python_env\lib\threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "f:\python_env\lib\site-packages\pymmcore_plus\mda\_thread_relay.py", line 144, in run
    emitter.emit(*args)
  File "f:\python_env\lib\site-packages\psygnal\_signal.py", line 1176, in emit
    self._run_emit_loop(args)
  File "f:\python_env\lib\site-packages\psygnal\_signal.py", line 1213, in _run_emit_loop
    raise loop_err from cb_err  # emit() call ABOVE || callback error BELOW
  File "f:\python_env\lib\site-packages\psygnal\_signal.py", line 1196, in _run_emit_loop
    self._run_emit_loop_inner()
  File "f:\python_env\lib\site-packages\psygnal\_signal.py", line 1225, in _run_emit_loop_immediate
    caller.cb(args)
  File "f:\python_env\lib\site-packages\psygnal\_weak_callback.py", line 453, in cb
    func(obj, *self._args, *args, **self._kwargs)
  File "f:\python_env\lib\site-packages\pymmcore_plus\mda\handlers\_5d_writer_base.py", line 157, in frameReady
    self.position_arrays[key] = ary = self.new_array(key, frame.dtype, sizes)
  File "f:\python_env\lib\site-packages\pymmcore_plus\mda\handlers\_ome_zarr_writer.py", line 279, in new_array
    ary.attrs["useq_MDASequence"] = to_builtins(
  File "f:\python_env\lib\site-packages\zarr\attrs.py", line 89, in __setitem__
    self._write_op(self._setitem_nosync, item, value)
  File "f:\python_env\lib\site-packages\zarr\attrs.py", line 83, in _write_op
    return f(*args, **kwargs)
  File "f:\python_env\lib\site-packages\zarr\attrs.py", line 102, in _setitem_nosync
    self._put_nosync(d)
  File "f:\python_env\lib\site-packages\zarr\attrs.py", line 150, in _put_nosync
    self.store[self.key] = json_dumps(d)
  File "f:\python_env\lib\site-packages\zarr\util.py", line 69, in json_dumps
    return json.dumps(
  File "f:\python_env\lib\json\__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "f:\python_env\lib\json\encoder.py", line 201, in encode
    chunks = list(chunks)
  File "f:\python_env\lib\json\encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "f:\python_env\lib\json\encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "f:\python_env\lib\json\encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "f:\python_env\lib\json\encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "f:\python_env\lib\json\encoder.py", line 438, in _iterencode
    o = _default(o)
  File "f:\python_env\lib\site-packages\zarr\util.py", line 64, in default
    return json.JSONEncoder.default(self, o)
  File "f:\python_env\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
psygnal.EmitLoopError:

  Signal emitted at: f:\python_env\lib\site-packages\pymmcore_plus\mda\_thread_relay.py:144, in run
    >  emitter.emit(*args)

  Callback error at: f:\python_env\lib\json\encoder.py:179, in default
    >  raise TypeError(f'Object of type {o.__class__.__name__} '

    Local variables:
       o = <OrderMode.row_wise_snake: 'row_wise_snake'>
```

tlambert03, 2024-09-23:
good catch, looks like there's not a test.  it's also annoying that the standard library json doesn't know how to serialize a standard library enum :joy:

i'm fine with adding msgspec to the default requirements if you guys are.  i think it's a generally superior library for serialization.  two alternatives would be:
- add a test to make sure that anything in the metadata is serializable with the standard json serializer
- add a custom json serializer that all the writers must use if msgspec isn't available.

preference?  just using msgspec is certainly the easiest and probably the fastest, if you don't mind the dependency

wl-stepp, 2024-09-24:
Seems at least not worth it testing both, so I vote adding the dependency.

tlambert03, 2024-09-24:
just uninstalled msgspec locally and the tests still passed actually.  So there's something more subtle going on here (a missing test case)

wl-stepp, 2024-09-24:
Hm, okay I will try to reproduce then with the DemoConfig and deliver more info.

tlambert03, 2024-09-24:
figured it out, and found a better fix, will post PR in a moment

closed 2024-09-24 by tlambert03