pymmcore-plus/pymmcore-plus #376: State of writers for MultiCamera acquisitions

wl-stepp, 2024-09-23:
Dear all,

I wanted to see what the current state for acquiring and storing MultiCamera acquisitions. I see that it runs Ok if I set it up correctly from the Engine and Runner side. I don't get two channels in the saved Zarrs however.
I think there are two problems here:

- event.index is the same for both

{'format': 'frame-dict', 'version': '1.0', 'runner_time_ms': 291.50379999919096, 'camera_device': **'HamamatsuHam_DCAM'**, 'property_values': ({'dev': 'CSUW1-Filter Wheel 1', 'prop': 'Label', 'val': 'Cy3'}, {'dev': 'CSUW1-Filter Wheel 2', 'prop': 'Label', 'val': 'DAPI'}, {'dev': 'DA TTL State Device', 'prop': 'State', 'val': '5'}, {'dev': 'Core', 'prop': 'Shutter', 'val': 'State Device Shutter'}, {'dev': 'CondenserTurret', 'prop': 'Label', 'val': '7-Closed'}), 'exposure_ms': 42.954, 'pixel_size_um': 0.065, 'position': {'x': 2344.8, 'y': -3700.2000000000003, 'z': 1968.02}, 'mda_event': MDAEvent(index=mappingproxy(**{'c': 0}**), channel=Channel(config='Dual-DAPI-Cy3', group='3-Channel'))}

 {'format': 'frame-dict', 'version': '1.0', 'runner_time_ms': 291.50379999919096, 'camera_device': **'HamamatsuHam_DCAM-1'**, 'property_values': ({'dev': 'CSUW1-Filter Wheel 1', 'prop': 'Label', 'val': 'Cy3'}, {'dev': 'CSUW1-Filter Wheel 2', 'prop': 'Label', 'val': 'DAPI'}, {'dev': 'DA TTL State Device', 'prop': 'State', 'val': '5'}, {'dev': 'Core', 'prop': 'Shutter', 'val': 'State Device Shutter'}, {'dev': 'CondenserTurret', 'prop': 'Label', 'val': '7-Closed'}), 'exposure_ms': 42.954, 'pixel_size_um': 0.065, 'position': {'x': 2344.8, 'y': -3700.2000000000003, 'z': 1968.02}, 'mda_event': MDAEvent(index=mappingproxy(**{'c': 0}**), channel=Channel(config='Dual-DAPI-Cy3', group='3-Channel'))}

Because of this, the writer just overwrites the frame from camera 0

- the writer does not know about the number of channels
Because of this if I hack in the incremented channel index in the runner, I get an out of bounds error in the writer. I have to additionally hack the extra channel into the writer in sequenceStarted

Is there another writer (Tensorstore?) that handles this better already? Have you thought about how this should be handled? So if I hack it in for now I do it in the least stupid way.

Thanks!

tlambert03, 2024-09-23:
i'd say the state is that the metadata was certainly a helpful step along the way, but that you might be the first to test it in a realistic situation.

In general, I prefer that TensorStore writer (if for no other reason than that I prefer tensorstore to zarr-python as a library), but I don't think that handler would do any better with this problem.

I haven't yet given it much though to how i think it should be handled, so we can discuss and implement together here.

my first naive thought is that `event.index` should indeed be the same (since it's triggered by the same `MDAEvent`), but I'm not set on that.  Obviously, we'd still need some way to disambiguate the two ... in the engine, [it would indeed be `camera_device` that would be different](https://github.com/pymmcore-plus/pymmcore-plus/blob/60a319e97c1bf466fd5336cb4987708158b2e723/src/pymmcore_plus/mda/_engine.py#L273-L278), as you've noted in your bolding above.  So, I suppose that means that all writers essentially need to have either a different file, or a different implicit channel for each camera device.   In MMCore, it *is* required that all multicamera members have the same ROI size, so one could conceptually add them to the same ND array... however, I wonder whether writing it that way might pin us into a future compatibility corner, and maybe we should just jump to writing different camera devices to different files?

So, i guess my proposal is: all writers should check the camera device and make N different files, where N is the number of unique camera devices encountered during the acquisition?

@marktsuchida, how do you guys deal with this in your writers?

wl-stepp, 2024-09-24:
I discussed this a little bit with the facility staff. Company softwares also seem to handle this in various ways, including several files. I get the thought and the user base for this is definitely limited.
However, intuitively I think this should go into the channel dimension.  You can take the exact same data with the same camera and would expect it to be organized in channels.
The ROI size is a good point... How would zarr handle different ROI sizes?

tlambert03, 2024-09-24:
> The ROI size is a good point... How would zarr handle different ROI sizes?

It would not... (zarr arrays cannot be jagged), that's the main point.  If you want to treat multiple cameras as multiple channels, then the images _must_ be the same dimensions.  Or, the smaller image could be padded by fill values.

I do see the intuitiveness of having it naturally in the channel axis though, so let's figure out how to make that an option, assuming the dimensions match.

marktsuchida, 2024-09-26:
In MMStudio we fold it into the channel axis, which I think is the right thing to do most of the time _in the saved (and displayed) dataset_, but at the stage of the acquisition engine it gets very nasty if there is a single axis for choice of camera and choice of (conventional) channel, when you have both (e.g., switching excitation wavelength as channel, while having 2 cameras with split emission).

There are also specialized cameras that produce more than 1 plane per frame (e.g., time-gated), so I feel that the acquisition code should not assume too much. Or maybe it defaults to treating as channel with the option to keep it separate.

When also thinking about metadata issues (say, saving to OME, which ideally incorporates a bunch of information that comes from places other than the acquisition engine), to me it seems natural to have a transformation between the acquisition engine and the final data representation, so that the acquisition engine doesn't need to know too much. In simple cases, this transformation may be omitted.