pymmcore-plus/pymmcore-plus #379: Acquisition stops temporarily while it catches up with frames

simonecoppola, 2024-09-27:
Related to this discussion: https://imagesc.zulipchat.com/#narrow/stream/442785-pymmcore.5B-plus.5D/topic/questions and to the micromanager-gui project as well.

Specifically: "Is it possible to run an acquisition in micromanager-gui without displaying the images? We've noticed that when performing fast acquisitions, often the acquisitions stop at the right time (say, a "fast" 20 or 50ms exposure acquisition for 5s - we see the led come on and off for those 5s) but then it waits until all the images have been displayed before starting on the second channel. We're not too interested in seeing all the images live (maybe the first for each channel would be good for sanity checking the focus is holding, but not too important) so if live display could be disabled, that would be great for us!"

The issue presents itself even if the acquisition is started as a script. I'm attaching here the full console log from the script run!
[console_log.txt](https://github.com/user-attachments/files/17166476/console_log.txt)

tlambert03, 2024-09-27:
thanks @simonecoppola, pasting the key part of the traceback here:

```pytb
Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_signal.py", line 1196, in _run_emit_loop
    self._run_emit_loop_inner()
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_signal.py", line 1225, in _run_emit_loop_immediate
    caller.cb(args)
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_weak_callback.py", line 453, in cb
    func(obj, *self._args, *args, **self._kwargs)
  File "C:\..\.virtualenv\Lib\site-packages\pymmcore_plus\mda\handlers\_tensorstore_handler.py", line 204, in frameReady
    self._store = self.new_store(frame, event.sequence, meta).result()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: ALREADY_EXISTS: Error opening "zarr" driver: Error writing local file "Z:\\Shared363\\data\\Coppola-Simone\\0_DATA\\openframe\\240926 PC19 BADA TADA\\test_speed\\console/.zarray" [source locations='tensorstore/internal/cache/kvs_backed_cache.h:208\ntensorstore/driver/driver.cc:112'] [tensorstore_spec='{\"context\":{\"cache_pool\":{},\"data_copy_concurrency\":{},\"file_io_concurrency\":{},\"file_io_sync\":true},\"create\":true,\"driver\":\"zarr\",\"dtype\":\"uint16\",\"kvstore\":{\"driver\":\"file\",\"path\":\"Z:\\\\Shared363\\\\data\\\\Coppola-Simone\\\\0_DATA\\\\openframe\\\\240926 PC19 BADA TADA\\\\test_speed\\\\console/\"},\"metadata\":{\"dimension_separator\":\"/\"},\"schema\":{\"chunk_layout\":{\"read_chunk\":{\"shape\":[1,1,2048,2048]},\"write_chunk\":{\"shape\":[1,1,2048,2048]}},\"domain\":{\"exclusive_max\":[2,100,2048,2048],\"inclusive_min\":[0,0,0,0],\"labels\":[\"c\",\"t\",\"y\",\"x\"]}},\"transform\":{\"input_exclusive_max\":[[2],[100],[2048],[2048]],\"input_inclusive_min\":[0,0,0,0],\"input_labels\":[\"c\",\"t\",\"y\",\"x\"]}}']

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\..\AppData\Local\Programs\Python\Python312\Lib\threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "C:\..\.virtualenv\Lib\site-packages\pymmcore_plus\mda\_thread_relay.py", line 144, in run
    emitter.emit(*args)
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_signal.py", line 1176, in emit
    self._run_emit_loop(args)
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_signal.py", line 1213, in _run_emit_loop
    raise loop_err from cb_err  # emit() call ABOVE || callback error BELOW
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_signal.py", line 1196, in _run_emit_loop
    self._run_emit_loop_inner()
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_signal.py", line 1225, in _run_emit_loop_immediate
    caller.cb(args)
  File "C:\..\.virtualenv\Lib\site-packages\psygnal\_weak_callback.py", line 453, in cb
    func(obj, *self._args, *args, **self._kwargs)
  File "C:\..\.virtualenv\Lib\site-packages\pymmcore_plus\mda\handlers\_tensorstore_handler.py", line 204, in frameReady
    self._store = self.new_store(frame, event.sequence, meta).result()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
psygnal.EmitLoopError:

While emitting signal 'pymmcore_plus.mda.events._psygnal.MDASignaler.frameReady', a ValueError occurred in a callback:

  Signal emitted at: C:\..\.virtualenv\Lib\site-packages\pymmcore_plus\mda\_thread_relay.py:144, in run
    >  emitter.emit(*args)

  Callback error at: C:\..\.virtualenv\Lib\site-packages\pymmcore_plus\mda\handlers\_tensorstore_handler.py:204, in frameReady
    >  self._store = self.new_store(frame, event.sequence, meta).result()

    Local variables:
       frame = array([[204, 197, 232, ..., 269, 224, 233],
       [214, 200...
       event = MDAEvent(index=mappingproxy({'t': 0, 'c': 0}), channel=Chann...
       meta = {'format': 'frame-dict', 'version': '1.0', 'runner_time_ms':...
```

tlambert03, 2024-09-27:
the problem in the log there is related to a pre-existing file that zarr doesn't want to overwrite.  You can use the `delete_existing` when instantiating the `TensorStoreHandler()` to avoid that.  (Of course, micromanager-gui would need to do that for you when using the GUI directly)

I'm not sure it's directly related to the question of pausing between channels...  could you perhaps also provide an example of the `MDASequence` you're trying to run, and perhaps the script you use to start the experiment (just in case there are any additional settings or details that might be relevant)

simonecoppola, 2024-09-30:
It looks like the delete_existing was not the issue with the pausing, I just tried it. Here's the code I used:

```from pymmcore_plus import CMMCorePlus
from useq import MDAEvent, MDASequence
from pymmcore_plus.mda.handlers import TensorStoreHandler
mmcore = CMMCorePlus.instance()
mmcore.loadSystemConfiguration(path_to_config)
mmcore.setExposure(50)
mmcore.setCircularBufferMemoryFootprint(10000)

sequence = MDASequence(
    channels=[{"config": "Ex470-QUAD", "exposure": 50}, {"config": "Ex550-QUAD", "exposure": 50}],
    time_plan={"interval": 0, "loops": 100},
    axis_order="ct",
)
mmcore.run_mda(sequence, output=TensorStoreHandler(path=path_to_savefile", delete_existing=True,
                                                   driver="zarr"))
```

Here's the output, this time with no errors:

```
2024-09-30 11:10:05,819 - pymmcore-plus - INFO - (_runner.py:321) MDA Started: axis_order=('c', 't') channels=(Channel(config='Ex470-QUAD', exposure=50.0), Channel(config='Ex550-QUAD', exposure=50.0)) time_plan=TIntervalLoops(interval=datetime.timedelta(0), loops=100)
2024-09-30 11:10:05,820 - pymmcore-plus - INFO - (_runner.py:283) index={'t': 0, 'c': 0} channel=Channel(config='Ex470-QUAD') exposure=50.0 min_start_time=0.0 reset_event_timer=True events=(MDAEvent(index=mappingproxy({'t': 0, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0, reset_event_timer=True), MDAEvent(index=mappingproxy({'t': 1, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 2, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 3, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 4, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 5, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 6, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 7, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 8, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 9, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 10, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 11, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 12, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 13, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 14, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 15, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 16, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 17, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 18, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 19, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 20, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 21, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 22, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 23, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 24, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 25, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 26, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 27, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 28, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 29, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 30, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 31, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 32, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 33, 'c': 0}), channel=Channel(config='Ex470-QUAD'), exposure=50.0, min_start_time=0.0), MDAEvent(index=mappingproxy({'t': 34, 'c': 0}), channel=Channel(confi [... cut ...]
2024-09-30 11:10:36,236 - pymmcore-plus - INFO - (_runner.py:408) MDA Finished: axis_order=('c', 't') channels=(Channel(config='Ex470-QUAD', exposure=50.0), Channel(config='Ex550-QUAD', exposure=50.0)) time_plan=TIntervalLoops(interval=datetime.timedelta(0), loops=100)
```

I should note that the behaviour is the same whether I run this just as a script, or with the micromanager-gui open. The only difference is that if the micromanager-gui is open, I see that it's catching up with the frames on the display.
Thanks again for all your help with this!

fdrgsp, 2024-09-30:
Hi @simonecoppola, I did a test on my microscope and it seems to works ok for me. Can you try too add the `keep_shutter_open_across` attribute to the MDASequence (this can be helpful if you have a shutter specified in the channels configs):

```py
sequence = MDASequence(
    channels=[{"config": "Ex470-QUAD", "exposure": 50}, {"config": "Ex550-QUAD", "exposure": 50}],
    time_plan={"interval": 0, "loops": 100},
    axis_order="ct",
    keep_shutter_open_across="t",
)
```
 
I've also notice that in our `MDAWidget` we do not currently support the `ct` axis order (only `tc`) and maybe @tlambert03 we should add it (https://github.com/pymmcore-plus/pymmcore-widgets/blob/3dab0c4c22cc68916f51c2d9deccd0567d686c7a/src/pymmcore_widgets/useq_widgets/_mda_sequence.py#L51-L62)

simonecoppola, 2024-09-30:
Hi @fdrgsp , I just tried adding the `keep_shutter_open_across="t"` but it made no difference :( it still acquired one channel, then paused for a bit, then did the other.

Regarding the support for `ct`, you totally should! It's useful for SRRF and not allowed in "normal" Micro-Manager (see https://forum.image.sc/t/scripting-complex-aquisitions-in-micro-manager/64555/7 and https://forum.image.sc/t/help-needed-to-set-up-leica-dmi4000b-with-micromanager/41211 for example of people being interested in the feature). It was actually what originally sent me towards pymmcore-plus, combined with needing to integrate some other stuff we already had written in python

fdrgsp, 2024-09-30:
@simonecoppola can you paste here your `cfg` file? Thanks!

simonecoppola, 2024-09-30:
Sure thing, here it is:
[openframe_config.txt](https://github.com/user-attachments/files/17192460/openframe_config.txt)

fdrgsp, 2024-09-30:
Hey @simonecoppola, is there any (do you hear any) hardware changing (e.g. shutter opening and closing, ...) while the acquisition is running and you see the "waiting" time between images? Do you have any delay set in any of the device you've added through the micromanager config wizard?

simonecoppola, 2024-10-01:
Hi @fdrgsp,
Nothing audible and no delays that I'm aware of!

I've done a bit more testing, and the delay between the two channels seems to be correlated with how many frames I'm acquiring (even if I'm just running a script and not the micromanager-gui), for example: if I acquire 10 frames for each channel, the delay is pretty minimal, if I increase this to 100 frames it gets longer etc. The behaviour does not change whether I am saving the data, or whether I'm just running `mmcore.run_mda(sequence)`. 

I've also run the same acquisitions while having the micromanager-gui open and not (i.e. just running a script), the delay between starting the acquisition for each channel is consistent regardless of the method, with the only difference being that when I have micromanager-gui open I can see that the acquisition for the second channel starts immediately after all the frames for the first channel have been displayed.

Thanks again for all your help with helping me troubleshoot this!

fdrgsp, 2024-10-01:
Hey @simonecoppola, I am not sure what is happening...can you try to run this code and get info about the actual timing between frames (maybe start form the `camera_metadata` `ElapsedTime-ms`)?

```py
import numpy as np
from pymmcore_plus import CMMCorePlus
from rich import print
from useq import MDAEvent, MDASequence

def on_frame_ready(img: np.ndarray, event: MDAEvent, metadata: dict) -> None:
    """Callback function to be called when a frame is ready."""
    print(metadata)

cfg = "/Users/fdrgsp/Desktop/test_config.cfg"
mmc = CMMCorePlus.instance()
mmc.loadSystemConfiguration(cfg)
mmc.mda.events.frameReady.connect(on_frame_ready)

sequence = MDASequence(
    channels=[{"config": "Ex470-QUAD", "exposure": 50}, {"config": "Ex550-QUAD", "exposure": 50}],
    time_plan={"interval": 0, "loops": 50},
    axis_order="ct",
    keep_shutter_open_across="t",
)

mmc.mda.run(sequence)
```

Another thing I would do is to try to use the demo camera instead of the Andor one and see if the "waiting" time is still there. 

You could also try to remake the channels configs step by step by adding one device at a time, running the sequence, and then see when the timing issue occurs...

simonecoppola, 2024-10-02:
Hi @fdrgsp ,
I've done some more testing while printing the metadata, here's my observations:

`ElapsedTime-ms` reset at the start of the second channel so I couldn't use that, so I decided to keep an eye on the `runner_time_ms` and `TimeReceivedByCore` properties.

Looking at those, if I ran the sequence with 10 frames per channel, I got a 2 second delay between the end of the first channel and the start of the second. This increased to 5 seconds for 50 frames and 20 seconds for 200 frames per channel.

Another thing I noticed is that the `images_remaining_in_buffer` just grows and grows during the acquisition and then goes down - this is particularly obvious when acquiring 200 frames per channel. The printing statements continue well after the first LED has gone off, though the timings etc look ok to me. I'm attaching the console output from that acquisition so you can have a look for yourself in case there's something else I missed.

I've also tried different combinations of the devices in the channel, or just having one at a time etc, and unfortunately I observed the same behaviour. 

I didn't see this behaviour when just running the acquisition through the demo camera, though the demo camera is 512x512 instead of 2048x2048 so if it's an issue of things building in the buffer maybe that plays a role? I'm not sure how to change the demo camera to being 2048x2048.

[acquisition_metadata_output.txt](https://github.com/user-attachments/files/17226666/acquisition_metadata_output.txt)

marktsuchida, 2024-10-02:
If it helps, you can change the DemoCamera image size by setting the (oddly-named) properties `OnCameraCCDXSize` and `OnCameraCCDYSize`.

tlambert03, 2024-10-02:
haven't had time to sit down with this, but i had a thought yesterday that this is all probably entirely explained by these lines in the `exec_sequenced_event` method:

https://github.com/pymmcore-plus/pymmcore-plus/blob/1cd0678c889ddf88a0ce43b8831289620317c952/src/pymmcore_plus/mda/_engine.py#L447-L451

in short: pymmcore-plus currently won't "move on" to the next event until the circular buffer is empty.  That's not a "good" design per se :joy: ... but it's almost certainly causing what you're seeing here.  assuming your hardware can't sequence across channels (safe assumption), that what creates the boundary between one "sequenced event" and another.  and the magnitude of the pause will depend on how many images are remaining in the circular buffer (explaining why it's probably all but undetectable with the 512x512 default size of the democam).

Sorry i didn't think of that sooner.  kind of an obvious explanation.  Also probably something that's relatively straightforward to fix, but may require a bit of a break in the interface between the engine (which pulls images off the buffer) and the runner (which is currently responsible for emitting frames as ready).  need to think about it

tlambert03, 2024-10-07:
> in short: pymmcore-plus currently won't "move on" to the next event until the circular buffer is empty. That's not a "good" design per se

actually, after talking with @marktsuchida, I realized that there is actually a core limitation here, which is that we can't reasonably move on to a new call to`startSequenceAcquisition` until we indeed have `getRemainingImageCount()` of 0.  So, while the bit I mentioned above is true (i.e. that `exec_sequenced_event` will not finish until it has finished processing the buffer), it's a little less apparent to me now how we should deal with this.

simonecoppola, 2024-10-18:
Hi @tlambert03  - could I confirm that I understood your previous post right and that the problem causing the pauses is the lack of triggering?
I have now set up my light sources to be triggered, and can generate events which are sequenceable, which I checked by first checking the relevant properties return `true` when I use `mmc.isPropertySequenceable` and that also `mmc.canSequenceEvents` returns `True` between all events. However, I'm still getting the same delays and same buffer problem (which I confirmed by looking at the metadata as done before) :( I make sure to also call `mmc.mda.engine.use_hardware_sequencing = True` before starting the acquisition. Is there anything else obvious I should check to make sure events are being triggered correctly?

tlambert03, 2024-10-18:
hey @simonecoppola, I know, it's a bit confusing... I'd love to chat with you about this directly! (I pinged you on zulip).

In short: the crux of this issue is this:

> in short: pymmcore-plus currently won't "move on" to the next event until the circular buffer is empty. 

And when I say "next event" here, I actually mean the next entire sequence of events (so, if you are able to sequence say 20 frames together from one channel, and then 20 frames from the next channel, that would be 2 "events" from the perspective of the engine...).  So, the engine is basically waiting for the buffer to clear from that first 20 before _starting_ the next 20.

However, after talking with @marktsuchida, I realized that that's actually probably the correct design, because MMCore itself (including when run on the Java side) will empty the existing circular buffer before starting a new sequence.  So we basically *have* to wait.

My main questions for you (apologies if you answered these elsewhere):
- how big is your image size? (2k x 2k?)
- how long is the exposure / frame during the sequence 
- how many images are you capturing into each sequenced "meta"-event?
- have you ever tried this in the Java micro-manager?  or is it simply not possible because of the experiment design (i.e. doing a timelapse on one channel before moving to the next channel)?

in short: i'm trying to determine whether this might just be a hard limitation of how micro-manager's sequenced acquisitions play with micro-manager's circular buffer.  If so, the very best we'll be able to do is to try to speed up the rate at which we *clear* the circular buffer and write to disk; an important thing to optimize to be sure... but also much trickier

tlambert03, 2024-11-08:
lots of off-github discussion has happened here.  Thanks a bunch to @simonecoppola for some very good sleuthing.  

I think at the end of the day this largely boils down to slow calls to `getXYPosition`, which was slowing down the speed of popping frames (with metadata) off the circular buffer.  that's a lesson that I think has already been learned before on the Java side (am I correct about that @marktsuchida)... so we should have done that better here.  #392 is an attempt to fix that

PR #392 (2024-11-15): perf: don't query position on triggered acquisition frames by default, add `include_frame_position_metadata` flag
This PR makes it so that the MDAEngine doesn't query stage positions during sequenced/triggered acquisitions by default.  This can be a rather slow call, depending on the stage, and ultimately was at the heart of @simonecoppola's performance issues seen in https://github.com/pymmcore-plus/pymmcore-plus/issues/379 (thank you **SO** much Simone for debugging that).

this PR adds:

```python
MDAEngine:
    include_frame_position_metadata: Literal[True, False, "unsequenced-only"] = "un [... cut]
closed 2024-11-15 by tlambert03