pymmcore-plus/pymmcore-plus #387: Feature Request - Remote Control/Usage

JojoDevel, 2024-10-31:
## Why?

Our microscopes (mostly NikonTi Eclipse) are controlled by computer just next to them running windows not always on the most up-to-date version and sometimes even air-gapped form the internet. Therefore, we usually try to touch these machines as little as possible, install as little software as possible and let them only interface with the microscope.

## What?

We would like to run the napari-micromanager on any other computer in our lab or institute and attach remotely to the running microscope control computer. This would allow us to much easier maintain/develop software, control imaging loops easier, and maybe even work remotely from our labs and dial into running image acquisitions from all over the world...

## How?

Not sure yet. I imagine a solution with as little programming overhead as possible but keeping the same functionality as right now. So the developer should not feel that everything goes through a remote connection (Though speed will possibly slower).

There is for example, [pyRO](https://pyro5.readthedocs.io/en/latest/) that allows to execute functions remotely. Maybe we could turn every device in the pymmcore python api into such a remote object. And then the python code just uses calls that are executed remotely. On the microscope computer then a pyro server would run initiated from a micromanager configuration. I never used the framework though.

So, there might be also be better opportunities to implement that. Any ideas?

**Hint:** We also discussed that on the EPFL Symposium for Smart Microscopy 2024

tlambert03, 2024-11-03:
hey @JojoDevel,  I extracted the logic from #208 and created a new package: https://github.com/pymmcore-plus/pymmcore-remote

when you have a moment, it would be great to get your eyes on it to see if it's going in a direction that would be useful to you.  See the readme there for details on installing, and using

JojoDevel, 2024-11-04:
Yes, this looks very good :+1: Great work!
Is there a way to get it to full network serialization? So that no shared memory is used and communication goes through network?

But maybe we should move the issue over to the new repo...

tlambert03, 2024-11-04:
Yes actually i found a good solution with msgpack-numpy and will use it 👍

I tried briefly to use it with napari micromanager and there are still some remaining issue with serialization of certain custom objects, so I'll keep working on it and let you know when I get a client side gui working with a remote core