pymmcore-plus/pymmcore-plus #398: TODO: decide convention for variable name for core/mmc/mmcore

tlambert03, 2024-11-15:
This is a rather bike-sheddy topic, but in various places like function signatures, examples, and docstrings, we use a different variable name for an instance of `CMMCorePlus()`.  sometimes it's called `mmcore`, occasionally `mmc`, and sometimes just `core`.  For uniformity, we should pick and stick to just one.

@marktsuchida, would appreciate your opinion here.

gselzer, 2024-11-15:
For what it's worth, I like `mmc` - it's the shortest of the three, and it's closest to an acronym.

marktsuchida, 2024-11-19:
I also like `mmc` and almost always use that myself (ever since BeanShell, in which the object is provided under that name). 

I think `core` has too much chance of clashing with other things.

`mmcore` might be more easily confused with `CMMCore[Plus]` (the type name) or pymmcore[-plus] (the package name), at least by complete novices.

tlambert03, 2024-11-19:
fantastic.  mmc it is :)