pymmcore-plus/pymmcore-plus #418: Blocking / Timing Issue with Event-Driven Acquisition Engine

alandolt, 2025-01-16:
Hello
First, I'd like to extend a big thanks to all the contributors of this project. It’s been incredibly helpful in advancing our microscopy work.

I’m currently working on a real-time feedback microscopy pipeline based on pymmcore-plus. As part of this, I’m implementing the event-driven acquisition engine outlined in the documentation (event_driven_acquisition.py, [Link to Documentation](https://pymmcore-plus.github.io/pymmcore-plus/guides/event_driven_acquisition/)).

In my setup, I have another thread that calculates stimulation masks based on the images captured. This process introduces a blocking queue (event_queue.get) before adding MDA events to the queue, which seems to cause an issue with the acquisition engine.

## Issue Description: 

Due to this blocking behavior, the engine does not operate as expected. Instead of sequentially processing events with proper timing, the events appear to be batched incorrectly. Below is a demonstration of the issue. The code simulates blocking by using time.sleep before adding events to the queue:

```
from useq import MDAEvent
import queue
import pymmcore_plus
import time
import threading
def run_controller_in_thread(event_queue, stop_event):
    time.sleep(5)
    print("1st")
    event_queue.put(MDAEvent(index={"t": 0}, exposure=2000, min_start_time=0))
    time.sleep(10)
    print("2nd")
    event_queue.put(MDAEvent(index={"t": 1}, exposure=1000, min_start_time=0))
    time.sleep(10)
    print("3rd")
    event_queue.put(MDAEvent(index={"t": 2}, exposure=1000, min_start_time=0))
    time.sleep(15)
    event_queue.put(stop_event)

if __name__ == "__main__":
    STOP_EVENT = object()
    event_queue = queue.Queue()
    mmc = pymmcore_plus.CMMCorePlus()
    mmc.run_mda(iter(event_queue.get, STOP_EVENT))
    mmc.loadSystemConfiguration()
    mmc.mda.events.frameReady.disconnect()

    @mmc.mda.events.frameReady.connect
    def on_frame(img, event):
        print(f'Frame {event.index} received: {img.shape}')
    process = threading.Thread(target=run_controller_in_thread, args=(event_queue, STOP_EVENT))
    process.start()
    process.join()
```

**Output:** 
```
[01/16/25 14:54:07] INFO     MDA Started: GeneratorMDASequence()                                                                                                                                                                                                          
1st
2nd
index={'t': 0} exposure=2000.0 min_start_time=0.0
[01/16/25 14:54:22] INFO     index={'t': 0} exposure=2000.0 min_start_time=0.0                                                                                                                                                                                            _runner.py:292
Frame {'t': 0} received: (512, 512)
3rd
index={'t': 1} exposure=1000.0 min_start_time=0.0 events=(MDAEvent(index={'t': 1}, exposure=1000.0, min_start_time=0.0), MDAEvent(index={'t': 2}, exposure=1000.0, min_start_time=0.0)) exposure_sequence=() x_sequence=() y_sequence=() z_sequence=()
[01/16/25 14:54:47] INFO     index={'t': 1} exposure=1000.0 min_start_time=0.0 events=(MDAEvent(index={'t': 1}, exposure=1000.0, min_start_time=0.0), MDAEvent(index={'t': 2}, exposure=1000.0, min_start_time=0.0)) exposure_sequence=() x_sequence=() y_sequence=()     _runner.py:292
                             z_sequence=()
Frame {'t': 1} received: (512, 512)
Frame {'t': 2} received: (512, 512)
[01/16/25 14:54:49] INFO     MDA Finished: GeneratorMDASequence()   
```

### Expected Behavior:

- The first image should be captured after a 5-second delay.
- The second image should be captured 10 seconds later, and the third image should be taken after an additional 10 seconds.

### Actual Behavior:

Instead of capturing the images sequentially, the first image is captured after 15 seconds, followed by both the second and third images being captured simultaneously at 25 seconds. This issue is not only reflected in the logs but also visible in the actual microscope operation.

## Troubleshooting Steps 

I have tried several approaches to resolve this issue:

1. Running the MDA part in a separate thread: I’ve attempted to move the MDA part, including frameReady and run_mda, to a separate thread, but the issue persists.

2. Using multiprocessing: Switching to multiprocessing rather than threading resulted in the same behavior.
3. Custom Iterable Queue: I’ve also attempted to implement my own IterableQueue class, but this did not resolve the problem either.

```
class IterableQueue(queue.Queue):
    def __init__(self, *args, **kwargs):
        super().__init__(*args, **kwargs)
        self._lock = threading.Lock()

    def _iter(self):
        return self

    def __next__(self):
        with self._lock:
            return self.get(block=True)
```

4. Manually retrieving events from the queue: As a temporary workaround, I manually retrieve events from the queue and feed them into mmc.run_mda, but this approach loses the ability to specify min_start_time, which is crucial for my application.
```
import threading
from useq import MDAEvent
import queue
import pymmcore_plus
import time

def run_controller_in_thread(event_queue, stop_event):
    time.sleep(5)
    print("1st")
    event_queue.put(MDAEvent(index={"t": 0}, exposure=2000, min_start_time=0))
    time.sleep(10)
    print("2nd")
    event_queue.put(MDAEvent(index={"t": 1}, exposure=1000, min_start_time=0))
    time.sleep(10)
    print("3rd")
    event_queue.put(MDAEvent(index={"t": 2}, exposure=1000, min_start_time=0))
    time.sleep(15)
    event_queue.put(stop_event)

def run_mda_in_thread(_mmc, event_queue, stop_event):
    while True:
        if event_queue.empty():
            time.sleep(0.1)
        else:
            current_event = event_queue.get()
            if current_event == stop_event:
                break
            _mmc.mda.run([current_event])

if __name__ == "__main__":
    STOP_EVENT = object()
    event_queue = queue.Queue()
    mmc = pymmcore_plus.CMMCorePlus()
    mmc.loadSystemConfiguration()

    mmc.mda.events.frameReady.disconnect()

    @mmc.mda.events.frameReady.connect
    def on_frame(img, event):
        print(f'Frame {event.index} received: {img.shape}')

    process = threading.Thread(target=run_controller_in_thread, args=(event_queue, STOP_EVENT))
    process_mda_runner = threading.Thread(target=run_mda_in_thread, args=(mmc, event_queue, STOP_EVENT))
    process_mda_runner.start()
    process.start()
    process.join()
    process_mda_runner.join()
```

I would greatly appreciate any advice or insights on how to resolve this blocking behavior while still being able to specify min_start_time for each MDA event.

Thank you in advance for your help!

tlambert03, 2025-01-16:
hey @alandolt 👋

you've been bitten by a bug (for which there is no issue) where running with a Queue plays poorly with the attempt to optimistically use hardware triggering where available.  It's actually fixed in https://github.com/pymmcore-plus/pymmcore-plus/pull/400 (see bullet point 5 in that PR description).... but I should have pulled out that fix and merged it on its own by now.  

To fix your use case, you can add `mmc.mda.engine.use_hardware_sequencing = False` before running the expeirment:

```python
if __name__ == "__main__":
    STOP_EVENT = object()
    event_queue = queue.Queue()
    mmc = pymmcore_plus.CMMCorePlus()
    mmc.mda.engine.use_hardware_sequencing = False
    ...
```

I will post a fix shortly

alandolt, 2025-01-16:
Thank you very much for your quick response! The solution works perfectly with the demo example I provided, it also works with real hardware.

closed 2025-01-16 by alandolt

tlambert03, 2025-01-16:
thanks, i will merge #419 soon and release a new version, after which you will no longer need to use `mmc.mda.engine.use_hardware_sequencing = False` (but using it won't be a problem)

PR #419 (2025-01-16): fix: Don't use `Engine.event_iterator` when an Iterable is directly passed to `MDARunner.run`
This fixes #418  (and is pulled out of #400).

When the user is directly passing an iterator object, such as `iter(queue.get, None)` to `MDARunner.run`, we don't want the MDAEngine to do complicated things like try to setup hardware triggering.  This PR essentially implies that using queues and passing iterables is an advanced use case, and we don't want to inject additional "magic".  This *does* mean that queue users won't be able to use hardware triggering (for now) but that will be solved by asking [... cut]