pymmcore-plus/pymmcore-plus #431: Best place to modify MDAsequence between MDAWidget and MDAEngine

dpshepherd, 2025-02-08:
more questions...

We have a mode of our instrument where we execute the same set of `MDAEvents`, but need to run a microfluidics program in between with a custom microfluidics controller setup that doesn't have MM drivers. Because that is hard to represent in the MDA window, I was wondering if the following callback strategy might work.

Ultimately, we want to do this in [pymmcore-gui](https://github.com/pymmcore-plus/pymmcore-gui) which is why I'm thinking about doing it the following way instead of programmatically defining the `MDASequence`.

1. set a `DemoDevice` to the type of fluidics program we want to run.
3. connect custom callback function to a signal from `MDAWidget` (maybe `run_btn.clicked`)? Looks like we'd have to disconnect to the existing signal too.
4. Rework the `MDASequence` to contain the custom metadata, pulled from the `DemoDevice` state, that we need for our custom engine to know that that should execute fluidics. I think this would entail adding a custom `MDAEvent` as a new first first event for the widget-generated `MDASequence`. Then make `n` copies of the modified `MDASequence`, corresponding to the `n` microfluidic programs we need to run.
5. Re-assign the modified `MDASequence `  using `MDAWidget.set_value()`
6. Call `run_mda()`

I thought about doing this work in our custom `MDAEngine` as well, but because it ingests the `MDASequence` I was wondering if we should do it before calling the engine.

Any and all feedback and insights (including this is the totally the wrong strategy) are welcome.

Thanks!!

ianhi, 2025-02-09:
I have a similar situation where I insert custom metadata after desinging the basic MDA in the GUI. I just take the MDASequence from the GUI, modify it, then directly run it by passing it to core run mda. I don't thnk you necessarily need to put the the modified verison back into the MDAWidget

https://github.com/ianhi/raman-mda-engine/blob/f7238bc286aa9642331eb5c6215c458e31327115/raman_mda_engine/utils.py#L12

dpshepherd, 2025-02-09:
Thanks! That definitely looks like what we need.

I searched through your repo and didn't see where you call that function. Do you have another repo where you modified the GUI that you are using?

tlambert03, 2025-02-09:
Hey @dpshepherd  (and thanks @ianhi!)

Broadly speaking, this is a great thing to bring up: the general question of "enhancing" a standard `MDASequence` to either intersperse custom events or inject custom metadata is an important one.  Programmatically and conceptually, the solution space is very broad, anything that maps an `Iterable[MDAEvent] -> Iterable[MDAEvent]` would work, so there will definitely be be many solutions depending on the needs.

I know the main question here is more about how to get handle on the important objects, at the right moments.  Here's a thought dump.

### injecting sequence-level metadata

Let's start with the question of injecting custom stuff.  There's many places this could happen

> I thought about doing this work in our custom MDAEngine as well, but because it ingests the MDASequence I was wondering if we should do it before calling the engine.

Let's explore this.  The MDAEngine (as a protocol) doesn't currently have much power to modify the sequence.  While the engine's `setup_sequence` is indeed [called shortly after `mda.run`](https://github.com/pymmcore-plus/pymmcore-plus/blob/b427f8ae10b5a7a763b612c77f4a5675cd94ea5e/src/pymmcore_plus/mda/_runner.py#L374), its ability to *mutate* the sequence is limited, because `MDASequence` is intentionally immutable at this point.  If *all* you need to do however, is add some metadata into the top-level `MDASequence.metadata` dict, that *will* work at the level of your custom MDAEngine, since that one field is still a mutable object.

```python
from pymmcore_plus.mda import MDAEngine

class MyEngine(MDAEngine):
    def setup_sequence(self, sequence: MDASequence) -> SummaryMetaV1 | None:
        # access self.mmcore to derive what you need and assign it in metadata
        sequence.metadata['custom-key'] = 'custom-value'
        return super().setup_sequence(sequence)
```

### mutating the event sequence itself

However, sequence-level metadata won't be enough if you need to actually change the `Iterable[MDAEvent]` that gets passed to the engine.  Since you say "We have a mode of our instrument where we execute the same set of MDAEvents" ... it sounds like you might not be in this situation.  That is: it's possible that you already have sufficient power by overriding the engine's `setup_event` method that you don't need to actually change the events themselves.

*If* you did need to though, then we need to actually change the sequence before it gets passed to `mmcore.mda.run`, and for that we'd need to patch the widget.

#### getting a handle for the `MDAWidget`

@ianhi's suggestion would definitely work for many cases, (side note: I'm also not finding where he would actually call that function: looks like the only time that function appears in [all of github](https://github.com/search?type=code&auto_enroll=true&q=%22get_seq_from_napari%22) is that definition).  

In terms of getting a handle on the MDAWidget itself, here's his code **for napari-micromanager**:

```python
mda_dock = main_window._dock_widgets["MDA"]
mda_widget = mda_dock.children()[4]
```

and here's how you would do that in **`pymmcore-gui`** (assuming you're starting with a handle to the `MicroManagerGui` as in [the example we previously discussed](https://github.com/pymmcore-plus/pymmcore-gui/issues/34#issuecomment-2644110420)).  The `MicroManagerGUI.get_widget` method will give you handle on any of the widgets defined for [`WidgetAction`](https://github.com/pymmcore-plus/pymmcore-gui/blob/95c4ed3c3089ffb4c1b1cadc0364ad309e989267/src/pymmcore_gui/actions/widget_actions.py#L138-L151)

```python
from pymmcore_gui import MicroManagerGUI, WidgetAction

win = MicroManagerGUI()
dock_widget = win.get_widget(WidgetAction.MDA_WIDGET)
```

#### modifying the sequence that gets run

This could be done many ways, but I think the safest way is probably *not* to use `set_value`. but rather to step in just before the value is passed on to the `run_mda` method, and that is [right here, in the `execte_mda` method on the widget.](https://github.com/pymmcore-plus/pymmcore-widgets/blob/9db494b19fd7da6774801b20489f37f3bb269caf/src/pymmcore_widgets/mda/_core_mda.py#L267-L271)

I may regret it :joy:, but I'm going to suggest you actually patch that method at runtime (an alternative would be for us to give you some way to specify a custom MDAWidget subclass to use instead of the one from `pymmcore-widgets`)

```python
from pymmcore_gui import MicroManagerGUI, WidgetAction

win = MicroManagerGUI()
dock_widget = win.get_widget(WidgetAction.MDA_WIDGET)
# Note: I dislike this line, and will be changing pymmcore-gui so that
# `get_widget` always gives you the widget itself, and not sometimes a dock widget
widget = dock_widget.widget()  # <--- THIS is now an MDAWidget

def custom_execute_mda(widget, output: Path | str | object | None) -> None:
    sequence = widget.value()
    # go nuts here, modify the sequence however you want
    # the only rule is that it has to still be an `Iterable[MDAEvent]` when you pass it in
    self._mmc.run_mda(sequence, output=output)

# modify the method on the instance
widget.execute_mda = custom_execute_mda
```

dpshepherd, 2025-02-09:
> If all you need to do however, is add some metadata into the top-level MDASequence.metadata dict, that will work at the level of your custom MDAEngine, since that one field is still a mutable object.

There are one or two acquisition modes where we could do it this way. However,

>If you did need to though, then we need to actually change the sequence before it gets passed to mmcore.mda.run, and for that we'd need to patch the widget.

Is what we need for the other modes.

> I may regret it 😂, but I'm going to suggest you actually patch that method at runtime (an alternative would be for us to give you some way to specify a custom MDAWidget subclass to use instead of the one from pymmcore-widgets)

Got it! Will move forward this way for those acquisition we need to mutate the sequence itself.

Thanks to both of you!

tlambert03, 2025-02-09:
quick note to update pymmcore-gui if you try my code. https://github.com/pymmcore-plus/pymmcore-gui/pull/35  changes the return type of `MicroManagerGui.get_widget` to be more predictable

closed 2025-02-09 by dpshepherd

tlambert03, 2025-02-10:
> The MDAEngine (as a protocol) doesn't currently have much power to modify the sequence. While the engine's setup_sequence is indeed [called shortly after [...]

@dpshepherd, I was thinking about this some more, and this statement I made is only partially true.  I forgot about a very powerful mechanism that the engine has (perhaps too powerful), and that is the [`MDAEngine.event_iterator` method](https://pymmcore-plus.github.io/pymmcore-plus/api/mda/#pymmcore_plus.mda._protocol.PMDAEngine.event_iterator)

That method is a function that maps an `Iterable[MDAEvent]` -> `Iterator[MDAEvent]`. It is [invoked by the runner in the main MDA loop](https://github.com/pymmcore-plus/pymmcore-plus/blob/b427f8ae10b5a7a763b612c77f4a5675cd94ea5e/src/pymmcore_plus/mda/_runner.py#L324-L329) and gets to spy on, and mutate if desired, every event.

It's what the base MDAEngine [uses to achieve opportunistic hardware triggering](https://github.com/pymmcore-plus/pymmcore-plus/blob/b427f8ae10b5a7a763b612c77f4a5675cd94ea5e/src/pymmcore_plus/mda/_engine.py#L209-L220) if the option is enables.  

A simple passthrough override would look like this:

```python
from pymmcore_plus.mda import MDAEngine

class CustomEngine(MDAEngine):
    def event_iterator(self, events: Iterable[MDAEvent]) -> Iterator[MDAEvent]:
        for event in super().event_iterator(events):
            # you can do what you need here
            yield event
```

A word of caution there: if your mutations change the number of frameReady events or the event indexes that get emitted, some widgets that display (or possible save) images might not work as expected  (but, I would be curious to find and squash those cases).

So, that's a general mechanism your lab might find useful for cases where you really need to get in there and muck about.

dpshepherd, 2025-02-10:
Most of the time we are going to be changing the number of images expected to be acquired, because of how the instrument works.

We are working to get the approach using `custom_excecute_mda()` above going. Once we got something hacky working, we'll come back to you for some feedback. 

thanks!