pymmcore-plus/pymmcore-plus #433: MDA file writing question

dpshepherd, 2025-02-12:
I was testing file writing for a sequenced acquisition and ran into a question. I don't get the same result if I pass in a directory for output vs a `.zarr`.

I'm using `pymmcore-plus = 0.13.4` and `useq-schema = 0.7.0`

Passing in `output = Path(r"c:/test.zarr")`
<details>

<summary>Results</summary>

```python
from useq import MDAEvent, CustomAction
from pymmcore_plus.core._mmcore_plus import CMMCorePlus
from types import MappingProxyType as mappingproxy
from pathlib import Path

core = CMMCorePlus()
core.loadSystemConfiguration()

image_mirror_step_um = 0.4
image_mirror_range_um = 20.
active_channels= [0]
exposure_channels = [100]
interleaved_acq = True

all_events: list[MDAEvent] = []
DAQ_event = MDAEvent(
    action=CustomAction(
        name="DAQ-projection",
        data = {
            'DAQ-mode' : 'projection',
            'DAQ-image_mirror_step_um' : float(image_mirror_step_um),
            'DAQ-image_mirror_range_um' : float(image_mirror_range_um),
            'DAQ-active_channels' : active_channels,
            "DAQ-exposure_channels_ms": exposure_channels,
            "DAQ-interleaved" : interleaved_acq,
            "DAQ-laser_powers" : [0,1,2,3,5],
            "DAQ-blanking" : bool(True),  
        }
    )
)

all_events.append(DAQ_event)

n_scan_steps = int(image_mirror_range_um/image_mirror_step_um)

for scan_idx in range(n_scan_steps): 
    image_event = MDAEvent(
        exposure=100.,
        index=mappingproxy({'t': 0, 'p': 0, 'g': 0, 'c': 0, 'z': scan_idx}),
        metadata = {
            "image_scan_step_um" : 0.4,
        }
    )
    all_events.append(image_event)

core.mda.run(all_events,output=Path(r"c:/test/"))
```

Console output:

```bash
[02/11/25 22:27:12] INFO     MDA Started: GeneratorMDASequence()                                   _runner.py:376
                    INFO     action=CustomAction(name='DAQ-projection', data={'DAQ-mode':          _runner.py:337
                             'projection', 'DAQ-image_mirror_step_um': 0.4,                                      
                             'DAQ-image_mirror_range_um': 20.0, 'DAQ-active_channels': [0],                      
                             'DAQ-exposure_channels_ms': [100], 'DAQ-interleaved': True,                         
                             'DAQ-laser_powers': [0, 1, 2, 3, 5], 'DAQ-blanking': True})                         
                    INFO     exposure=100.0 properties=[]                                          _runner.py:337
C:\Users\dpshe\miniforge3\envs\opmv2\Lib\site-packages\pymmcore_plus\mda\handlers\_util.py:32: UserWarning: This sequence is a placeholder for a generator of events with unknown length & shape. Iterating over it has no effect.
  main_sizes = dict(seq.sizes)
[02/11/25 22:27:17] INFO     MDA Finished: GeneratorMDASequence()                                  _runner.py:463
C:\Users\dpshe\miniforge3\envs\opmv2\Lib\site-packages\pymmcore_plus\mda\handlers\_ome_zarr_writer.py:215: UserWarning: This sequence is a placeholder for a generator of events with unknown length & shape. Iterating over it has no effect.
  sizes = {**seq.sizes}
```

Screenshot of inner directory:

[image]

The `.zattrs` has the metadata for all of the images, but there is only one chunk in the zarr.

</details>

Passing in `output = Path(r"c:/test/")`
<details>

<summary>Results</summary>

```python
from useq import MDAEvent, CustomAction
from pymmcore_plus.core._mmcore_plus import CMMCorePlus
from types import MappingProxyType as mappingproxy
from pathlib import Path

core = CMMCorePlus()
core.loadSystemConfiguration()

image_mirror_step_um = 0.4
image_mirror_range_um = 20.
active_channels= [0]
exposure_channels = [100]
interleaved_acq = True

all_events: list[MDAEvent] = []
DAQ_event = MDAEvent(
    action=CustomAction(
        name="DAQ-projection",
        data = {
            'DAQ-mode' : 'projection',
            'DAQ-image_mirror_step_um' : float(image_mirror_step_um),
            'DAQ-image_mirror_range_um' : float(image_mirror_range_um),
            'DAQ-active_channels' : active_channels,
            "DAQ-exposure_channels_ms": exposure_channels,
            "DAQ-interleaved" : interleaved_acq,
            "DAQ-laser_powers" : [0,1,2,3,5],
            "DAQ-blanking" : bool(True),  
        }
    )
)

all_events.append(DAQ_event)

n_scan_steps = int(image_mirror_range_um/image_mirror_step_um)

for scan_idx in range(n_scan_steps): 
    image_event = MDAEvent(
        exposure=100.,
        index=mappingproxy({'t': 0, 'p': 0, 'g': 0, 'c': 0, 'z': scan_idx}),
        metadata = {
            "image_scan_step_um" : 0.4,
        }
    )
    all_events.append(image_event)

core.mda.run(all_events,output=Path(r"c:/test.zarr"))
```

Console output:

```bash
[02/11/25 22:34:44] INFO     MDA Started: GeneratorMDASequence()                                   _runner.py:376
                    INFO     action=CustomAction(name='DAQ-projection', data={'DAQ-mode':          _runner.py:337
                             'projection', 'DAQ-image_mirror_step_um': 0.4,                                      
                             'DAQ-image_mirror_range_um': 20.0, 'DAQ-active_channels': [0],                      
                             'DAQ-exposure_channels_ms': [100], 'DAQ-interleaved': True,                         
                             'DAQ-laser_powers': [0, 1, 2, 3, 5], 'DAQ-blanking': True})                         
                    INFO     exposure=100.0 properties=[]                                          _runner.py:337
[02/11/25 22:34:50] INFO     MDA Finished: GeneratorMDASequence()                                  _runner.py:463
```

Screenshot of directory:
[image]

</details>

Should I setup up in a different way to use the zarr writer?

tlambert03, 2025-02-12:
the docs for the output argument here sort of explain this what's going on here: https://pymmcore-plus.github.io/pymmcore-plus/api/mda/#pymmcore_plus.mda._runner.MDARunner.run

> "output" can be any of the following:
> - A string or Path to a directory to save images to. A handler will be created automatically based on the extension of the path.
> - .zarr files will be handled by `OMEZarrWriter`
> - .ome.tiff files will be handled by `OMETiffWriter`
> - A directory with no extension will be handled by `ImageSequenceWriter`
> - A handler object that implements the DataHandler protocol, currently meaning it has a frameReady method. See mda_listeners_connected for more details.

cutting to the chase, if you pass a string or a path, you leave it up to pymmcore-plus what handler to pick, and a bare directory goes to `ImageSequenceWriter`

For you, I'd suggest explicitly using a handler from [`pymmcore_plus.mda.handlers`](https://github.com/pymmcore-plus/pymmcore-plus/blob/main/src/pymmcore_plus/mda/handlers/__init__.py)  (these, I'm afraid, aren't in the documentation at the moment).  The one that receives the most attention is [`TensorStoreHandler`](https://github.com/pymmcore-plus/pymmcore-plus/blob/3b1e80600c376150a24bb926fcaa45c15fef6ff2/src/pymmcore_plus/mda/handlers/_tensorstore_handler.py#L35)

```python
from pymmcore_plus.mda.handlers import TensorStoreHandler

handler = TensorStoreHandler(path='some/path', delete_existing=True)
mda.run(..., output=handler)
```

but you can also try OMEZarrWriter which i think writes valid ome-zarr (v 0.4)

```python
from pymmcore_plus.mda.handlers import OMEZarrWriter

handler = OMEZarrWriter(store='some/path', overwrite=True)
mda.run(..., output=handler)
```

dpshepherd, 2025-02-12:
Sorry, I wasn't clear. The ome-zarr only has 1 out of 50 images written to disk. The sequence of tiffs has all 50.

From the docs, this seems like a bug to me.

tlambert03, 2025-02-12:
ah, sorry, i didn't look closely enough.  Yeah, sounds like a bug

dpshepherd, 2025-02-12:
`TensorStoreHandler` gives a different output. Now there are 300 sub-folders written, but only the first 50 (folders 0 - 49) contain data. The rest of the sub-folders are empty.

[image]

dpshepherd, 2025-02-12:
Also, there is no `.zattrs` file. Looking at the `TensorstoreFileHandler` code, I would expect it to write it if/when `finalize_metadata()` is called. The other two handlers I tried in the first post do write the metadata.

I get the same result (300 folders and no `.zattrs`) if I use one `CustomAction` `MDAEvent` followed by (1) sequence-able `MDAEvents` as in my first post's example or (2) non-sequence-able `MDAEvents`.

If I remove the first `CustomAction` `MDAEvent`, I get the same result (300 folders and no `.zattrs`).

tlambert03, 2025-02-12:
thanks, this is the bit I was referring to when I said that there were things I was dissatisfied with surrounding the writing/visualization :) https://github.com/pymmcore-plus/pymmcore-gui/issues/34#issuecomment-2644143083

basically, want to take a fine-toothed-comb to the io pipeline; and these details are helpful, thanks!  Will need a bit more time to dig in.  If you're so inclined, feel free to much about with the handler code itself, you can also write your own (the only method the object really needs is a `frameReady()` method, but can also take `sequenceStarted` or `sequenceFinished`)

hopefully more soon

dpshepherd, 2025-02-12:
> thanks, this is the bit I was referring to when I said that there were things I was dissatisfied with surrounding the writing/visualization :) https://github.com/pymmcore-plus/pymmcore-gui/issues/34#issuecomment-2644143083

I wondered, but wasn't sure if you meant something specific to the GUI or more general. Regardless, this is helpful to know!

tlambert03, 2025-02-12:
> we don't dynamically expand the zarr store

yeah, this is one of the biggest issues.  I started without that, but that doesn't handle the event-driven cases for @wl-stepp and others well.  I think at the moment, that single class is too overloaded: by making the complicated things possible, we've made the simple things complex.  So I may have a more formal way to determine whether a sequence is *guaranteed* to produce deterministic outputs or not, and then pick a handler accordingly (rather than having the handler lazily ready to handle anything that comes at it)

wl-stepp, 2025-02-12:
Yeah, sounds like a good idea to me. We've not had any problems with the resizing so far I think. The challenge is more to make the output readable in a nice way by other programs (separate channels in napari etc). This is something that ndv might do directly from the index information? Other than that we are working towards a translation into a valid ome.zarr from the tensorstore to load in napari etc for post processing.

dpshepherd, 2025-02-21:
I believe the reason we are getting a flat array is that we don't have list of `MDAEvents`, not an `MDASequence`.

[Here](https://github.com/pymmcore-plus/pymmcore-plus/blob/72e724c9cc5fd02482091311cfb11652e683ad38/src/pymmcore_plus/mda/handlers/_tensorstore_handler.py#L272) the code is looking at `seq` object provided to the `TensorStoreHandler`.

When we call `mda.run()`, this [code](https://github.com/pymmcore-plus/pymmcore-plus/blob/72e724c9cc5fd02482091311cfb11652e683ad38/src/pymmcore_plus/mda/_runner.py#L226) checks if it received `list[MDAEvent]` or `MDASequence`. If it is not an `MDASequence`, it instead returns a `GeneratorMDASequence()`. That triggers the `TensorStoreHandler` to create a flat store of shape `(nframes, y, x)`.

Does all that seem correct?

To expedite our testing, I made a new version of the `TensoreStoreHandler` that accepts the maximum indices from our `list[MDAEvent]` and uses those initialize the zarr store. With that, we get the appropriately shaped zarr store. In the long run, it might be worth modify the logic to also work with `list[MDAEvent]` that is finite and the handler does't need to deal with non-deterministic events.

tlambert03, 2025-02-21:
Yep that sounds correct. 

Basically we punted for the moment on anything that doesn't come in as nice tidy predetermined mdasequence.  Even a list of events was cast to a flat array (so as to not have to determine if it's jagged, etc...).  I would say it's perfect reasonable for you to subclass and modify the writer as needed for now.  While we work on more generally applicable handlers

dpshepherd, 2025-02-21:
From my perspective, this question is answered and we have a working solution. I'll close it, but feel free to re-open if you want.

closed 2025-02-21 by dpshepherd