pymmcore-plus/pymmcore-plus #435: No Metadata object returned from CMMCorePlus.popNextImageAndMD

Gronemeyer, 2025-02-16:
`python version 3.13.1`
`pymmcore_plus version 0.13.0`

I noticed my metadata output from an MDA hardware sequence no longer contains camera metadata for each frame as it previously did. After investigating, I understand that the MDA Runner utilizes the `popNextImageAndMD` method when capturing frames from the Core. So, I made a simplified tester function and noticed that I _can_ receive camera metadata when using the `getLastImageAndMD` while running a continuous sequence acquisition. 

### Tester:
```python
from pymmcore_plus import CMMCorePlus

def test_mmcore_mda():
    import time
    
    mmcore = CMMCorePlus()
    mmcore.loadSystemConfiguration("C:/Program Files/Micro-Manager-2.0/mm-sipefield.cfg")
    
    mmcore.startContinuousSequenceAcquisition()
    time.sleep(1)
    img, getLast_metadata = mmcore.getLastImageAndMD()
    time.sleep(1)
    img, popNext_metadata = mmcore.popNextImageAndMD()
    mmcore.stopSequenceAcquisition()
    
    print(f"getLastImageAndMD Metadata object: {getLast_metadata}")
    print(F"popNextImageAndMD Metadata object: {popNext_metadata}")
```

### Outputs:
```terminal
getLastImageAndMD Metadata object: {'Camera': 'Dhyana', 'ElapsedTime-ms': '983.57', 'Height': '512', 'ImageNumber': '39', 'PixelType': 'GRAY16', 'ROI-X-start': '0', 'ROI-Y-start': '0', 'Temperature': '12.424', 'TimeReceivedByCore': '2025-02-15 20:40:27.150329', 'Width': '512'}

popNextImageAndMD Metadata object: {}
```

Of course, I am interested in understanding a bit more about how I might investigate further on my end. In my broader program I am using the OME Tiff writer/handler and storing frame metadata in a JSON file similar to the img sequence handler. Particularly, the `TimeReceivedByCore` is the most vital piece of metadata for synchronizing timestamps between multiple concurrent acquisitions. 

### In case it becomes relevant, here is my MicroManager configuration file used in the test_mmcore_mda function:
```cfg
# Generated by Configurator on Fri Sep 20 17:01:22 EDT 2024

# Reset
Property,Core,Initialize,0

# Devices
Device,COM3,SerialManager,COM3
Device,Dhyana,TUCam,TUCam
Device,Arduino-Hub,Arduino,Arduino-Hub
Device,Arduino-Switch,Arduino,Arduino-Switch
Device,Arduino-Shutter,Arduino,Arduino-Shutter

# Pre-init settings for devices
Property,Arduino-Hub,Logic,Normal
Property,Arduino-Hub,Port,COM3

# Pre-init settings for COM ports
Property,COM3,AnswerTimeout,500.0000
Property,COM3,BaudRate,57600
Property,COM3,DTR,Disable
Property,COM3,DataBits,8
Property,COM3,DelayBetweenCharsMs,0.0000
Property,COM3,Fast USB to Serial,Disable
Property,COM3,Handshaking,Off
Property,COM3,Parity,None
Property,COM3,StopBits,1
Property,COM3,Verbose,1

# Hub (parent) references
Parent,Arduino-Switch,Arduino-Hub
Parent,Arduino-Shutter,Arduino-Hub

# Initialize
Property,Core,Initialize,1

# Delays

# Focus directions

# Roles
Property,Core,Camera,Dhyana
Property,Core,Shutter,Arduino-Shutter
Property,Core,AutoShutter,0

# Camera-synchronized devices

# Labels
# Arduino-Switch
Label,Arduino-Switch,4,Blue
Label,Arduino-Switch,2,Violet

# Configuration presets
# Group: Channel
# Preset: Blue
ConfigGroup,Channel,Blue,Arduino-Switch,State,4

# Preset: Violet
ConfigGroup,Channel,Violet,Arduino-Switch,State,2

# Group: System
# Preset: Startup
ConfigGroup,System,Startup,Dhyana,Binning,512x512(4x4Bin)
ConfigGroup,System,Startup,Dhyana,Exposure,19.9980
# ConfigGroup,System,Startup,Dhyana,Gain,HDR
ConfigGroup,System,Startup,Dhyana,Output Trigger Port,2
ConfigGroup,System,Startup,Arduino-Switch,Sequence,On
```

Gronemeyer, 2025-02-16:
Was able to test in another conda environment with `pymmcore_plus version 0.12.0` and did not have the same issue. Therefore, I suspect the issue has arisen either from version 0.13.0 or a more specific dependency issue between my virtual environments

tlambert03, 2025-02-16:
thanks for the very clear issue.  with git bisect i have this changing at #413   and [this (erroneous) line](https://github.com/pymmcore-plus/pymmcore-plus/pull/413/files#diff-ef27d81e1fb28d6095de5e5db5411c37b8d780decea194ec6a393834793b0cddR649)

will fix, add a test, and cut a new release today.

thanks for catching this!  apologies

closed 2025-02-16 by tlambert03

tlambert03, 2025-02-16:
will be fixed in v0.13.5, which is releasing now

PR #436 (2025-02-16): fix: Fix missing metadata in popNextImageAndMD
fixes #435

This also fixes what should have been a fine call to `md = Metadata(md)` by making `pymmcore_plus.Metadata` behave *both* as a subclass of `pymmcore.Metadata`, and also a (virtual) subclass of `abc.collections.Mapping`