pymmcore-plus/pymmcore-plus #446: bug: Consider preserving hardware state before MDA

gselzer, 2025-03-11:
Last week @marktsuchida and I noticed that after running a MDA the hardware state during the final state persists after the final snap.

mmstudio seems to instead revert to the state existing before the MDA started. This is easily seen with the demo config if you:

1. Set the `Channel` Group to the `DAPI` preset.
2. Run a MDA with the `DAPI` channel and then the `Cy5` channel.

In mmstudio, once the MDA completes, `Channel` will revert to `DAPI`. The `GroupPresetTableWidget`'s `Channel` group will remain set to `Cy5`.

In this case, the mmstudio behavior seems better to me. If your settings are configured for optimal panning (maybe snapping one channel is much faster or less toxic than the other), you likely want to revert to those settings before your next MDA. However I'm too naive to suggest that this is the correct behavior. Thoughts @fdrgsp @tlambert03 @marktsuchida?

gselzer, 2025-03-11:
> And all state or just some state? For example, do we necessarily move the stage back to where it was to begin with?

Well there's only so much state that can be affected by an MDA, so I suppose we only need to consider stage positions and channel?

> Hmm, but why?...It feels to me like something that should be up to the user, a configuration option

A configuration option may make sense. @marktsuchida may know any reasons for this behavior in mmstudio better than I...I'm mostly interested in fostering this conversation because it came up in our in-person discussion.

tlambert03, 2025-03-11:
Yes thanks for opening,  it’s definitely worth having an “answer” for