pymmcore-plus/pymmcore-plus #447: Sending command to hardware after sequenced acquisition started

dpshepherd, 2025-03-12:
More questions!

We have an acquisition mode that at a high level looks like this:

1. Setup stage controller in special mode where it knows the distance to move at a constant speed using serial commands. Stage then waits for a "start" serial command.
2. Start camera in sequenced acquisition mode **BUT** the camera does not actually start acquiring until a TTL trigger is received in the input port
3. Send the stage controller command to start the constant speed stage scan. The stage controller automatically sends a TTL output when it passes the predefined start position, which then starts the camera

This might be a better question for the right `MDAEvent` structure, but I realized that this is a bit a tricky one to get right because we want the `mmcore.startSequenceAcquisition()` to be called for a hardware triggered event sequenced that we have generated, but also need to execute another event that won't be hardware sequenced, so the events would not collapse if we insert it there.

Any advice on how best to accomplish this?

dpshepherd, 2025-03-13:
I forgot we had this conversion, #318 and and you implemented #319. I'll give that a shot.

tlambert03, 2025-03-13:
ah, yes.  Sorry for the delayed response.  was out of town.  I was going to mention #319 and `post_sequence_started` ... but I wasn't entirely sure whether that would work in this case.   I *think* it will, but I guess the one question I have is what exactly will happen (with the camera) when that line `_mmc.startSequenceAcquisition` is called [just before](https://github.com/pymmcore-plus/pymmcore-plus/blob/a50a45b52da0b253811b19d702795b1ddd047396/src/pymmcore_plus/mda/_engine.py#L490) `post_sequence_started`.  Will it not go ahead and start the camera sequence for you?  (I suppose it won't, as long as micromanager doesn't think your camera itself is sequenceable right?)  Anyway.  That would be my main question... but let me know how it goes implementing the `post_sequence_started` method in your engine

dpshepherd, 2025-03-13:
> ah, yes. Sorry for the delayed response. was out of town

you are always super fast, don't apologize!

> but I wasn't entirely sure whether that would work in this case. I think it will, but I guess the one question I have is what exactly will happen (with the camera) when that line [...]

Starting the camera is actually the desired behavior. There is an external trigger mode for Hamamatasu cameras called `START` that allows you to start the camera, but not actual begin returning frames until it sees a TTL pulse on the input port.

tlambert03, 2025-03-13:
ah perfect

dpshepherd, 2025-03-20:
This approach works for our needs!

closed 2025-03-20 by dpshepherd