pymmcore-plus/pymmcore-plus #457: Question: How to add a delay before MDA to allow brightfield lamp to reach full intensity

boydcpeters, 2025-04-04:
Hi everyone,

I'm pretty new to Micro-Manager and `pymmcore-plus`, and I’m still learning the ropes. First of all, thank you for the fantastic work on these tools—despite being new, I've really enjoyed working with them!

I have a quick question and wasn't 100% sure if this is the best place to ask. Should I be posting this kind of question here, or would it be better suited for [image.sc](https://forum.image.sc/) or [Microforum](https://forum.microlist.org/)?

---

### Issue Description

I'm working with a **Nikon Ti microscope** and trying to perform a **brightfield grid acquisition**. The setup works well, except for one small hiccup:

When the MDA sequence begins, the **brightfield lamp takes a few seconds (1–3s) to reach the configured intensity** after being turned on. Since I'm imaging with a 30 ms exposure time, the first few images are dim because the lamp hasn't fully "warmed up" yet.

Even if I **manually turn the lamp on and wait before starting the acquisition**, the MDA sequence **still turns it off and back on again**, restarting the delay.

After the initial startup, everything works beautifully when setting `keep_shutter_open_across=("p", "c", "g")`. It's just this initial intensity ramp-up that's causing issues in the first few images.

---

### ⚙️ Configuration Notes

- The **AutoShutter is connected to the Dia lamp**, since we’re using a **LU4A laser box**, but we **can’t control this from Micro-Manager or pymmcore-plus**.
- Full config file is included below for reference.

---

### What I've Tried

I attempted to manually turn on the brightfield lamp and insert a delay before running the MDA:

```python
mmc.getDeviceObject("TIDiaLamp").getPropertyObject("State").value = 1
mmc.getDeviceObject("TIDiaLamp").getPropertyObject("Intensity").value = 6
mmc.waitForSystem()
time.sleep(5)  # Wait for the lamp to warm up
```

But as mentioned, the MDA engine turns the lamp off and then on again at the start of acquisition, negating the manual warm-up.

---

### What I’m Looking For

Is there a recommended way to introduce a delay after the lamp turns on but before the first image is acquired? Or perhaps a way to prevent the MDA engine from toggling the lamp at the start, if it’s already in the correct state?

Also—just in case—am I missing something obvious in my configuration that could solve this?

---

### Config file

<details><summary>Click to expand full config</summary>

```text
# Generated by Configurator on Fri Mar 21 11:17:29 CET 2025

# Reset
Property,Core,Initialize,0

# Devices
Device,TIScope,NikonTI,TIScope
Device,TIDiaLamp,NikonTI,TIDiaLamp
Device,TINosePiece,NikonTI,TINosePiece
Device,TIFilterBlock1,NikonTI,TIFilterBlock1
Device,TILightPath,NikonTI,TILightPath
Device,TIZDrive,NikonTI,TIZDrive
Device,TIXYDrive,NikonTI,TIXYDrive
Device,TIPFSOffset,NikonTI,TIPFSOffset
Device,TIPFSStatus,NikonTI,TIPFSStatus
Device,TITIRF,NikonTI,TITIRF
Device,Camera-1,PVCAM,Camera-1

# Pre-init settings for devices

# Pre-init settings for COM ports

# Hub (parent) references
Parent,TIDiaLamp,TIScope
Parent,TINosePiece,TIScope
Parent,TIFilterBlock1,TIScope
Parent,TILightPath,TIScope
Parent,TIZDrive,TIScope
Parent,TIXYDrive,TIScope
Parent,TIPFSOffset,TIScope
Parent,TIPFSStatus,TIScope
Parent,TITIRF,TIScope

# Initialize
Property,Core,Initialize,1

# Delays
Delay,TINosePiece,1000.0
Delay,TIFilterBlock1,1000.0
Delay,TILightPath,1000.0

# Focus directions
FocusDirection,TIZDrive,0
FocusDirection,TIPFSOffset,0

# Roles
Property,Core,Camera,Camera-1
Property,Core,Focus,TIPFSOffset
Property,Core,AutoShutter,1

# Camera-synchronized devices

# Labels
# TINosePiece
Label,TINosePiece,5,6-Unknown
Label,TINosePiece,4,5-Plan Apo VC 60x NA 1.20 WI
Label,TINosePiece,3,4-Plan Apo 60x NA 1.20 WI
Label,TINosePiece,2,3-Plan Apo TIRF 100x NA 1.45 Oil
Label,TINosePiece,1,2-Plan Apo 40x NA 0.95 Dry
Label,TINosePiece,0,1-Plan Apo 20x NA 0.75 Dry
# TIFilterBlock1
Label,TIFilterBlock1,5,6-Cy3b:
Label,TIFilterBlock1,4,5-N-ST2
Label,TIFilterBlock1,3,4-TRF89
Label,TIFilterBlock1,2,3-C3:48
Label,TIFilterBlock1,1,2-C2: 5
Label,TIFilterBlock1,0,1-DC: F
# TILightPath
Label,TILightPath,3,4-Left80
Label,TILightPath,2,3-Right100
Label,TILightPath,1,2-Left100
Label,TILightPath,0,1-Eye100

# Configuration presets
# Group: Channel
# Preset: Brightfield
ConfigGroup,Channel,Brightfield,TIDiaLamp,Intensity,6
ConfigGroup,Channel,Brightfield,TIDiaLamp,State,1
ConfigGroup,Channel,Brightfield,TIFilterBlock1,Label,4-TRF89
ConfigGroup,Channel,Brightfield,Core,Shutter,

# Preset: Cy3B
ConfigGroup,Channel,Cy3B,TIDiaLamp,Intensity,0
ConfigGroup,Channel,Cy3B,TIDiaLamp,State,0
ConfigGroup,Channel,Cy3B,TIFilterBlock1,Label,6-Cy3b:
ConfigGroup,Channel,Cy3B,Core,Shutter,

# PixelSize settings
# Resolution preset: 60x
ConfigPixelSize,60x,TINosePiece,Label,5-Plan Apo VC 60x NA 1.20 WI
PixelSize_um,60x,0.108333
PixelSizeAffine,60x,0.108333,0.0,0.0,0.0,0.108333,0.0
```

</details>

---

### Minimal reproducible example

Here’s a simplified script that reproduces the "issue":

<details><summary>Click to expand simplified script</summary>

```python
import time
import useq
from useq import MDASequence, GridWidthHeight
from pymmcore_plus import CMMCorePlus
from pymmcore_plus.mda import mda_listeners_connected
from pymmcore_plus.mda.handlers import ImageSequenceWriter

config_path = "config_path"
mmc = CMMCorePlus.instance()
mmc.loadSystemConfiguration(config_path)

# Try to turn on the brightfield lamp before the acquisition to overcome start-up time
mmc.getDeviceObject("TIDiaLamp").getPropertyObject("State").value = 1
mmc.getDeviceObject("TIDiaLamp").getPropertyObject("Intensity").value = 6

mmc.waitForSystem()
# Wait a bit until the brightfield lamp is at set intensity (in reality it takes 1-3 seconds)
time.sleep(5)

fov_width_um = 194.94
fov_height_um = 194.94
perc_overlap = 0.2

sequence = MDASequence(
    channels=[useq.Channel(group="Channel", config="Brightfield", exposure=30)],
    grid_plan=GridWidthHeight(
        width=1000,
        height=1000,
        overlap=(fov_width_um * perc_overlap, fov_height_um * perc_overlap),
        fov_width=fov_width_um,
        fov_height=fov_height_um,
        relative_to="center"
    ),
    keep_shutter_open_across=("p", "c", "g",),
    axis_order="pcg",
)

# Start the acquisition and write the files
with mda_listeners_connected(ImageSequenceWriter("data_dir", overwrite=True)):
    mmc.mda.run(sequence)

mmc.waitForSystem()
print("Sequence finished!")
```

</details>

Thanks so much in advance for any pointers you can give! I really appreciate the help.

tlambert03, 2025-04-04:
welcome @boydcpeters, thanks for the thorough issue!  There are many angles from which this can be attacked.

> Should I be posting this kind of question here

you're always welcome to ask here.  I do also pay attention to image.sc ... so either works.

> Is there a recommended way to introduce a delay after the lamp turns on but before the first image is acquired?

The first thing I would try (correct me if I'm wrong @marktsuchida) is to use [setDeviceDelayMs](https://pymmcore-plus.github.io/pymmcore-plus/api/cmmcoreplus/#pymmcore.CMMCore.setDeviceDelayMs), which is core method that modifies how long a device might wait when you call [waitForDevice](https://pymmcore-plus.github.io/pymmcore-plus/api/cmmcoreplus/#pymmcore.CMMCore.waitForDevice).  So:

```python
mmc.setDeviceDelayMs("TIDiaLamp", 5000)

# after which:
mmc.setProperty("TIDiaLamp", "State", 1)
mmc.waitForDevice("TIDiaLamp")  # should now work
```

> But as mentioned, the MDA engine turns the lamp off and then on again at the start of acquisition, negating the manual warm-up.

Yeah, this is a good point.  First, lemme point you to the code in the [MDA engine](https://github.com/pymmcore-plus/pymmcore-plus/blob/main/src/pymmcore_plus/mda/_engine.py) that may modify the state of the shutter (according to whether `autoShutter` was set in the core, and depending on whether you've set `keep_shutter_open`, (and welcome you to suggest ways in which that code code be modified to avoid unnecessary changes for your case, without breaking existing functionality):

1. At the beginning of *every* event, [this block](https://github.com/pymmcore-plus/pymmcore-plus/blob/d4562125bff30fac88a84e43f0602aa210208776/src/pymmcore_plus/mda/_engine.py#L268-L279) is checked.
2. and then again [here](https://github.com/pymmcore-plus/pymmcore-plus/blob/d4562125bff30fac88a84e43f0602aa210208776/src/pymmcore_plus/mda/_engine.py#L302-L303) depending on the `keep_autoshutter_open` setting.

Now, I'm *hoping* that just setting `mmc.setDeviceDelayMs("TIDiaLamp", 5000)` above will make it such that *any* change in the state of the light will wait for warmup, and it will just work.  But if that doesn't work for you, perhaps you could help us identify where in our engine code improvements could be me for your case.

let's start there and see what we get?

marktsuchida, 2025-04-08:
Some devices honor the delay set by `setDeviceDelayMs()`; others (those that can detect their motion completion) don't.
For the devices that do, the delay can be configured in the hardware configuration.

For the Nikon Ti `TIDiaLamp`, the delay setting should work, I think.

boydcpeters, 2025-04-09:
Thank you for the input, @marktsuchida, especially this comment:

> others (those that can detect their motion completion) don't.

This was super helpful, since I also tried to add a delay after a XYStage change, but setting a delay didn't do anything, but that makes perfect sense now. I fixed that by connecting the `XYStagePositionChanged` event as follows:

```python
mmc.events.XYStagePositionChanged.connect
def pause_before_imaging():
    time.sleep(0.5)
```

And @tlambert03, your suggestion to use `mmc.setDeviceDelayMs("TIDiaLamp", 5000)` worked beautifully! This was exactly what I needed, thank you so much! The acquisition waits now for long enough so the there are no changes in brightness between FOV 1 and FOV 2.

closed 2025-04-09 by boydcpeters