pymmcore-plus/pymmcore-plus #478: Error during MDA in StateDevice if the state doesn't have a label

jeskesen, 2025-07-01:
When running an MDA sequence using a TriggerScopeMM device, I got the following error:  

> [06/20/25 13:54:15] INFO     MDA Started: GeneratorMDASequence()                                                                                                                            _runner.py:376
>    INFO     channel=Channel(config='State0', group='Channel - LF') exposure=10.0 z_pos=-10.0 properties=[PropertyTuple(device_name='TS1_TTL1-8', property_name='State',    _runner.py:337                             property_value='1')] z_sequence='(615 items)' property_sequences='(2 items)'
> [06/20/25 13:54:16] ERROR    Error getting properties ('State', 'Label') on TS1_TTL1-8: Cannot get value of property "Label". Cannot ensure signal emission                           _mmcore_plus.py:2124
> 
> Aborted!

I was confused, because I only tried to set the state.  Tracing the error message to [here](https://github.com/pymmcore-plus/pymmcore-plus/blob/61119dd0f362e772227b1f3d4d20e0ed53485cec/src/pymmcore_plus/core/_mmcore_plus.py#L2389), I see (a few lines above) that the label is explicitly added to the tuple (for context, see #105 ).  

As far as I can tell, the [TriggerscopeMMTTL device](https://github.com/micro-manager/mmCoreAndDevices/blob/main/DeviceAdapters/TriggerScopeMM/TriggerScopeMMTTL.cpp) never calls SetPositionLabel().  Should the label be required?  I think this particular use case would work fine without it.

Another solution would be to add the label to the TriggerScopeMMTTL class, [as the older implementation did](https://github.com/micro-manager/mmCoreAndDevices/blob/1352b67e2467b53cd0a6e4f1aa5adc9e4126ccdd/DeviceAdapters/TriggerScope/TriggerScope.cpp#L769).  @marktsuchida

tlambert03, 2025-07-02:
There's a couple things going on here I guess,  so I'll just comment on the python side:

The purpose of the [code you found](https://github.com/pymmcore-plus/pymmcore-plus/blob/61119dd0f362e772227b1f3d4d20e0ed53485cec/src/pymmcore_plus/core/_mmcore_plus.py#L2389) is to generally ensure that the  `core.propertyChanged` event is emitted whenever a property change occurs.  (You can see exactly what we're trying to ensure in the tests added in #105).  The C++ core code unfortunately doesn't yet guarantee this (we're steadily working on it), and it's pretty important for a lot of application-level things, so pymmcore-plus does do a quick `getValue` as you found there.  I've also struggled a bit with the duplicate-source-of-truth for state devices:  Really, only one of those properties (state OR label) should be considered as writeable, and the other should be read only... but that's not the case, so additional considerations have to be made.  If only one were writable, we wouldn't have the ability for devices to sometimes not implement Label, as found here.

And yes, there are potential costs to this, both in terms of performance, and in error handling when setting state.

However, I'm surprised that to see that your thing aborted:

```
[06/20/25 13:54:16] ERROR Error getting properties ('State', 'Label') on TS1_TTL1-8: Cannot get value of property "Label". Cannot ensure signal emission _mmcore_plus.py:2124

Aborted!
```

Did you abort that manually?  I would have expected this to have simply been logged, but your MDA to continue as normal.  Regardless of how we fix the TriggerScope Label issue, i would still want pymmcore-plus to not stop your MDA because of this

jeskesen, 2025-07-02:
Ok.  It sounds like it needs to added to the C++ side.  I'll add it in my mmCoreAndDevices fork, and submit a PR when I've verified it fixes this python error.  Sound good?  If there's a good reason **not** to add it there, you can simply reject the PR.

> Did you abort that manually?

Yes.  The acquisition hung at this point.  The error message _seemed_ like it could be a smoking gun, but it could be totally unrelated.  Sorry for the confusion.

tlambert03, 2025-07-02:
> The error message seemed like it could be a smoking gun,

it does definitely seem related.

@marktsuchida, can you think of any reason why calling

```python
try:
    core.getProperty('label', 'NonExistent')
except:
    pass
```

would leave the system in an inconsistent state that would have broken the MDA for any reason?

tlambert03, 2025-07-02:
possible solution for the config file bloat https://github.com/micro-manager/mmCoreAndDevices/pull/680 ... that would then let us go back and double check that state devices are implemented uniformly

jeskesen, 2025-07-02:
@tlambert03 , back on the python side, if the label is considered to be required, do you think there could/should be an assertion that [the number of labels == the number of states] when loading the config so that the error happens before the mda is running?

jeskesen, 2025-07-02:
> I'll add it in my mmCoreAndDevices fork, and submit a PR when I've verified it fixes this python error.

In case it wasn't obvious, I'm holding off on this because there's discussion on how to proceed.

tlambert03, 2025-07-02:
> if the label is considered to be required, do you think there could/should be an assertion that [the number of labels == the number of states] when loading the config so that the error happens before the mda is running?

It's not that the label is required in the sense that the number of states and labels must be the same... it's that we had previously  assumed that all state devices create a property named Label.

Yes, we could theoretically, after device initialization, maintain a list of non-compliant state devices.  But that seems to me like a pretty gross workaround.  Ultimately, I would be pretty surprised (and need more of a smoking gun) if simply querying getProperty(..., "Label") is the real problem here.  It really should just log an error and move on with things

jeskesen, 2025-07-03:
Sorry for the delay, I can confirm that the once smoking gun is now a red herring, to mix my metaphors.  If we expect the program to proceed without the label, maybe it makes sense to demote the message from an error to a warning?

tlambert03, 2025-07-03:
> Sorry for the delay, I can confirm that the once smoking gun is now a red herring, to mix my metaphors.

:joy:  thanks for confirming!  That's a relief!

> maybe it makes sense to demote the message from an error to a warning?

totally agree. 👍

jeskesen, 2025-08-22:
Thanks all!