pymmcore-plus/pymmcore-plus #486: feat: Check "is-core-like"

gselzer, 2025-07-29:
Over in [pymmcore-remote](https://github.com/pymmcore-plus/pymmcore-remote) @tlambert03 [noted](https://github.com/pymmcore-plus/pymmcore-remote/pull/11#issuecomment-3114884948) that functionality should not rely upon the concrete `CMMCorePlus` type to implement behavior, as it prevents proxying a `CMMCorePlus`. When it came up, I was trying to use pymmcore-remote's `MMCorePlusProxy` within pymmcore-widget's [`StageWidget`](https://github.com/pymmcore-plus/pymmcore-widgets/blob/bbacae1cd6e204b898faa318fac5cc8ba8614b4a/src/pymmcore_widgets/control/_stage_widget.py#L193) (which does an `isinstance` check [here](https://github.com/pymmcore-plus/pymmcore-widgets/blob/bbacae1cd6e204b898faa318fac5cc8ba8614b4a/src/pymmcore_widgets/control/_q_stage_controller.py#L88-L91))

Not sure about the best implementation here (could be some `is_corelike` function here, or could write a `Protocol`)...

tlambert03, 2025-07-29:
i'm tempted to say that this sort of this should be simple little helper functions in target libraries.  We only really know of 2 proxies at the moment (local and remote CMMCorePlus) ... and I don't expect there to be many more in the future, and that can be solved with `isinstance(obj, (CMMCorePlus, MMCorePlusProxy))`.  I'm not sure it warrants a new base-level api from `pymmcore-plus`?

tlambert03, 2025-07-30:
Also one more thing: that isinstance check you point to in widgets could just as easily be "is not None"...

gselzer, 2025-09-07:
Sorry, I guess this issue slipped my mind 🫣 

You're definitely right that there are other options beyond a protocol, but seeing as it did slip my mind I think it's probably not necessary for this to happen (anytime soon). Can always reconsider implementing some form of isinstance helper functionality later if there is further need.

closed 2025-09-07 by gselzer