pymmcore-plus/pymmcore-plus #493: issue: circular buffer is empty when calling earlier `popNextImage()` after continous acquisition

jacopoabramo, 2025-08-25:
I'm using a Daheng Galaxy camera for a project and I want to drive it with `pymmcore-plus`. I have something like this:

```py
import pymmcore_plus as mmc

core = mmc.CMMCorePlus()

core.loadDevice("MimirDetector", "DahengGalaxy", "DahengCamera")
core.initializeDevice("MimirDetector")
core.setCameraDevice("MimirDetector")
core.startContinuousSequenceAcquisition()

while True:
    core.popNextImage()
    print("Got image")
```

Due to the fact that most probably the camera has still not acquired an image, I get the following:

```
Traceback (most recent call last):
  File "C:\sandbox\python\galaxy_mmcore_sandbox.py", line 12, in <module>
    core.popNextImage()
  File "C:\sandbox\python\.venv\lib\site-packages\pymmcore_plus\core\_mmcore_plus.py", line 787, in popNextImage
    img: np.ndarray = super().popNextImage()
IndexError: Circular buffer is empty.
```

is there anyway to add some sort of waiting condition to wait when there's something in the buffer rather than throwing an exception? I tried both `waitForDevice` and `waitForSystem` but unsurprisingly they both didn't work.

gselzer, 2025-08-25:
> is there anyway to add some sort of waiting condition to wait when there's something in the buffer rather than throwing an exception?

To my knowledge there is no event currently - the [`ImagePreview` widget of `pymmcore-widgets` uses a `QTimer`](https://github.com/pymmcore-plus/pymmcore-widgets/blob/eb950ad45b91c169bc5fd6ecc0312ac7ade5af78/src/pymmcore_widgets/views/_image_widget.py#L66) to poll the buffer for new images.

Are you running any event loop within your code already? There's most certainly a timer mechanism that you could use. E.g. for asyncio:
```python
import asyncio
from contextlib import suppress

# Maybe needed if you want core callback events
# from psygnal import set_async_backend

import pymmcore_plus as mmc

core = mmc.CMMCorePlus()
core.loadSystemConfiguration()
core.startContinuousSequenceAcquisition()

async def check_buffer():
    while True:
        with suppress(IndexError):
            arr = core.getLastImage()
            print("got a new image")
        await asyncio.sleep(0.1)

if __name__ == "__main__":
    asyncio.run(check_buffer())
```

gselzer, 2025-08-25:
> [@gselzer](https://github.com/gselzer) thanks for the reply; that could work but I still feel that it's not very optimal (i never trust too much the accuracy of sleep); it would be nice if the circular buffer would behave like a blocking queue with a timeout option in case no image is stored after a certain time but that would mean rummaging on the C++ side so I imagine it's not trivial

Yeah, I hear you, but since the [C++ code](https://github.com/micro-manager/mmCoreAndDevices/blob/ccac7dea805a5e167d3ace1429e5732b1d771ad3/MMCore/MMEventCallback.h#L27) provides no callback feature as of now for inserting an image into the buffer, it would require changes upstream to do better.

> i'm leaving the issue open until tomorrow in case any other comments from interested parties pops up

I'm not as familiar with core internals as others so not a bad idea 😅

dpshepherd, 2025-08-25:
A common strategy we use is to poll how many images are in the circular buffer using the core helper function, `getRemainingImageCount()`

Something like this,
```python
import pymmcore_plus as mmc

core = mmc.CMMCorePlus()

core.loadDevice("MimirDetector", "DahengGalaxy", "DahengCamera")
core.initializeDevice("MimirDetector")
core.setCameraDevice("MimirDetector")
core.startContinuousSequenceAcquisition()

while True:
    while core.getRemainingImageCount() > 0:
      core.popNextImage()
      print("Got image")
```

jacopoabramo, 2025-08-25:
good enough for me, i'll follow the suggested pattern

closed 2025-08-25 by jacopoabramo