pymmcore-plus/pymmcore-plus #503: Second MDA ignores filter block in channel config if filter block was manually changed after first MDA within the same script (Nikon Ti)

boydcpeters, 2025-09-15:
Hi all,

### Issue description

I am working with a **Nikon Ti** microscope and noticed an "issue" when running two multi-dimensional acquisitions (MDAs) in the same script.

Specifically, if I manually set the filter block after an MDA finishes, the next MDA does **not** correctly change to the filter block defined in its channel config. Instead, it stays on the manually-set filter block from after the first MDA.

I’m not sure if this is expected behavior (e.g., the MDA engine reuses its previous configuration), or if there’s a way to reset the engine so that each MDA always starts from a clean state. Any clarification or suggestions would be very helpful. Or if I am not using `pymmcore-plus` properly, please let me know as well.

I’ve attached both a minimal reproducible example and the configuration file.

Maybe some additional information, before any MDA I have started to change the filter block to a "random" position as sometimes `pymmcore-plus` does not seem to read out the correct filter block during initialization. If I set the filter block, then the microscope and `pymmcore-plus` are in sync. Hence, I came across this when putting two MDAs in a single script. I hope this makes some sense, if not please let me know.

Currently, this is not a major issue for us, as we are only imaging in a single channel at the moment so we can hard-code the correct filter block position, but I thought it would be nice to ask to find out what causes this "issue".

### Minimal reproducible example

Here's a simplified script that reproduces the issue:

<details><summary>Click to expand the script</summary>

```python
import time

import useq
from pymmcore_plus import CMMCorePlus
from useq import MDASequence

if __name__ == "__main__":
    bottom = 3133
    top = 3138
    step = 0.25
    go_up = True

    system_config_path = (
        r"..\microscope_configs\NikonTi_left_port_kinetix22_default.cfg"
    )
    mmc = CMMCorePlus.instance()
    mmc = mmc.loadSystemConfiguration(system_config_path)

    sequence = MDASequence(
        channels=(useq.Channel(config="Cy3B", exposure=50),),
        z_plan=useq.ZTopBottom(bottom=bottom, top=top, step=step, go_up=go_up),
        axis_order=("c", "z"),
        keep_shutter_open_across=("c", "z"),
    )

    print("Current filter block: ", mmc.getDeviceObject("TIFilterBlock1"))
    mmc.getDeviceObject("TIFilterBlock1").getPropertyObject("State").setValue(2)
    time.sleep(1)  # Allow the filter block to change, this works.
    mmc.waitForSystem()

    mmc.mda.run(
        sequence
    )  # Here the filter block is correctly changed to the Cy3B filter block

    print("First run is finished.")

    print("Current filter block: ", mmc.getDeviceObject("TIFilterBlock1"))
    mmc.getDeviceObject("TIFilterBlock1").getPropertyObject("State").setValue(2)
    time.sleep(1)  # Allow the filter block to change, this works.
    mmc.waitForSystem()

    # During this second MDA, the filter block is not correctly set to the filter block
    # of the Cy3B config (filter block position 5)... It remains on the filter block at
    # position 2.
    mmc.mda.run(sequence)

    print("Second run is finished.")
```

</details>

### Config file

<details><summary>Click to expand full config</summary>

```text
# Generated by Configurator on Fri Jul 11 11:09:16 CEST 2025

# Reset
Property,Core,Initialize,0

# Devices
Device,TIScope,NikonTI,TIScope
Device,TIDiaLamp,NikonTI,TIDiaLamp
Device,TINosePiece,NikonTI,TINosePiece
Device,TIFilterBlock1,NikonTI,TIFilterBlock1
Device,TILightPath,NikonTI,TILightPath
Device,TIZDrive,NikonTI,TIZDrive
Device,TIXYDrive,NikonTI,TIXYDrive
Device,TIPFSOffset,NikonTI,TIPFSOffset
Device,TIPFSStatus,NikonTI,TIPFSStatus
Device,TITIRF,NikonTI,TITIRF
Device,Camera-1,PVCAM,Camera-1

# Pre-init settings for devices

# Pre-init settings for COM ports

# Hub (parent) references
Parent,TIDiaLamp,TIScope
Parent,TINosePiece,TIScope
Parent,TIFilterBlock1,TIScope
Parent,TILightPath,TIScope
Parent,TIZDrive,TIScope
Parent,TIXYDrive,TIScope
Parent,TIPFSOffset,TIScope
Parent,TIPFSStatus,TIScope
Parent,TITIRF,TIScope

# Initialize
Property,Core,Initialize,1

# Delays
Delay,TIDiaLamp,5000.0
Delay,TINosePiece,1000.0
Delay,TIFilterBlock1,1000.0
Delay,TILightPath,1000.0

# Focus directions
FocusDirection,TIZDrive,0
FocusDirection,TIPFSOffset,0

# Roles
Property,Core,Camera,Camera-1
Property,Core,Shutter,TIDiaLamp
Property,Core,Focus,TIZDrive
Property,Core,AutoShutter,1

# Camera-synchronized devices

# Labels
# TINosePiece
Label,TINosePiece,5,6-Unknown
Label,TINosePiece,4,5-Plan Apo VC 60x NA 1.20 WI
Label,TINosePiece,3,4-Plan Apo SR 60x NA 1.27 WI
Label,TINosePiece,2,3-Plan Apo TIRF 100x NA 1.45 Oil
Label,TINosePiece,1,2-Plan Apo 40x NA 0.95 Dry
Label,TINosePiece,0,1-Plan Apo 20x NA 0.75 Dry
# TIFilterBlock1
Label,TIFilterBlock1,5,6-Cy3b
Label,TIFilterBlock1,4,5-N-ST2
Label,TIFilterBlock1,3,4-TRF89
Label,TIFilterBlock1,2,3-C3:48
Label,TIFilterBlock1,1,2-C2: 5
Label,TIFilterBlock1,0,1-DC: F
# TILightPath
Label,TILightPath,3,4-Left80
Label,TILightPath,2,3-Right100
Label,TILightPath,1,2-Left100
Label,TILightPath,0,1-Eye100

# Configuration presets
# Group: Channel
# Preset: brightfield
ConfigGroup,Channel,brightfield,TIDiaLamp,Intensity,5
ConfigGroup,Channel,brightfield,TIDiaLamp,State,1
ConfigGroup,Channel,brightfield,TIFilterBlock1,Label,4-TRF89

# Preset: Cy3B
ConfigGroup,Channel,Cy3B,TIDiaLamp,Intensity,0
ConfigGroup,Channel,Cy3B,TIDiaLamp,State,0
ConfigGroup,Channel,Cy3B,TIFilterBlock1,Label,6-Cy3b

# Preset: quad-band
ConfigGroup,Channel,quad-band,TIDiaLamp,Intensity,0
ConfigGroup,Channel,quad-band,TIDiaLamp,State,0
ConfigGroup,Channel,quad-band,TIFilterBlock1,Label,4-TRF89

# Group: System
# Preset: Startup

# PixelSize settings
# Resolution preset: 60x VC
ConfigPixelSize,60x VC,TINosePiece,Label,5-Plan Apo VC 60x NA 1.20 WI
PixelSize_um,60x VC,0.108333
PixelSizeAffine,60x VC,0.108333,0.0,0.0,0.0,0.108333,0.0

# Resolution preset: 60x
ConfigPixelSize,60x,TINosePiece,Label,4-Plan Apo SR 60x NA 1.27 WI
PixelSize_um,60x,0.108333
PixelSizeAffine,60x,0.108333,0.0,0.0,0.0,0.108333,0.0

# Resolution preset: 100x
ConfigPixelSize,100x,TINosePiece,Label,3-Plan Apo TIRF 100x NA 1.45 Oil
PixelSize_um,100x,0.065
PixelSizeAffine,100x,0.065,0.0,0.0,0.0,0.065,0.0
```

</details>

tlambert03, 2025-09-15:
thanks for the clear report!  Should be fixed in https://github.com/pymmcore-plus/pymmcore-plus/pull/504

closed 2025-09-15 by tlambert03

boydcpeters, 2025-09-19:
I just had some microscopy time again and tested it out, and now it works like a charm, thanks again!